dtcooper/raspotify #393: Build aarch64 packages

samip5, 2021-05-09:
Currently the Debian repository is only for armhf, most people will use RPi3 and RPi4's in 64-bit mode, so support them.

timfi-amsterdam, 2021-09-08:
this seems to just work if line 32 of install.sh is changed to something like:
`if uname -a | fgrep -ivq -e aarch -e arm; then`

64bit kernel supports 32 bit just fine

samip5, 2021-09-09:
> this seems to just work if line 32 of install.sh is changed to something like:
> `if uname -a | fgrep -ivq -e aarch -e arm; then`
> 
> 64bit kernel supports 32 bit just fine

I think you misunderstood, with that change:
```
 ./install-rasp.sh 
gpg: pub  rsa4096/09146F2F7953A455 2017-06-15  David Cooper <david@dtcooper.com>
gpg: no running gpg-agent - starting '/usr/bin/gpg-agent'
gpg: waiting for the agent to come up ... (5s)
gpg: connection to agent established
OK
deb https://dtcooper.github.io/raspotify raspotify main
Hit:1 https://download.docker.com/linux/ubuntu focal InRelease
Get:2 https://dtcooper.github.io/raspotify raspotify InRelease [1706 B]                                                                                           
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal InRelease                                                                                                        
Get:4 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease [101 kB]                           
Get:5 https://packages.cloud.google.com/apt kubernetes-xenial InRelease [9383 B]
Get:7 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease [114 kB]
Fetched 339 kB in 2s (141 kB/s)                         
Reading package lists... Done
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://dtcooper.github.io/raspotify raspotify InRelease' doesn't support architecture 'arm64'
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package raspotify
```

JasonLG1979, 2021-11-17:
PR's welcome to enable aarch64 support.

JasonLG1979, 2021-11-17:
Duplicate of https://github.com/dtcooper/raspotify/issues/263

closed 2021-11-17 by JasonLG1979