dtcooper/raspotify #394: Update to librespot v0.2.0

JasonLG1979, 2021-05-13:
Librespot was updated to v0.2.0 a few days ago. Along with the usual bug fixes and whatnot the most user noticeable changes are higher bit depth formats are available reducing quantization noise and providing for higher quality software volume. Volume normalization now also uses a real limiter to better adhere to the ReplayGain standard and how official Spotify clients actually work.

AFAIK no args have changed and the everything is backward compatible with previous releases so it should be just a matter of rebuilding the binary.

 https://github.com/librespot-org/librespot/releases/tag/v0.2.0

heinzgruber, 2021-05-22:
Ogg passthrough sounds interesting. Which format uses librespot as default?

JasonLG1979, 2021-05-22:
> Ogg passthrough sounds interesting. Which format uses librespot as default?
> 
> I’m looking for a 1:1 playback with no quality loss.

Spotify streams Ogg vorbis which is lossy. Eventually when the folks behind librespot figure out the new API and Spotify opens up their "HiFi" tier it will be lossless 16 bit 44.1 audio, FLAC I would imagine?

heinzgruber, 2021-05-22:
That sounds great. 

But I wonder what Ogg passthrough exactly means. At the moment I use the 320 kbits option in the config.

JasonLG1979, 2021-05-22:
> But I wonder what Ogg passthrough exactly means. At the moment I use the 320 kbits option in the config.

It works like pipe except instead of pipping PCM It pippes the Ogg Vorbis stream without decoding it.

JasonLG1979, 2021-05-22:
Pipping is one of those things that if you don't know what it's good for you probably don't need it. It's like a catch all option. You can basically put anything at the other end of the pipe that either understands PCM or Vorbis and do whatever you wanted with it. For the basic use case of running Spotify Connect on the same Pi you plan on outputting sound from pipping to another program is overkill IMHO. Unless your Pi is connected to a nice USB DAC your Pi is most definitely the bottleneck as far as sound quality not whatever program is playing the audio.

heinzgruber, 2021-05-23:
> Unless your Pi is connected to a nice USB DAC your Pi is most definitely the bottleneck as far as sound quality not whatever program is playing the audio.  

My Pi is connected to the high end DAC RME ADI-2 DAC. 

If I understand correctly, there is currently an unnecessary conversion from ogg to pcm?

JasonLG1979, 2021-05-23:
> If I understand correctly, there is currently an unnecessary conversion from ogg to pcm?

No. Unless your DAC can decode an Ogg Vorbis stream naively.

JasonLG1979, 2021-05-23:
Ogg Vorbis is a lossy audio format and container like an MP3.

JasonLG1979, 2021-05-27:
@dtcooper Maybe hold off, `0.3.0` it should be released shortly. The only breaking change is `--mixer-linear-volume` has been removed but raspotify doesn't use that in it's default args anyway.

From `0.1.6` to `0.3.0` (once it's released) is a no-brainer update.

dtcooper, 2021-05-27:
Okay thanks for the info. Will track the 0.3.0 release.

giggywithit, 2021-06-01:
> Maybe hold off, `0.3.0` it should be released shortly. The only breaking change is `--mixer-linear-volume` has been removed but raspotify doesn't use that in it's default args anyway.
> 
> From `0.1.6` to `0.3.0` (once it's released) is a no-brainer update.

@JasonLG1979 
What will be the syntax changes for me to keep using the linear volume?
I've been having to do somewhat of a hack to get it to work without crashing, using the following in specific order..

OPTIONS="--mixer=alsa --mixer-linear-volume --mixer-name=Digital"

VOLUME_ARGS="--volume-ctrl=linear --initial-volume=65"

Also, is it possible to get some sort of delay when starting playback?
I have a system that needs 1s to start an amplifier before processing the audio.
The sequence would be - open audio device - then after 1s - process audio.

Thx

JasonLG1979, 2021-06-01:
> What will be the syntax changes for me to keep using the linear volume?

Actual linear or linear to the volume controls in `alsamixer`? Because the controls in `alsamixer` are logarithmic if I remember correctly? Logarithmic is perceptually linear, a true linear volume control doesn't sound linear to the human ear. We don't perceive sound linearly, that's why dB's are log.

There's been a lot of work done by @roderickvd on the volume controls among other things. He's the mastermind behind basically all of the audio quality and volume improvements. I don't generally use volume controls in `librespot` (I just use the volume knob on my receiver) so I set it to fixed and 100 so I am far from an authority. But I'm Pretty sure if you want your volume controls to be linear to `alsamixer`  all you should need is `--mixer=alsa` since logarithmic is the default mode in `librespot` you should be able to skip `--volume-ctrl=linear` unless you want a true linear volume control (that won't match the volume controls in `alsamixer`.

If I have misunderstood or misstated anything please correct me @roderickvd.

> Also, is it possible to get some sort of delay when starting playback?
> I have a system that needs 1s to start an amplifier before processing the audio.
> The sequence would be - open audio device - then after 1s - process audio.

You may be able to use `--onevent` (which allows you to run a program or script when events happen) along with `emit-sink-events` to run a script when `librespot` opens and closes the sink (basically audio device) I'm not sure if that event is blocking or if the script is fired before or after the sink is opened. Again sorry I am not the authority on that. My set up is pretty basic, I've never had the need for anything that complicated.

JasonLG1979, 2021-06-02:
Anyone that want's to play with the dev branch here's a x86_64 Linux binary that I built that you can download with this command (that will not run on a Raspberry Pi):

`wget --no-check-certificate 'https://docs.google.com/uc?export=download&id=17PavN0y1ILhQdfTNlt6wkNRnDoQb0RJd' -O librespot`

And then run `chmod +x librespot` to make it executable.

An ARMv6 binary for testing purposes is coming as soon as it's done compiling. But really compiling `librespot` is super easy and documented pretty well at their [wiki](https://github.com/librespot-org/librespot/wiki/Compiling).

Please though if you do run into any bugs while testing these out do not report them here. Report them to   [librespot](https://github.com/librespot-org/librespot).

JasonLG1979, 2021-06-02:
Here is the ARMv6 binary that will run on a 32bit Raspberry Pi:

`wget --no-check-certificate 'https://docs.google.com/uc?export=download&id=1mqXKACTrIQZ-2C6enDYniE7m2vowcBDT' -O librespot`

Same as before,  run `chmod +x librespot` to make it executable and report any bugs to librespot.

`./librespot -h` will print out the help menu in the terminal giving you a list of possible options and flags to choose from.

roderickvd, 2021-06-02:
> @dtcooper Maybe hold off, `0.3.0` it should be released shortly. The only breaking change is `--mixer-linear-volume` has been removed but raspotify doesn't use that in it's default args anyway.

I don't know about your release cycles but a `0.3.0` release date has not been settled yet. I'd like to release one early summer but it's not just up to me. In the meantime `0.2.0` is a pretty solid release that I think is worth your while.

> > What will be the syntax changes for me to keep using the linear volume?
> 
> Actual linear or matching the volume controls in `alsamixer`? Because the controls in `alsamixer` are logarithmic if I remember correctly? Logarithmic is perceptually linear, a true linear volume control doesn't sound linear to the human ear. We don't perceive sound linearly, that's why dB's are log.
> 
> There's been a lot of work done by @roderickvd on the volume controls among other things. He's the mastermind behind basically all of the audio quality and volume improvements. I don't generally use volume controls in `librespot` (I just use the volume knob on my receiver) so I set it to fixed and 100 so I am far from an authority. But I'm pretty sure if you want your volume controls to match `alsamixer` all you should need is `--mixer=alsa` since logarithmic is the default mode in `librespot` you should be able to skip `--volume-ctrl=linear` unless you want a true linear volume control (that won't match the volume controls in `alsamixer`).

Thanks for the appreciation.

Up to and including `0.2.0` you can't have true linear volume control using the `alsamixer`. Because Alsa's mixer maps everything logarithmically internally, when launching `librespot --volume-ctrl linear` you will be getting a logarithmic control which is generally what you want (or cubic, depending on whether your DAC has hardware volume control).

If you launch with `--volume-ctrl log` then you will get a nonsensical log-on-log mapping, this is probably what you were referring to.

This has been worked around in `dev` for release in `0.3.0`. Doing `--volume-ctrl {linear|log}` will then actually do what you think it'll do.

Unless you have some special requirements, for best sound quality I recommend using `softvol` instead of `alsamixer`, it does volume control in 64-bit floating point which is better than doing it in DAC hardware or Alsa Softvol. In which case `--volume-ctrl {linear|log}` is already working fine.

> You may be able to use `--onevent` (which allows you to run a program or script when events happen) along with `emit-sink-events` to run a script when `librespot` opens and closes the sink [...]

Not natively in `librespot` but maybe you can do something with the GStreamer backend. I don't know.

dtcooper, 2021-06-02:
> @JasonLG1979
> What will be the syntax changes for me to keep using the linear volume?

Unsure until I test the release, post v0.3.0, however I will make sure your config @ `/etc/default/raspotify` is properly `sed`'d to reflect any changed arguments.

dtcooper, 2021-11-01:
Extremely busy week on my end. Will get a chance to delve into this weekend. Thanks for your patience, all. (As always, patches welcome!)

closed 2021-11-17 by JasonLG1979