dtcooper/raspotify #398: multiple raspotify at same time

tonygils, 2021-06-01:
Hi,

Is it possible to install raspotify multiple times and run them at the same time?
so 2 installations with 2 different spotify users on the same raspberry pi.

JasonLG1979, 2021-06-02:
You could theoretically run multiple instances they would just need to have different names AFAIK? Your limiting factors would basically be bandwidth, output devices, and CPU/RAM. If you had a Raspberry Pi 4, a USB hub and a bunch of USB DACs you should be able to run up to maybe 8 instances simultaneously?

Raspotify is a systemd service. Google running multiple instances of the same service, specifically template unit files.

JasonLG1979, 2021-06-02:
The quick and dirty way is to just write a shell script to call several instances of `librespot` with whatever args you want for each instance.

closed 2021-11-17 by JasonLG1979

mmiller42, 2023-03-28:
Would this be something that a wiki article could be written about? I would love to be able to set up Spotify for myself and my partner via my RPi 4 and connect it to our speakers for a budget alternative to wifi speakers.

JasonLG1979, 2023-03-28:
> Would this be something that a wiki article could be written about? I would love to be able to set up Spotify for myself and my partner via my RPi 4 and connect it to our speakers for a budget alternative to wifi speakers.

My above advice still applies. Look up template unit files. You're also going to have to figure out where you want the other config to live. When someone figures everything out I'll add it to the wiki as an unsupported option. It's nothing I care to support or help with.

JasonLG1979, 2023-03-28:
Again you could also just do it quick and dirty with a shell script.

SimonCoulombe, 2023-05-19:
just wondering, would this multiple user setup require multiple DACs and receiver/speaker sets, or could it output to a single set?

JasonLG1979, 2023-05-20:
No. So long as discovery is enabled (it is by default) when one user jumps on it just kicks the previous user.

SimonCoulombe, 2023-05-20:
thanks Jason