dtcooper/raspotify #402: [BUG] Raspotify needs to be restarted after internet break

AdriDevelopsThings, 2021-06-04:
I will reference issue #106 here because I think the author has the same problem as me, but the issue was closed.
### Environment
First of all information about my environment:

Hardware: Raspberry Pi 3B
System: ``Ubuntu 20.04.2 LTS focal``
Kernel: ``5.4.0-1035-raspi``
Version of raspotify: (0.16.0)
```
raspotify:armhf:
  Installed: 0.16.0~librespot.v0.1.3-37-g0adb851
  Candidate: 0.16.0~librespot.v0.1.3-37-g0adb851
```

### Problem

When my internet goes down (e.g. my DSL goes down) my Raspotify player will be removed from my Spotify and I can't control the player. But when my internet goes up again the Raspotify player does not reconnect.

### Workaround

I have to restart the Raspotify service:
```bash
sudo systemctl restart raspotify
```

I hope somebody can help me fix the issue.

Janeai, 2021-06-04:
I think you can try this approach 
https://unix.stackexchange.com/questions/159641/systemd-restart-service-when-ip-address-changed

JasonLG1979, 2021-06-04:
Maybe something like this? My bash foo isn't the greatest and I haven't tested it but it should be at least a starting point. I'm sure there are probably better and more proper ways to check if you have a network connection. ping is also an option. 

```bash
#!/bin/sh

# zero = active, non-zero = not active 
not_active=$(systemctl is-enabled --quiet raspotify)
# echo "$not_active"

bad_status=$(systemctl status raspotify | grep -E "(ERROR|101)")
# echo "$bad_status"

# zero = success/connected, non-zero failure/not connected
not_connected=$(nc -zw1 google.com 443 >/dev/null)
# echo "$not_connected"

if [ "$not_connected" ]; then
    if [ -n "$not_active" ]; then
        sudo systemctl stop raspotify
        # echo "Stop Raspotify"
    fi
else
    if [ -n "$bad_status" ] || [ "$not_active" ]; then
        sudo systemctl restart raspotify
        # echo "Restart Raspotify"
    fi
fi
```

The general idea is that it checks to see if raspotify is active, if there are any errors and if your Pi has an internet connection.

If there is no internet connection it stops the raspotify service. If there is a connection but the service has been stopped because the internet went down or the service has thrown any errors it restarts it.

You can either use a systemd timer unit or a cron job to fire the script at whatever interval you'd like.

giggywithit, 2021-07-04:
Just wanted to say that I encounter the same problems. The activity controls the power of my system and it will hang and remain on until someone has restarted it. Ping, with my router, will not function as intended as the router itself will return the ping when offline.

Not sure how to deal with this one for my mother's machine, as I am not usually around to restart it.

AdriDevelopsThings, 2021-07-07:
> Maybe something like this? My bash foo isn't the greatest and I haven't tested it but it should be at least a starting point. I'm sure there are probably better and more proper ways to check if you have a network connection. ping is also an option.
> 
> ```shell
> #!/bin/sh
> 
> # zero = active, non-zero = not active 
> not_active=$(systemctl is-active --quiet raspotify)
> # echo "$not_active"
> 
> bad_status=$(systemctl status raspotify | grep -E "(ERROR|101)")
> # echo "$bad_status"
> 
> # zero = success/connected, non-zero failure/not connected
> not_connected=$(nc -zw1 google.com 443)
> # echo "$not_connected"
> 
> if [ "$not_connected" ]; then
>     if [ "$not_active" ]; then
>         sudo systemctl stop raspotify
>         # echo "Stop Raspotify"
>     fi
> else
>     if [ -n "$bad_status" ] || [ ! "$not_active" ]; then
>         sudo systemctl restart raspotify
>         # echo "Restart Raspotify"
>     fi
> fi
> 
> exit 0
> ```
> 
> The general idea is that it checks to see if raspotify is active, if there are any errors and if your Pi has an internet connection.
> 
> If there is no internet connection it stops the raspotify service. If there is a connection but the service has been stopped because the internet went down or the service has thrown any errors it restarts it.
> 
> You can either use a systemd timer unit or a cron job to fire the script at whatever interval you'd like.

Thank you for your solution, but it doesn't work. If I execute the script while Raspotify works and my internet is up, it restarts my Raspotify without any reason.
The echo's at the top of the script don't echo anything.

JasonLG1979, 2021-11-24:
@AdriDevelopsThings Like I said I'm not a shell script guru.

Could you update to the latest Raspberry Pi OS version (Bullseye) and Raspotify and see if you're still having this issue. If so I'd be interested in PR's for a proper watchdog service.

JasonLG1979, 2021-12-02:
@AdriDevelopsThings I was just revisiting this and if you're like me and your Pi is used primarily for thing that require an internet connection (for me that's Rapotify and Plex) and it's not really useful otherwise you can go the big hammer route and use `watchdog`. (`sudo apt-get install watchdog`)

https://linux.die.net/man/8/watchdog
https://linux.die.net/man/5/watchdog.conf

I'm still playing with it but I set the timeout to 6 min and for it to ping `8.8.8.8` (Google's public DNS server) every 5 min. The idea being that the Pi will reboot after 6 min if `8.8.8.8` can't be reached.

The relevant values in `/etc/watchdog.conf` being:

```
watchdog-timeout                = 360
watchdog-refresh-use-settimeout = auto
watchdog-refresh-ignore-errors  = yes
interval                        = 300
realtime                        = yes
priority                        = 1
sigterm-delay                   = 30
ping                            = 8.8.8.8
ping-count                      = 10
```

Take those values with a grain of salt though I've just stumbled upon `watchdog` they may be way off but it seems to work?

giggywithit, 2021-12-02:
My workaround on one machine was to set up a USB volume knob and use the push mute as push reboot. If you don't see it on the network, reboot it with the button.

JasonLG1979, 2021-12-02:
Yep mine sits on my receiver. If I need it to reconnect I usually just unplug and plug it back in.

JasonLG1979, 2021-12-03:
I'm going to close this as I feel like monitoring your internet connection status is a little out of scope of Raspotify, @giggywithit and I have offered different sized hammers for the job and if you Google how to monitor your internet connection and/or how to restart a system service in response to your internet connection status there are like a thousand and one ways.

closed 2021-12-03 by JasonLG1979