dtcooper/raspotify #404: raspotify stops when restarting

franvillamil, 2021-06-05:
I restarted raspotify (`sudo systemctl restart raspotify`) after changing the configuration and it stops working. Not sure why, but it seems it only works again after doing a full uninstall and installing it again. I'm not sure why (I'm new to Linux fwiw).

This is what I get from `sudo service raspotify status`:

```
● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2021-06-05 20:51:12 CEST; 9s ago
  Process: 5613 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 5614 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 5615 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --device-type ${DEVICE_TYPE} $BACKEND_ARGS --bitrate ${BITRA
 Main PID: 5615 (code=exited, status=1/FAILURE)

jun 05 20:51:12 raspberrypi librespot[5615]:         --normalisation-pregain PREGAIN
jun 05 20:51:12 raspberrypi librespot[5615]:                         Pregain (dB) applied by volume normalisation
jun 05 20:51:12 raspberrypi librespot[5615]:         --volume-ctrl VOLUME_CTRL
jun 05 20:51:12 raspberrypi librespot[5615]:                         Volume control type - [linear, log, fixed]. Default is
jun 05 20:51:12 raspberrypi librespot[5615]:                         logarithmic
jun 05 20:51:12 raspberrypi librespot[5615]:         --autoplay      autoplay similar songs when your music ends.
jun 05 20:51:12 raspberrypi librespot[5615]:         --disable-gapless
jun 05 20:51:12 raspberrypi librespot[5615]:                         disable gapless playback.
jun 05 20:51:12 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
jun 05 20:51:12 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
 ESCOC

ed; vendor preset: enabled)
Sat 2021-06-05 20:51:12 CEST; 9s ago
aspotify (code=exited, status=0/SUCCESS)
r/cache/raspotify (code=exited, status=0/SUCCESS)
ME} --device-type ${DEVICE_TYPE} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS (code=exited, status=1/FA

ation-pregain PREGAIN
      Pregain (dB) applied by volume normalisation
trl VOLUME_CTRL
      Volume control type - [linear, log, fixed]. Default is
      logarithmic
      autoplay similar songs when your music ends.
gapless
      disable gapless playback.
 process exited, code=exited, status=1/FAILURE
ed with result 'exit-code'.
```

And this is what I see in `apt list`:

```
raspotify/unknown,now 0.16.0~librespot.v0.1.3-37-g0adb851 armhf [installed]
```

giggywithit, 2021-06-21:
I had trouble with my volume args at one point...
apparently they've been changed up some, although I haven't had the chance to test any yet.
Try using default settings and get it to work first, then change the settings one at a time and see what happens

JasonLG1979, 2021-11-19:
It's puking the help menu because you've feed it an invalid argument.

JasonLG1979, 2021-11-21:
`librespot -h` will give a list of valid options and flags and valid values for the options.

closed 2021-11-21 by JasonLG1979