dtcooper/raspotify #407: Can't choose Raspberry in Spotify

LuigiSetola, 2021-07-13:
It's impossible to choose my raspoberry with raspotify in spotify application.
I'm using the last version of raspotify on my raspberry pi4 and raspberry pi zero.

[image]

JasonLG1979, 2021-07-25:
I get a similar message from my daughter's Spotify's account because she does not have a Spotify Premium account. Free accounts are not supported by Raspotify/librespot.

bastiandoetsch, 2021-08-06:
I got the same issue, and I got a Premium account :(. I suppose something else could be the reason.

felixhusse, 2021-08-08:
I get this message when trying to start a Podcast directly. I choose first a music Track (can select raspotify) and then switch to podcast.

JasonLG1979, 2021-08-08:
If it's a matter of podcasts then yes librespot has issues currently with podcasts sometimes. Podcasts were not a thing when librespot was initially written AFAIK. Librespot can only decode Ogg Vorbis audio at the moment which works fine for music 99.9% of the time because that's generally an available format. Some podcasts however are only available in MP3 or AAC formats.

There is a lot of work being done to reverse engineer the latest and greatest Spotify API(s) to stay current and to be able to enable Spotify HiFi when it rolls out. Part of that would be switching to some kind of audio framework maybe or at least a decoder that can handle multiple different formats. The ideal decoder would be able to do MP3, AAC, Ogg Vorbis, and FLAC gapeless, be relatively small, cross-platform and be written in pure Rust. (the current Ogg Vorbis is pure Rust) I think generally the idea is to try to stay away from huge (non-Rust) frameworks like Gstreamer and FFmpeg.

JasonLG1979, 2021-11-17:
I'm going to call this an upstream librespot issue.

closed 2021-11-17 by JasonLG1979