dtcooper/raspotify #408: Troubleshooting No Sound without HDMI (headless) -- Hint: check using 'aplay -l'

rcrouthamel, 2021-07-17:
I was chasing the no sound issue on a RPi 3B.  Initially, with the HDMI monitor connected, I configured /etc/default/raspotify with: ```OPTIONS="--device hw:0,1"``` (as in the instructions and another issue) and this appeared to work well.  On reboot without HDMI, no audio.

What I realized is that these IDs change whether HDMI is connected at boot or not.  The trick is to boot the pi in the configuration you plan to use it, then run the command ```aplay -l``` to figure out what card and device is assigned to the headphone jack.  In my case it was actually 0,0:
```
     $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
  ```

So my configuration worked with ```OPTIONS="--device hw:0,0"```

closed 2021-07-17 by rcrouthamel

bji219, 2021-11-06:
I am trying to use raspotify with a Focusrite 2i2. It is listing the device as a Subdevice. Have you seen anything like this before? I tried changing my ``OPTIONS="--device hw:1,0"`` (see screenshot below). I have also updated the alsa configuration in ``/usr/share/alsa/alsa.conf`` to read ``defaults.ctl.card 0`` and ``defaults.pcm.card 0``. Please let me know what you think! Thanks.

[image]

bji219, 2021-11-06:
Here is the output of the journal command:
[raspi_error.txt](https://github.com/dtcooper/raspotify/files/7489923/raspi_error.txt)