dtcooper/raspotify #409: Raspotify not visible when my phone is the access point

Bram-diederik, 2021-07-20:
I have a pi zero w with a bluetooth speaker.
Works great. But only the pi is not visible if i use my phone as access point. (And i have no wifi connected on my phone need to test that out. Im reinstalling after i gave volumio one go)

I have premium and my device credentials set in the config.

Bram-diederik, 2021-07-24:
[image]
[image]

The chromecast works fine. Is connected to my phone hotspot. Raspotify only accessible when my phone is connected to a wifi network.

Ej747, 2021-08-04:
I have the same problem where I can only use it if my phone is connected to a wifi network

Bram-diederik, 2021-08-04:
Well after reboot the chromecast also looses connection. And is also only available when the wifi is up. Try to get this bug to spotify support but that is slow.

JasonLG1979, 2021-11-17:
duplicate of https://github.com/dtcooper/raspotify/issues/413

closed 2021-11-17 by JasonLG1979