dtcooper/raspotify #411: jackaudio

regie-techno, 2021-07-29:
I would like to transmit audio via jack.
So I tried to use alsajack plugin

My .asoundrc file loks like

`pcm.!default {
     type plug
     slave { pcm "jack" }
     }

pcm.jackplug {
     type plug
     slave { pcm "jack" }
     }

pcm.jack {
     type jack
     playback_ports {
          0 system:playback_1
          1 system:playback_2
          }
     capture_ports {
          0 system:capture_1
          1 system:capture_2
          }
     }`
It works well with : aplay -f cd media.wav

jack_lsp give
`system:capture_1
system:capture_2
system:capture_3
system:capture_4
system:capture_5
system:capture_6
system:capture_7
system:capture_8
system:playback_1
   alsa-jack.jackP.11798.0:out_000
system:playback_2
   alsa-jack.jackP.11798.0:out_001
system:playback_3
system:playback_4
system:playback_5
system:playback_6
system:playback_7
system:playback_8
alsa-jack.jackP.11798.0:out_000
   system:playback_1
alsa-jack.jackP.11798.0:out_001
   system:playback_2`

But when I use with raspotify, nothing works and jack_lsp gives me not connexion.

So I tried to use --backend=jackaudio (as explained in librespot github). 
This is what `systemctl status raspotify.service` gives :

`● raspotify.service - Raspotify
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2021-07-29 17:38:41 CEST; 1s ago
  Process: 11749 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 11750 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 11751 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --device-type ${DEVICE_TYPE} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS (code=exited, status=101)
 Main PID: 11751 (code=exited, status=101)

juil. 29 17:38:41 msTorpez librespot[11751]:   23:   0x905d30 - std::rt::lang_start_internal::hf9465186f7b70c75
juil. 29 17:38:41 msTorpez librespot[11751]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/rt.rs:51
juil. 29 17:38:41 msTorpez librespot[11751]:   24:   0x4df460 - main
juil. 29 17:38:41 msTorpez librespot[11751]:   25: 0xb6c6a718 - __libc_start_main
juil. 29 17:38:41 msTorpez systemd[1]: raspotify.service: Main process exited, code=exited, status=101/n/a
juil. 29 17:38:41 msTorpez systemd[1]: raspotify.service: Failed with result 'exit-code'.`

Can you give me any help ?

regie-techno, 2021-07-30:
Success !
I succeeded in making jack work !
But it does not work with raspotify as a service ....

closed 2021-07-30 by regie-techno

microfx, 2022-01-29:
So you're now not using alsa at all?

JasonLG1979, 2022-01-29:
> My .asoundrc file loks like

As in `~/.asoundrc`? If so that config only works for that particular user, That's why it doesn't work for the Raspotify service. The Raspotify service is a system level service not a user level service.

> So I tried to use --backend=jackaudio

The audio backends are optional. At compile time you can decide to compile in any, all or none (which will get you just the pipe backend). The version of `librespot` packaged in Raspotify only has the ALSA backend and the pipe backend (again because you always get at least the pipe backend).

> This is what `systemctl status raspotify.service` gives:

Judging from that output you are using an old version of Raspotify (and `librespot`).

If you want to continue to use Raspotify I would advise that you update your OS and Raspotify to the latest version otherwise you will get no support here.

JasonLG1979, 2022-01-29:
> So you're now not using alsa at all?

No. ALSA is at the bottom of the Linux audio stack. ALSA will always be there, it's just a matter of how directly you're interacting with it. Jack is just an abstraction layer on top of ALSA just like PulseAudio just build with different goals in mind. In the end they both have to talk to ALSA.