dtcooper/raspotify #413: Doesn't show pi in list of devices when I am using data

Ej747, 2021-08-03:
I wanted to be able to use raspotify for my car instead of having to use an aux cord.

I planned to connect the pi to my phone's hotspot and use my data plan for my phone (since I obviously won't be able to connect to wifi). The problem is that my pi doesn't show up in Spotify if I'm not connected to a wifi network.

I tried entering my username and password in the /etc/default/raspotify file but when I do that, I am completely unable to connect to it in any situation. If my phone is on wifi, with my username and password in the raspotify file, the pi will show up as a speaker in spotify but when I click on it to try and connect, it will disappear a few seconds after it says it is connecting. When my phone isn't on wifi, it just doesn't show up at all.

Is there something I'm misunderstanding or doing wrong?

PatMulligan, 2021-08-09:
Edit the file **`/lib/systemd/system/raspotify.service`** on your pi and in the field for **EnvironmentFile** (near the end of the [Service] block), remove the "-" before "/etc/default/raspotify". Seems to be a typo...

that line should look like this : `EnvironmentFile=/etc/default/raspotify`

then reload the daemon and restart Raspotify with:
`#sudo systemctl daemon-reload`
`#sudo service raspotify stop`
`#sudo service raspotify start`

Ej747, 2021-08-10:
This didn't seem to work. I made a spreadsheet of all the possible scenarios if that helps
[raspotify troubleshooting.xlsx](https://github.com/dtcooper/raspotify/files/6958263/raspotify.troubleshooting.xlsx)

PatMulligan, 2021-08-10:
yeah nevermind, @dtcooper corrected me, the "-" is supposed to be there.

> The argument passed should be an absolute filename or wildcard expression, optionally prefixed with "-", which indicates that if the file does not exist, it will not be read and no error or warning message is logged. This option may be specified more than once in which case all specified files are read. If the empty string is assigned to this option, the list of file to read is reset, all prior assignments have no effect.
-[systemd docs](https://www.freedesktop.org/software/systemd/man/systemd.exec.html#EnvironmentFile=)

PatMulligan, 2021-08-10:
I didn't do enough investigative work. I was misled, because when I restart the service "raspotify" shows up in my spotify device list (with the "-" as well as per above). Then after a while it seems to disappear.

JasonLG1979, 2021-11-17:
@Ej747 Your problem is because of Network Segregation. When your phone is in hotspot mode it's basically functioning as a wifi router/AP. For security purposes generally devices on that hotspot network will not be able to see the phone and vice versa. They may not even be able to see other devices connected to the hotspot for that reason. This is not a bug in Raspotify or librespot.

closed 2021-11-17 by JasonLG1979

Ej747, 2021-11-17:
Alright, that makes sense. Thanks for the help. Would you have any idea for a workaround or is my idea just not plausible?

JasonLG1979, 2021-11-17:
> @JasonLG1979 Alright, that makes sense. Thanks for the help. Would you have any idea for a workaround or is my idea just not plausible?

Short of seeing if devices on the same hotspot network can see each other and then using that other device to control Raspotify no not really.

Alternatively you could just get a Bluetooth receiver for your car or turn your Raspberry Pi Zero into a  Bluetooth receiver with these instructions I wrote a while ago:

https://github.com/JasonLG1979/PiZero-Bluetooth-Audio-Receiver

Ej747, 2021-11-17:
Alright. Thanks again for the help!!