dtcooper/raspotify #417: Raspotify not discoverable without credentials

mckennajones, 2021-08-09:
Thanks for a great project!

I've been using Raspotify for about a year now without issues. I had not added my username/password to the config file. For some reason, during the past couple days, my Raspotify instance no longer showed in the "connect" menu of Spotify both on my Android Phone and macOS Spotify apps. I tried some basic things like restarting/re-installing, but the same issue persists. If I do enter my username/password to the config file, I am able to connect as before. 

I'm happy to share any logs that would be useful to diagnosing the problem.

mckennajones, 2021-08-10:
Quick follow up here. About 24 hours after adding my credentials to the config file, I was then also unable to see raspotify in my connect menu... A quick restart solved the issue.

galehar, 2021-09-02:
I have the same issue. Using OSMC on a vero 4k, it used to work without credentials. Now, I have to add credentials, but it's only visible when connected to Spotify using the same account. I was hoping I could make it a shared speaker so that everyone can stream to it regardless of the Spotify account they are logged in with.

mckennajones, 2021-09-02:
I actually determined that the issue lied in a firmware update to my CenturyLink router. After the update, mDNS is completely broken across 2.4Ghz and 5Ghz signals. I previously had my router setup to broadcast the same SSID for 2.4Ghz and 5Ghz. Since the Pi is connected to 2.4, I couldn't discover it from devices connected to the 5Ghz signal. The "solution" was to broadcast separate SSID's, allowing me to discover the Pi if I'm connected to 2.4Ghz on my computer or phone.

closed 2021-09-02 by mckennajones