dtcooper/raspotify #419: Doesnt play whole playlists

Pukesmiley, 2021-08-19:
My standard playlist is 2500+ songs big, but when i play it on the Pi it only shuffles the first 80+- and beginns with the first song in the playlist.
Im using the 3B+

JasonLG1979, 2021-11-20:
The new release is live. Please update and see if that solves your issue.

maxmidnite, 2021-11-20:
Now I can't connect at all anymore. I don't have time to check what's wrong right now. I'll get to it later, worst case I'll reset the system and let you know.

JasonLG1979, 2021-11-24:
This particular problem is an upstream issue. If after updating to the latest version of Raspotify the issue persists please file a bug with `librespot`.

Otherwise if the reason you can't connect is something different please feel free to open a different issue here if it's related to packaging or configuration or `librespot` if it's related to how the `librespot` binary behaves.

closed 2021-11-24 by JasonLG1979