dtcooper/raspotify #420: Raspotify not visible on wired network without credentials

galehar, 2021-09-03:
I'm running raspotify on a vero 4k connected to the network with ethernet. I can only see the device if I enter credentials in the config. I'd like to be able to use it with multiple different Spotify accounts which doesn't work if I enter credentials.
Official spotify-connect documentation says that devices need to be on the same wifi network. Is it a requirement? Can't it work on an ethernet network? How does the discovery works? I can't find any doc about ports and protocols used. How do I diagnose the issue with tcpdump?
Alternatively, could I run multiple instances of Raspotify, each with different credentials? Is this supported?
Thanks.

galehar, 2021-09-03:
I noticed that when I reboot the vero, raspotify becomes visible (without credentials) for about 2~3 minutes, then, if I don't use it disappears. Restarting the service won't help, I have to reboot the computer and then it's visible again for a couple of minutes.
If I connect to it before it disappears, then it stays. So it's an acceptable workaround, I just have to remember to connect to it after any reboot. But there's something weird going on I think.

kraaijmakers, 2021-09-23:
If you are on the same network with your Spotify Client and the Spotify Connect running on Raspotify it should be able to find. For example an iPhone needs extra permission to find clients using the Network. Last week my cousin turned on my Raspberry Pi with Spotify Connect using his own credentials. Since this software is based on librespot, maybe the docs from them guide you in the right direction, here's the link: [wiki](https://github.com/librespot-org/librespot/wiki).

alex-go, 2021-09-30:
Had a similar issue, related to a VPN starting shortly after and routing all multicast traffic through the VPN. 
What worked for me was forcing all multicast traffic to go via eth0 instead of tun0: route add -net 224.0.0.0 netmask 240.0.0.0 eth0

you can use tcpdump to check for mdns packets being sent on your LAN interfaces (replace eth0 with your local interface): tcpdump -n "port mdns" -i eth0 -v

JasonLG1979, 2021-11-20:
The new release is live. Please update and see if that solves your issue.

JasonLG1979, 2021-12-03:
Closing this due to inactivity and the fact that the original poster has not responded to requests for information.

closed 2021-12-03 by JasonLG1979

VilterPD, 2024-07-01:
> Had a similar issue, related to a VPN starting shortly after and routing all multicast traffic through the VPN. What worked for me was forcing all multicast traffic to go via eth0 instead of tun0: [...]

This has worked for me, with wlan0, for the interface I use