dtcooper/raspotify #426: apt-get update warning on RPi4

jodo234, 2021-09-22:
Hi,

I just updated to an RPi4 from RPi3. I use raspotify for a long time now and saw that it is also supported for the "new" RPi4 as well. As this one now is arm64, after installing i get the following message each time I run sudo apt-get update:

N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://dtcooper.github.io/raspotify raspotify InRelease' doesn't support architecture 'arm64'

Does this mean I dont get any further updates now for the software? Currently, raspotify is working, but I am not sure what this message means or if I now will have problems for future updates.

kraaijmakers, 2021-09-23:
This is because Raspotify is compiled for the 32bit Raspberry Pi OS, not for the 64-bit version. You can silence the warning by editing the file /etc/apt/sources.list.d/raspotify.list
and enter after 'deb' [ARCH=armhf]. This makes it only look for the current armhf (32 bit Raspberry Pi OS) executables in the repository. But if and when an arm64 will be released you have to remove it.

jodo234, 2021-09-23:
Thx for your reply and the information. But if I do so, I will still get updates for the 32bit version right? :)

kraaijmakers, 2021-09-23:
Yes it just narrows the search of all packages down to just armhf

jodo234, 2021-09-23:
Ok manx thx for clarifying :)

closed 2021-09-23 by jodo234