dtcooper/raspotify #427: Wrong directory permission on cache by default

kraaijmakers, 2021-09-23:
Since the credentials will be saved in /var/cache/raspotify
The librespot team recommends setting the directory permission to 0700 but the systemd Unit file uses 0755.

Setting the right permission is as easy as:
`sudo chmod 700 /var/cache/raspotify/`

For now this seems to work for me fine without any other modifications (except for the RUST_LOG environment.

JasonLG1979, 2021-11-14:
If you want to get technical it should be `600` since `librespot` doesn't need to execute anything from the cache folder. All it needs is read/write access. Anything but `600` is technically a security flaw. No one should be allowed to execute anything in the cache folder and no one except the raspotify user should be allowed to read anything in the cache folder since it could contain cached credentials.

JasonLG1979, 2021-11-14:
If you want to get really crazy you can edit the raspotify systemd unit file with `sudo systemctl edit --full raspotify` and replace it's content with:

```
[Unit]
Description=Raspotify
After=network.target

[Service]
User=raspotify
Group=raspotify
Restart=always
RestartSec=10
PermissionsStartOnly=true
UMask=077
CapabilityBoundingSet=
RestrictAddressFamilies=AF_UNIX AF_INET AF_INET6 AF_NETLINK
RestrictNamespaces=true
NoNewPrivileges=true
ProtectHostname=true
ProtectControlGroups=true
ProtectKernelLogs=true
ProtectKernelModules=true
ProtectKernelTunables=true
ProtectProc=invisible
ProcSubset=pid
PrivateTmp=true
PrivateUsers=true
ProtectSystem=strict
ReadWritePaths=/var/cache/raspotify
ProtectHome=true
LockPersonality=true
MemoryDenyWriteExecute=true
RemoveIPC=true
SystemCallArchitectures=native
RestrictRealtime=true
RestrictSUIDSGID=true
SystemCallFilter=~@clock
SystemCallFilter=~@debug
SystemCallFilter=~@module
SystemCallFilter=~@mount
SystemCallFilter=~@raw-io
SystemCallFilter=~@reboot
SystemCallFilter=~@swap
SystemCallFilter=~@privileged
SystemCallFilter=~@resources
SystemCallFilter=~@cpu-emulation
SystemCallFilter=~@obsolete
ExecStartPre=/bin/mkdir -m 0600 -p /var/cache/raspotify ; /bin/chown raspotify:raspotify /var/cache/raspotify
Environment="DEVICE_NAME=raspotify (%H)"
Environment="BITRATE=160"
Environment="CACHE_ARGS=--disable-audio-cache"
Environment="VOLUME_ARGS=--enable-volume-normalisation --volume-ctrl linear --initial-volume 100"
Environment="BACKEND_ARGS=--backend alsa"
Environment="DEVICE_TYPE=speaker"
EnvironmentFile=-/etc/default/raspotify
ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --device-type ${DEVICE_TYPE} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS

[Install]
WantedBy=multi-user.target
```

That will security harden the raspotify service.

You can analyze the security of any service with `systemd-analyze security <service name>`.

Before hardening:
```
→ Overall exposure level for raspotify.service: 9.2 UNSAFE 😨
  ```
  
After hardening:
 ```
→ Overall exposure level for raspotify.service: 1.5 OK 🙂
 ```

JasonLG1979, 2021-11-17:
This will addressed in https://github.com/dtcooper/raspotify/pull/449

closed 2021-11-17 by JasonLG1979