dtcooper/raspotify #430: No Autoplay Similar Songs

fruitofgeorge, 2021-10-05:
I've selected the appropriate option in my settings but when playing to my raspotify device, it never autoplays similar songs when the queue ends.

fruitofgeorge, 2021-11-17:
So is anyone still maintaining this?

JasonLG1979, 2021-11-17:
> So is anyone still maintaining this?

The owner of the repo has been busy with real life lately so I've stepped up to help. I'm currently dealing with the backlog of issues and getting ready for a new release. I would say that you can either try the package I posted [here](https://github.com/dtcooper/raspotify/pull/449#issuecomment-970553764) or wait until I get the new release out the door.

JasonLG1979, 2021-11-20:
The new release is live. Please update and see if that solves your issue.

fruitofgeorge, 2021-11-20:
Thanks for working on it. Sadly the update didn't fix it. Still just stops at the end rather than autoplaying a radio.

JasonLG1979, 2021-11-20:
Strange. I very rarely have Raspotify stop. I can listen for hours and it never stops?

JasonLG1979, 2021-11-20:
What does the tail-end of `journalctl -u raspotify -b` have to say before it stops?

fruitofgeorge, 2021-11-20:
The spotify app on iphone only lets you play one album at a time so this is when it gets to the end of the album. The song Racing Stripes is the last song of the album and Get Up is the first. It just finishes playing the last song and then loads up the first of the album again but doesn't play it.

```
Nov 20 18:16:12 raspy librespot[453]: [2021-11-20T23:16:12Z INFO  librespot_playback::player] Loading <Racing Stripes> with Spotify URI <spotify:track:3mOlGrCrh3rJYbjmhfMTiQ>
Nov 20 18:16:12 raspy librespot[453]: [2021-11-20T23:16:12Z INFO  librespot_playback::player] <Racing Stripes> (246506 ms) loaded
Nov 20 18:20:18 raspy librespot[453]: [2021-11-20T23:20:18Z INFO  librespot_playback::player] Loading <Get Up> with Spotify URI <spotify:track:1RyoaXDDV5cPO5g0yWZi89>
Nov 20 18:20:19 raspy librespot[453]: [2021-11-20T23:20:19Z INFO  librespot_playback::player] <Get Up> (153840 ms) loaded
```
If I play the same album (or any album) but instead of selecting raspotify as the spotify connect speaker, I play it to my Denon receiver or my desktop computer, when it gets to the end of the album it continues to autoplay similar songs.

JasonLG1979, 2021-11-21:
And you have `--autoplay` in your config I assume?

Could you post it (removing your username and password if it's in there)?

fruitofgeorge, 2021-11-21:
That's a big assumption. First I've ever heard of an autoplay flag. I just edited the default config file that raspotify installs. No mention of autoplay in there. Seems like it should just adopt the settings of the device you're playing from. Added it to the options environment and it magically works now! Thanks.

closed 2021-11-21 by fruitofgeorge

JasonLG1979, 2021-11-21:
> That's a big assumption. First I've ever heard of an autoplay flag. I just edited the default config file that raspotify installs. No mention of autoplay in there. Seems like it should just adopt [...]

Apparently you have not "selected the appropriate option".

`--autoplay` is a client-side flag. The controlling device does not decide that. It's a pretty big assumption on your part maybe you should take a peak at the [librespot wiki](https://github.com/librespot-org/librespot/wiki/Options)? I mean there are like 75 links to and mentions of librespot in this repo,lol!!!

Sort of a RTFM moment on your part...

fruitofgeorge, 2021-11-21:
That goes against the whole concept of raspotify. To quote the readme "A Spotify Connect client for Raspberry Pi's that Just Works™". A default config file is provided so that I can set the options and it works. Now all of a sudden I have to go to a completely different project and examine the endless flag options to get it to work?? No thanks.
Sort of a douchebag moment on your part...

JasonLG1979, 2021-11-21:
librespot isn't a completely different. Raspotify ***is*** librespot. Again maybe read something, anything in the readme... 

And Raspotify does just work. That doesn't mean you shouldn't maybe learn about the available options. I'm sorry, you asked a question that you could have figured out yourself in a couple mins, and then you pop off like it's my fault that Spotify designed their API so that autoplay is not decided by the controlling app.

The info is out there with links plastered everywhere I'm not sure how much easier I can make it. I can't read and learn it for you. You may actually have to put forth some effort... It's amazing you managed to install it with that attitude...