dtcooper/raspotify #436: Question: Track Information?

PetersOtto, 2021-11-03:
Hello together!
I have just a question.

I use raspotify together with mpc/mpd (webradio) on a Pi Zero W as kitchen radio. I use a webserver to control mpc/mpd. I adjusted the webpage in that way, that the radiostation and track will shown when I use mpc/mpd (`mpc --format [%title%] | head -n 1`).

When I stop the radio and use raspotify, it would be great to have the same behavior.

Is there a way to get these information from raspotify by using a commandline command?

[image]

JasonLG1979, 2021-11-17:
Raspotify is just a packaged up version of librespot as such this feature request is out of scope. You can get a track ID from librespot via an onevent script but you'd have to roll your own control mechanism or use one of the dozen that are floating around github. Everything is documented by Spotify in their public API docs.

https://developer.spotify.com/documentation/web-api/

closed 2021-11-17 by JasonLG1979

PetersOtto, 2021-11-17:
It seems not so easy as I wished :)
Thank you!