dtcooper/raspotify #437: Connecting to McIntosh MA5300 via RPI4 but no sound

Gru79, 2021-11-04:
I followed different tutorials on this and cannot get the sound out of RPI4 and MA5300 on Spotify. That's with USB connected to MA5300 stereo amp via its DAC1. Any help would be appreciated.

pi@raspberrypi:~ $  aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7

# /etc/default/raspotify -- Arguments/configuration for librespot

# Device name on Spotify Connect
DEVICE_NAME="McIntosh MA5300"

# The displayed device type in Spotify clients.
# Can be "unknown", "computer", "tablet", "smartphone", "speaker", "tv",
# "avr" (Audio/Video Receiver), "stb" (Set-Top Box), and "audiodongle".
#DEVICE_TYPE="speaker"

# Bitrate, one of 96 (low quality), 160 (default quality), or 320 (high quality)
#BITRATE="160"

# Additional command line arguments for librespot can be set below.
# See `librespot -h` for more info. Make sure whatever arguments you specify
# aren't already covered by other variables in this file. (See the daemon's
# config at `/lib/systemd/system/raspotify.service` for more technical details.)
#
# To make your device visible on Spotify Connect across the Internet add your
# username and password which can be set via "Set device password", on your
# account settings, use `--username` and `--password`.
#
# To choose a different output device (ie a USB audio dongle or HDMI audio out),
# use `--device` with something like `--device hw:0,1`. Your mileage may vary.
#
OPTIONS="--device usbstream:0,0 --username dstyga79 --password aQ=~s%-98hqj]7J#cuvy"

# Uncomment to use a cache for downloaded audio files. Cache is disabled by
# default. It's best to leave this as-is if you want to use it, since
# permissions are properly set on the directory `/var/cache/raspotify'.
#CACHE_ARGS="--cache /var/cache/raspotify"

# By default, the volume normalization is enabled, add alternative volume
# arguments here if you'd like, but these should be fine.
#VOLUME_ARGS="--enable-volume-normalisation --volume-ctrl linear --initial-volume=100"

# Backend could be set to pipe here, but it's for very advanced use cases of
# librespot, so you shouldn't need to change this under normal circumstances.
#BACKEND_ARGS="--backend alsa"

handcc, 2021-11-05:
You should remove your username and password from the post. It shouldn't be necessary anyway. 

play -l and -L doesn't appear to recognize your DAC. Is it turned on? You should see something like this: 

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 1: DAC1 [DAC-1], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

Gru79, 2021-11-05:
Ahh. good catch on the password. THANKS!

MA5300: It is turned on and connected with USB (the same USB cable I used with Nuc ROCK Roon before and it worked).

Gru79, 2021-11-05:
Just to be clear, I am not using ANY DAC on the PI itself. Trying to use my Mcintosh DAC. Am I getting this right?

handcc, 2021-11-05:
Yep, the output I posted above is using USB output to an external DAC. I'm using an XMC-1, here's my play -L

pi@raspberrypi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
sysdefault:CARD=XMC1
    XMC-1, USB Audio
    Default Audio Device
front:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    Front speakers
surround21:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    Direct sample mixing device
dsnoop:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    Direct sample snooping device
hw:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    Direct hardware device without any conversions
plughw:CARD=XMC1,DEV=0
    XMC-1, USB Audio
    Hardware device with all software conversions
usbstream:CARD=XMC1
    XMC-1
    USB Stream Output

Gru79, 2021-11-05:
So ... what am I doing wrong? I just replaced the cable and same thing. It appears the PI4 does not see my external Mc DAC1.

Gru79, 2021-11-05:
Ok. I just updated using sudo apt update and magically McIntosh HD USB Audio device showed up. ;)

Now, how do I configure it in the config files?

Here is the output now:

pi@raspberrypi:~ $ aplay -L
default
    Playback/recording through the PulseAudio sound server
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
sysdefault:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
dmix:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample mixing device
dsnoop:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample snooping device
hw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct hardware device without any conversions
plughw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Hardware device with all software conversions
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
sysdefault:CARD=Audio
    McIntosh HD USB Audio, USB Audio
    Default Audio Device
front:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    Front speakers
surround21:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    Direct sample mixing device
dsnoop:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    Direct sample snooping device
hw:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Audio,DEV=0
    McIntosh HD USB Audio, USB Audio
    Hardware device with all software conversions
usbstream:CARD=Audio
    McIntosh HD USB Audio
    USB Stream Output
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: Audio [McIntosh HD USB Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@raspberrypi:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2afd:0005
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $

Gru79, 2021-11-05:
I commented out the config file as I now see I did not need to do that. Still no sound.

Gru79, 2021-11-05:
Decided to start from scratch knowing what I know now. No sound.

Gru79, 2021-11-10:
Can anyone can assist here?

Gru79, 2021-11-11:
@dtcooper Hi there. Can you please assist me on this? Thanks.

JasonLG1979, 2021-11-17:
What you want is:

`--device hw:CARD=Audio,DEV=0`

if that doesn't work you can route it though a plug with:

`--device plughw:CARD=Audio,DEV=0`

Gru79, 2021-11-18:
@JasonLG1979 THAT WORKED! Thank you so much! I had to use the mixer to raise the volume to 100%. Now all is working!

closed 2021-11-18 by JasonLG1979