dtcooper/raspotify #438: Random changes in sound as music plays

franvillamil, 2021-11-05:
I have a Raspberry Pi connected to an analog 5.1 system through a 3.5 jack. When music plays through raspotify, the volume goes up and down without apparent reason. I have tried to change the sound normalization setting, but it doesn't seem to work either way. Does anyone know which setting should change, or why is this happening?

Output from `sudo service raspotify status` when loading raspotify:

```
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-10-30 11:30:58 CEST; 6 days ago
  Process: 17440 ExecStartPre=/bin/mkdir -m 0755 -p /var/cache/raspotify (code=exited, status=0/SUCCESS)
  Process: 17441 ExecStartPre=/bin/chown raspotify:raspotify /var/cache/raspotify (code=exited, status=0/SUCCESS)
 Main PID: 17442 (librespot)
    Tasks: 3 (limit: 2062)
   CGroup: /system.slice/raspotify.service
           └─17442 /usr/bin/librespot --name raspotify (raspberrypi) --device-type speaker --backend alsa --bitrate 160 --disable-audio-cache --volume-ctrl linear --initial-volume=100
```

JasonLG1979, 2021-11-06:
Change the volume to `fixed` and see if that has any effect.

franvillamil, 2021-11-06:
That would be `--fixed-volume` in `VOLUME_ARGS`?

JasonLG1979, 2021-11-06:
No. `--volume-ctrl fixed` in `VOLUME_ARGS`.

closed 2021-11-24 by JasonLG1979