dtcooper/raspotify #442: Unable to get `mixer=alsa` working

agneevX, 2021-11-09:
Hello!

I cannot seem to get `--mixer=alsa` working. 

I'm using the `derkades/raspotify` with a modified entrypoint script. This does works without this parameter.

```py
$ /usr/bin/librespot --name=Speakers --mixer=alsa --backend=alsa --enable-volume-normalisation --normalisation-pregain=3 --disable-discovery --disable-audio-cache --bitrate=320 --initial-volume=70 --username=xxxxx --password=xxxxx
[2021-11-09T10:44:44Z INFO  librespot] librespot (raspotify v0.16.0) 0adb851 (2020-12-15). Built on 2020-12-30. Build ID: KLdkAvgt
[2021-11-09T10:44:44Z INFO  librespot_core::session] Connecting to AP "gae2-accesspoint-a-v84t.ap.spotify.com:4070"
[2021-11-09T10:44:45Z INFO  librespot_core::session] Authenticated as "xxxxx" !
[2021-11-09T10:44:45Z INFO  librespot_playback::mixer::alsamixer] Setting up new mixer: card:default mixer:PCM index:0
thread 'main' panicked at 'Couldn't find simple mixer control for PCM,0', playback/src/mixer/alsamixer.rs:39:44
```

Trace here: https://paste.ubuntu.com/p/pNYKMD4gd8/

```
root@6bf06b9dadaf:/# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
plugequal
equal
hw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct hardware device without any conversions
plughw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Hardware device with all software conversions
sysdefault:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
dmix:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample mixing device

root@6bf06b9dadaf:/# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
```

JasonLG1979, 2021-11-12:
https://github.com/Derkades/raspotify-docker is fork of a fork of a fork. It get's pretty muddy. You should file a bug report there and walk back though the forks. Follow the chain of command if you will. But really it will probably be all for naught as all of those projects are using an extremely old version of librespot. The mixer and audio backends have been basically completely rewritten in the meantime since that version was released so this bug is more than likely invalid anymore.

agneevX, 2021-11-12:
I'm not sure that make a difference. The Dockerfile references with the latest version of raspotify.

```Dockerfile
RUN apt-get update && \
    apt-get -y install alsa-utils libasound2-plugin-equal gettext curl apt-transport-https && \
    update-ca-certificates --fresh && \
    curl -sSL https://dtcooper.github.io/raspotify/key.asc | sudo apt-key add -v - && \
    echo 'deb https://dtcooper.github.io/raspotify raspotify main' | sudo tee /etc/apt/sources.list.d/raspotify.list && \
    sudo apt-get update && \
    sudo apt-get -y install raspotify && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*
```

JasonLG1979, 2021-11-12:
Which is not the latest version of librespot...

JasonLG1979, 2021-11-12:
And it's not running on Raspberry Pi OS.

JasonLG1979, 2021-11-12:
You're bug is not valid here because you're not running Raspotify on it's intended target. And not valid with librespot because it's an out of date version. You need to start with filing a bug with https://github.com/Derkades/raspotify-docker to rule out docker related issues and then if it's found that it's not a docker related issue then you file a bug here and then if it's found to not be a bug in Raspotify then file a bug with librespot. This is more than likely a configuration issue. As the alsa mixer option should work since the build in audio of a Raspberry Pi does have hardware volume control.

agneevX, 2021-11-12:
How are you sure that librespot is out of date with what raspotify is shipping? Per the README, the build instructions seems to be the same.

JasonLG1979, 2021-11-12:
The version of librespot that Raspotify and the docker image you're using is way out of date.

JasonLG1979, 2021-11-12:
Raspotify is at something like v0.16.0 I think (?) that was built in Dec 2020. Librespot is at verison v0.3.1 that was released about 3 weeks ago.

agneevX, 2021-11-12:
> The version of librespot in Raspotify and the docker image you're using is way out of date.

If you could please help me make sense of this. The raspotify script uses and bundles a custom compiled librespot image IIRC?

The docker image was built on the Oct 31 2021, so it should be running the latest version given v0.16.0 of raspotify was released on Dec 31 2020.

I've tried librespot multi-arch docker images, but they offer `rodio` output.

JasonLG1979, 2021-11-12:
The current version of Raspotify != the latest version of librespot. Raspotify is just a build of librespot made into a systemd service. The binary is librespot. Librespot in Raspotify is out of date so anything that uses Raspotify is also out of date.

JasonLG1979, 2021-11-12:
> I've tried librespot multi-arch docker images, but they offer `rodio` output.

Then change them to build with the alsa backend.

agneevX, 2021-11-16:
Thanks, that's what [I did](https://hub.docker.com/r/agneev/librespot-alsa). My image works fine, as long as I use `softvol`.

JasonLG1979, 2021-11-16:
Awesome. I will mention that to others when they ask about Raspotify/librespot docker.

JasonLG1979, 2021-11-16:
> My image works fine, as long as I use softvol.

Volume in alsa on most devices is done with integer division. Softare volume in librespot is done in 64bit float. Unless you absolutely need some feature to do with hardware volume control in alsa, software volume control is much, much better.

JasonLG1979, 2021-11-16:
Hardware volume control in alsa is also buggy as all Hell. Like as in the code is crusty and crufty as Hell but can't be changed because it would break things that depend on it.

agneevX, 2021-11-17:
Ok, thanks for your help 👍

closed 2021-11-17 by agneevX