dtcooper/raspotify #443: Raspotify won`t start playing

aleksa3123, 2021-11-10:
It start loading song and crash
I got this error when check status:

```
Nov 10 15:23:44 raspberrypi librespot[3396]:   17:   0x553304 - <librespot_playback::player::PlayerInternal as futures::future::Future>::poll::he6e04a5bccf94e34
Nov 10 15:23:44 raspberrypi librespot[3396]:   18:   0x4b7300 - futures::task_impl::std::set::hab1c1811473e3998
Nov 10 15:23:44 raspberrypi librespot[3396]:   19:   0x4ad838 - std::sys_common::backtrace::__rust_begin_short_backtrace::h8286c5a0fd04a387
Nov 10 15:23:44 raspberrypi librespot[3396]:   20:   0x4c84f4 - core::ops::function::FnOnce::call_once{{vtable.shim}}::h76e6c7604d7daced
Nov 10 15:23:44 raspberrypi librespot[3396]:   21:   0x8f4940 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::h6b7ad05f87ff90ae
Nov 10 15:23:44 raspberrypi librespot[3396]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/alloc/src/boxed.rs:1042
Nov 10 15:23:44 raspberrypi librespot[3396]:   22:   0x8f4940 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::he9130e675b8d97ee
Nov 10 15:23:44 raspberrypi librespot[3396]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/alloc/src/boxed.rs:1042
Nov 10 15:23:44 raspberrypi librespot[3396]:   23:   0x8f4940 - std::sys::unix::thread::Thread::new::thread_start::h7a3f6590d8e03e95
Nov 10 15:23:44 raspberrypi librespot[3396]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/sys/unix/thread.rs:87
```

dtcooper, 2021-11-10:
Any other reporters here? Could be related to librespot being out of date. Fixing #300 will fix this in that case. Cheers.

aleksa3123, 2021-11-11:
I getting this shit now:
```
Nov 11 17:27:28 raspberrypi librespot[3091]:   36:   0x935d30 - std::panic::catch_unwind::hb91533e13ff48846
Nov 11 17:27:28 raspberrypi librespot[3091]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/panic.rs:382
Nov 11 17:27:28 raspberrypi librespot[3091]:   37:   0x935d30 - std::rt::lang_start_internal::hf9465186f7b70c75
Nov 11 17:27:28 raspberrypi librespot[3091]:                        at /rustc/7eac88abb2e57e752f3302f02be5f3ce3d7adfb4/library/std/src/rt.rs:51
Nov 11 17:27:28 raspberrypi librespot[3091]:   38:   0x50f460 - main
Nov 11 17:27:28 raspberrypi librespot[3091]:   39: 0x76c64be0 - __libc_start_main
Nov 11 17:27:28 raspberrypi librespot[3091]: [2021-11-11T17:27:28Z ERROR librespot_playback::player] Player thread panicked!
Nov 11 17:27:28 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=101/n/a
Nov 11 17:27:28 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Nov 11 17:27:28 raspberrypi systemd[1]: raspotify.service: Consumed 1.125s CPU time.
```

aleksa3123, 2021-11-11:
@dtcooper yo man i reinstall os and install raspotify again and it work, but when i install drivers for my HifiBerry digi it imidietly stop working and displaying this error.

JasonLG1979, 2021-11-17:
I put a lot of work into actually presenting the user useful error messages as apposed to just puking a stack trace upstream in librespot. I'm gearing up for a new release of Raspotify which will include A LOT of fixes, new features and my improvements. Hopefully when that hits it will solve your issue or at least make debugging it easier.

JasonLG1979, 2021-11-20:
The new release is live. Please update and see if that solves your issue.

Grovespaz, 2021-11-20:
I had the same issue and updating solved it for me!

JasonLG1979, 2021-11-20:
> I had the same issue and updating solved it for me!

There's no way to know if your issue was the same issue. The service status above contains an incomplete stack trace so there's no way to really know what went wrong from that output.

closed 2021-11-27 by github-actions[bot]