dtcooper/raspotify #446: Raspotify working with new release of Raspberry OS?

DaMuhr, 2021-11-11:
Hey there,
was just wondering if the Raspotify service already works for the new version of raspberry OS, Bullseye. I upgraded today and raspotify isn't working now. Any help would be appreciated.

aleksa3123, 2021-11-12:
I have sam issue, but I was thinking that my Hifiberry digi driver is bugging the shit out of it. You can see my status output

JasonLG1979, 2021-11-12:
LIbrespot works great on Raspberry Pi OS Bullseye if you compile it yourself but Raspotify maybe not so much judging from bug reports.

@dtcooper is looking for someone to [take over maintanership](https://github.com/dtcooper/raspotify/issues/300#issuecomment-966696609) see the top of the README.md:

> warning NOTE: I have limited to no time to work on this package these days!

> I am looking for a maintainer to help with issues and updating the package. Please reach out via email at david@jew.pizza if > you have interest in helping.

> Cheers,

> David

I'm no docker expert by any means but I would say that the docker file could be updated to Bullseye (it's currently stretch) and some deprecated option names need to be dealt with and it should be good.

In the mean time it's really easy to build librespot yourself on your raspberry pi and just replace the librespot binary that ships with raspotify.

JasonLG1979, 2021-11-16:
Update: I have a PR in that updates raspotify and I have volunteered to help @dtcooper maintain raspotify. Just waiting for him to get back to me.

dtcooper, 2021-11-16:
@JasonLG1979 I've added you as a maintainer.

dtcooper, 2021-11-16:
> I'm no docker expert by any means but I would say that the docker file could be updated to Bullseye (it's currently stretch) and some deprecated option names need to be dealt with and it should be good.

Me neither! :)

JasonLG1979, 2021-11-17:
A new release is on the way. I'm going to close this.

closed 2021-11-17 by JasonLG1979