dtcooper/raspotify #451: Security hardening causes librespot to not start

jan-be, 2021-11-17:
Hi,
I tried the new version on my Pi 4 running Ubuntu and the changed systemd service file causes librespot to not start.
a844a33428018cae34e4c9642aaaca5964ae8502 appears to be the issue. Reverting to the old version fixes the issue.

Systemctl log:
```
Nov 17 22:56:03 jans-rpi systemd[1]: raspotify.service: Scheduled restart job, restart counter is at 11.
Nov 17 22:56:03 jans-rpi systemd[1]: Stopped Raspotify (Spotify Connect Client).
Nov 17 22:56:03 jans-rpi systemd[1]: Started Raspotify (Spotify Connect Client).
Nov 17 22:56:03 jans-rpi systemd[1]: raspotify.service: Main process exited, code=killed, status=31/SYS
Nov 17 22:56:03 jans-rpi systemd[1]: raspotify.service: Failed with result 'signal'.
Nov 17 22:56:13 jans-rpi systemd[1]: raspotify.service: Scheduled restart job, restart counter is at 12.
Nov 17 22:56:13 jans-rpi systemd[1]: Stopped Raspotify (Spotify Connect Client).
Nov 17 22:56:13 jans-rpi systemd[1]: Started Raspotify (Spotify Connect Client).
Nov 17 22:56:14 jans-rpi systemd[1]: raspotify.service: Main process exited, code=killed, status=31/SYS
Nov 17 22:56:14 jans-rpi systemd[1]: raspotify.service: Failed with result 'signal'.
```

running `journalctl _AUDIT_TYPE_NAME=SECCOMP` repeatedly shows messages like

```
Nov 17 22:56:24 jans-rpi audit[2564]: SECCOMP auid=4294967295 uid=114 gid=122 ses=4294967295 subj=unconfined pid=2564 comm="librespot" exe="/usr/bin/librespot" sig=31 arch=40000028 syscall=45 compat=1 ip=0xf7b32d56 code=0x0
```

syscall 45 is apparently `brk` on ARM, but I'm not sure about that.

JasonLG1979, 2021-11-18:
What OS are you running? And What device? I have tested this release on a Raspberry Pi Zero and a Raspberry Pi 4 both on Raspberry Pi OS (not the Aarch64 version).

JasonLG1979, 2021-11-18:
The solution is not removing the line but doing proper native builds.

jan-be, 2021-11-18:
Oh yeah that makes sense, that's correct, I'm running arm64.
I agree, native builds would be even better.

JasonLG1979, 2021-11-18:
> I agree, native builds would be even better.

Librespot builds and runs just fine on Aarch64 I know a Aarch64 build of Raspotify has been on people's wish list for a long time. I'll see what I can do.

closed 2021-11-18 by JasonLG1979

JasonLG1979, 2021-11-18:
Good catch @jan-be Thanks!!! I'm rebuilding as I speak.