dtcooper/raspotify #456: [0.31.1] raspotify.service: Main process exited, code=exited, status=127/n/a

kennymc-c, 2021-11-20:
I just updated to 0.31.1 but unfortunately raspotify does not start via systemd with the following errors::

`Nov 20 13:45:12 Raspi-Kueche systemd[1]: raspotify.service: Main process exited, code=exited, status=127/n/a
Nov 20 13:45:12 Raspi-Kueche systemd[1]: raspotify.service: Unit entered failed state.
Nov 20 13:45:12 Raspi-Kueche systemd[1]: raspotify.service: Failed with result 'exit-code'.
Nov 20 13:45:18 Raspi-Kueche systemd[1]: Stopped Raspotify (Spotify Connect Client).
Nov 20 13:46:44 Raspi-Kueche systemd[1]: [/lib/systemd/system/raspotify.service:20] Unknown lvalue 'RestrictSUIDSGID' in section 'Service'
Nov 20 13:46:44 Raspi-Kueche systemd[1]: [/lib/systemd/system/raspotify.service:24] Unknown lvalue 'ProtectHostname' in section 'Service'
Nov 20 13:46:44 Raspi-Kueche systemd[1]: [/lib/systemd/system/raspotify.service:26] Unknown lvalue 'ProtectKernelLogs' in section 'Service'
Nov 20 13:46:44 Raspi-Kueche systemd[1]: [/lib/systemd/system/raspotify.service:29] Unknown lvalue 'ProtectProc' in section 'Service'
Nov 20 13:46:44 Raspi-Kueche systemd[1]: [/lib/systemd/system/raspotify.service:32] Unknown lvalue 'ProtectClock' in section 'Service'
Nov 20 13:46:44 Raspi-Kueche systemd[1]: [/lib/systemd/system/raspotify.service:42] Unknown lvalue 'CacheDirectory' in section 'Service'
Nov 20 13:46:44 Raspi-Kueche systemd[1]: [/lib/systemd/system/raspotify.service:48] Unknown lvalue 'ProcSubset' in section 'Service'
Nov 20 13:46:44 Raspi-Kueche systemd[1]: [/lib/systemd/system/raspotify.service:51] Unknown lvalue 'LockPersonality' in section 'Service'
Nov 20 13:46:44 Raspi-Kueche systemd[1]: [/lib/systemd/system/raspotify.service:58] Don't know system call group, ignoring: @system-service
Nov 20 13:46:44 Raspi-Kueche systemd[1]: [/lib/systemd/system/raspotify.service:69] Unknown lvalue 'ConfigurationDirectory' in section 'Service'`

karlossus, 2021-11-20:
Same error code on same version.

JasonLG1979, 2021-11-20:
What OS and what architecture?

kennymc-c, 2021-11-20:
armv6l (Pi Zero W) with Raspbian Stretch

JasonLG1979, 2021-11-20:
Raspbian Stretch is 2 versions old, pretty crusty by now, definitely not supported by Raspotify. You'll need to upgrade to Raspberry Pi OS (Bullseye).

If you don't want to do that as a totally unsupported workaround you can edit the service unit file and uncomment things until it works.

`sudo systemctl edit --full raspotify`

closed 2021-11-20 by JasonLG1979

kennymc-c, 2021-11-20:
@JasonLG1979 Strange, since librespot is running fine when I start it manually. Probably unsupported systemd parameters?
You automatically get this update with stretch when running apt update/upgrade. Could this be somehow prevented? I could imagine there are some users that still run pi's with older raspbian versions and only install security updates since they are just using it for audio-streaming.
If this is a stretch compatibly issue could stretch be marked as not compatible with stretch in the ReadMe, release notes?

JasonLG1979, 2021-11-20:
> You automatically get this update with stretch when running apt update/upgrade. Could this be somehow prevented?

Sure. Keep your Pi up to date,lol!!!

> I could imagine there are some users that still run pi's with older raspbian versions and only install security updates since they are just using it for audio-streaming.

If you never plan on upgrading your system you shouldn't install Raspotify via the repo. You should install a specific [release](https://github.com/dtcooper/raspotify/releases). The repo isn't OS version specific. (and hasn't been for 3 years) You're going to get basically rolling releases that assume the latest stable version of Raspberry Pi OS. The only reason it's worked this long for you is because the build tooling and package configs were ancient. Eventually they have to be upgraded and support for old versions dropped.

If this is a stretch compatibly issue could this release be marked as not compatible with stretch in the ReadMe, release notes?

Done.

MrEasy, 2021-11-28:
Had the same issue with 0.31.1.
Workaround if you want to keep the new version:
Replace the service file with the one from 0.16.0 and it worked again.