dtcooper/raspotify #457: [0.3.1] librespot Audio Sink Error Connection Refused after update

kirenida, 2021-11-20:
On raspberry pi 3B+ with hifiberry Dac+ and running raspberry pi os buster.

I am running pulseaudio as a system instance (yes, I know it's not recommended), so I can send audio from my laptop to the raspi, which is connected to my amplifier. I chose this solution as no user is constantly logged in to the raspi, and other people also use it as a Spotify Connect speaker.

After updating to the new version, I get the following error:

```
Nov 20 13:57:38 raspberrypi librespot[3670]: socket(): Address family not supported by protocol
Nov 20 13:57:38 raspberrypi librespot[3670]: [2021-11-20T12:57:38Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Nov 20 13:57:38 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Nov 20 13:57:38 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
```

After purging the new version and manually installing the previous package, everything works with the default configuration, without any `--backend` or `--device` options

Running `librespot -n raspi` on its own gets me the same error.

I have seen [this issue ](https://github.com/librespot-org/librespot/issues/877), but can't figure out what I could change in my pulseaudio configuration for it to work. I tried commenting out the `DevicePolicy` lines in the raspotify.service file, but that didn't help.

Any chance of a suggestion or fix for this, as I don't think that this is a very niche use case?

Should I just switch to spotifyd, as mentioned in the README?

JasonLG1979, 2021-11-20:
> I chose this solution as no user is constantly logged in to the raspi, and other people also use it as a Spotify Connect speaker.

A headless install is what you want in that case you don't want a system with PulseAudio, especially if Raspotify is the only thing making noise on the Pi.

Raspotify running on a Raspberry Pi OS lite (headless) install does not require a user be logged in for it to work. It works as a system level service that starts at boot.

kirenida, 2021-11-20:
> A headless install is what you want in that case you don't want a system with PulseAudio, especially if Raspotify is the only thing making noise on the Pi.
>
> Raspotify running on a Raspberry Pi OS [...]

I am already running a headless raspberry OS, and I **need** Pulseaudio because I send audio over the network from my (Linux) laptop to the Pulseaudio server running on the Raspberry.

JasonLG1979, 2021-11-20:
Everything worked fine prior to the latest update, and works fine if I install the previous package, even with the default settings.

What OS version? The new version is only tested to work with the current stable version of Raspberry Pi OS (Bullseye).

> I didn't even do much hacking to get pulseaudio to work, I just set up the systemd service to start on boot and enabled the zeroconf and tcp modules.

That's a pretty unique use case, and not really supported, your mileage may vary. If all else is configured correctly `librespot` should pickup PulseAudio as your default output (as I'm guessing it was doing). My guess is it's time for you to upgrade your Pi to the latest version of Raspberry Pi OS.

kirenida, 2021-11-20:
> My guess is it's time for you to upgrade your Pi to the latest version of Raspberry Pi OS.

I was just waiting for some time to pass and any potential bugs to be squashed.

I'll try a fresh install tommorow and report back. Anything special I should look out for? I plan on just doing the same thing as before - set up pulseaudio as a system instance and install raspotify from this repo.

JasonLG1979, 2021-11-20:
> I was just waiting for some time to pass and any potential bugs to be squashed.

Yep. It is what it is. With stuff like this no one actually pays attention to requests to test so you just gotta scream test.

> I'll try a fresh install tommorow and report back. Anything special I should look out for? I plan on just doing the same thing as > before - set up pulseaudio as a system instance and install raspotify from this repo.

> Just out of curiosity, why is this a unique case? Just because of the pulseaudio running as a system instance?

Well one using PulseAudio for streaming is a pretty advanced use case. And Raspotify isn't really meant to run on systems with PulseAudio installed. It only has the alsa backend compiled into it.

JasonLG1979, 2021-11-20:
It's not a good idea to run PulseAudio as a system service (and is in fact frowned upon by even them) that's generally considered the wrong thing to do and in my experience network streaming with PulseAudio was kinda janky last I tried it. You might checkout [snapcast](https://github.com/badaix/snapcast). You can then edit `/etc/asound.conf` as described in  https://github.com/dtcooper/raspotify/wiki/Troubleshooting to get software mixing capabilities similar to PulseAudio. So then you would run snapcast along side librespot and they could coexist happily on the same system.

kirenida, 2021-11-20:
Sure, I know that they don't recommend running it system-wide, but as they state in the [official documentation](https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/WhatIsWrongWithSystemWide/): `System mode is around for usage on thin client or embedded setups, where no real local user exists, where access is exclusively via the network, and where state data is flushed on each session termination.`, so I think my case is OK. 

I basically just want to send all/any audio from my computer (a laptop with shitty speakers) to my Hifiberry-equipped raspi connected to my amp+speakers. And I want my wife to be able to play Spotify on that same raspberry.

Up to now it has worked flawlessly for me and it's pretty simple to configure, especially after you figure out that you just need to enable the zeroconf and network transport pulseaudio modules. Then I can just choose it as the output device from my taskbar mixer applet.

I've had a look at snapcast, but it seems a little overkill for my use case. PulseAudio + Raspotify **_just worked™_** up to now.

But now we are getting way off-topic, so I'll try Raspberry OS bullseye tommorow and report back.

JasonLG1979, 2021-11-20:
> But now we are getting way off-topic, so I'll try Raspberry OS bullseye tommorow and report back.

You might be better served by either spotifyd or configuring pulse so that it's just another thing accessing dmix instead of it completely taking over your audio devices.

You can always just use the old systemd unit file if that's the issue?

I've also seriously considered compiling in the PulseAudio backend into Raspotify and offering an alternative systemd unit file so users could pick depending on their use case. It could be in the same package with the same binary (both backends can be compiled in and chosen at runtime) It would just be a different service name something like `raspotify-pulse`? I think I can make them mutually exclusive so that enabling one would disable the other. They could use the same cache folder but could have different config files. Just an idea.

kirenida, 2021-11-20:
> You can always just use the old systemd unit file if that's the issue?

I will try this on the new installation if defaults don't work.

> I've also seriously considered compiling in the PulseAudio backend into Raspotify and offering an alternative systemd unit file so users could pick depending on their use case. It could be in the [...]

Maybe it would be enough to just figure out what needs to be changed in the systemd unit file and add that in the wiki/Troubleshooting page? It seems that not many people have this configuration or problems with pulseaudio, so it would be less work for you?

Anyway, as long as I can just manually install the previous .deb release, I would be glad to help test any changes in this direction, up to my abilities, as I don't have much experience at this level.

JasonLG1979, 2021-11-20:
> Maybe it would be enough to just figure out what needs to be changed in the systemd unit file and add that in the wiki/Troubleshooting page? It seems that not many people have this configuration or problems with pulseaudio, so it would be less work for you?

Well if it in fact just a matter of just changing the unit file I will include a drop in override file in the repo. Drop ins are nice because in the event of a package up date they don't get overwritten. They're basically overlays.

JasonLG1979, 2021-11-20:
You might look into this:

https://wiki.archlinux.org/title/PulseAudio/Examples#PulseAudio_as_a_minimal_unintrusive_dumb_pipe_to_ALSA

kirenida, 2021-11-21:
I think I fixed it:

In the raspotify.service file, I added `AF_UNIX` to the `RestrictAddressFamilies=AF_INET AF_INET6 AF_NETLINK` option, so now it works without any other changes.

According to [this](https://gist.github.com/ageis/f5595e59b1cddb1513d1b425a323db04), AF_UNIX should be included for local communication.

All of this was done on my current installation (Buster), with pulseaudio running as a system instance.

Regarding your last suggestion to run Pulseaudio as a pipe to ALSA, I don't think it would work in my case, as I need the server running and advertising on the local network so my laptop can see it available as a sink.

JasonLG1979, 2021-11-21:
> According to [this](https://gist.github.com/ageis/f5595e59b1cddb1513d1b425a323db04), AF_UNIX should be included for local communication.

I will add that for the sake of backwards compatibility. It was not included originally because it's not necessary without PulseAudio.

> Regarding your last suggestion to run Pulseaudio as a pipe to ALSA, I don't think it would work in my case, as I need the server running and advertising on the local network so my laptop can see it available as a sink.

Yep I wasn't sure what the issue was. If adding `AF_UNIX` solves your problem then yeah not really a point to that,lol!!!

Sorry about the breakage and thanks for helping debug it. I'll add `AF_UNIX` and keep this open until a new version is released.

JasonLG1979, 2021-11-21:
https://github.com/dtcooper/raspotify/commit/f7ff40becfd608e5f4912e43dfbbc78425e0d92d

JasonLG1979, 2021-11-23:
@kirenida I forgot to ask, can you possibly document your setup in the wiki for other users?

kirenida, 2021-11-23:
> @kirenida I forgot to ask, can you possibly document your setup in the wiki for other users?

I can give it a go when I do a clean install of bullseye. What should I include?

JasonLG1979, 2021-11-25:
Can you test against https://github.com/dtcooper/raspotify/releases/tag/0.31.2-beta-1

kirenida, 2021-11-25:
> Can you test against https://github.com/dtcooper/raspotify/releases/tag/0.31.2-beta-1

purged installed release, downloaded & installed the beta, playback works OK with no extra configuration.

JasonLG1979, 2021-11-25:
Good deal!!!

JasonLG1979, 2021-11-25:
I know this sounds weird but could you purge again and then reinstall. I also fixed a bug where purging the package didn't actually purge everything. It left the the old systemd service file and then when you install after a purge the old unit file still exists and so isn't over written.

JasonLG1979, 2021-11-25:
To me purge means remove EVERYTHING not so much I guess to the deb-helper scripts...

kirenida, 2021-11-25:
> I know this sounds weird but could you purge again and then reinstall. I also fixed a bug where purging the package didn't actually purge everything. It left the the old systemd service file and then when you install after a purge the old unit file still exists and so isn't over written.

puged, checked /lib/systemd/system/ and /etc/defaults/, no raspotify files left.

after install, I can connect and it plays normally.

can't test more at this moment, I'm off to work.

JasonLG1979, 2021-11-25:
Right on thanks!!!

JasonLG1979, 2021-11-25:
0.31.2-beta-2 is up. If all goes well I will release 0.31.2 in a couple days.

https://github.com/dtcooper/raspotify/releases/tag/0.31.2-beta-2

Gudio, 2021-11-25:
I had similar issue with error
`Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'ENOMEM: Out of memory'`
I have pi-btaudio installed. It works fine with 0.31.2-beta-2 installed on my side.

JasonLG1979, 2021-11-26:
Thanks everyone for all your help. I've released [0.31.2](https://github.com/dtcooper/raspotify/releases/tag/0.31.2) It will be pushed to the repo shortly.

closed 2021-11-26 by JasonLG1979

lblabr, 2022-07-26:
hey,

i'm quite new to the audio stuff @ raspi. i try to get respotify working....

i'm running a scoreboard, TV connected on HDMI 1, autologin with p default user ... i would liek to use the setup also to play spotify music ... i installed respotify. how do i get the audio to the hdmi ? pulseaudio ist installed (by default ??) at the moment everything is working. what is the best way to get the things done??

- audio output from lxde-pi user
- respotify
- command line (with user pi it also works fine at the moment) 

thanks a lot!

JasonLG1979, 2022-07-26:
@lblabr Public presentation of content from Spotify is a violation of their terms of service. While I personally do not have a problem with it, they certainly do.

I'm sorry but I can't help you flagrantly violate their terms of service because it could potentially lead to them blocking all librespot users.

Please to not use Raspotify or librespot to violate Spotify's terms of service.