dtcooper/raspotify #458: systemd issue (using fifo sink)

bozakov, 2021-11-22:
Running raspotify as a systemd service crashes with the following error:

```
[2021-11-22T09:43:05Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: No such file or directory (os error 2)                            
```

however running the command (copied from journalctl) from the shell everything works as expected:

`sudo -u raspotify /usr/bin/librespot --name pi3 --device-type speaker --backend pipe --device /tmp/snapcast/raspotify_fifo --bitrate 160 --disable-audio-cache --enable-volume-normalisation --volume-ctrl linear --initial-volume 100`

Any ideas what the issue might be?

kirenida, 2021-11-22:
Maybe try what I did [here](https://github.com/dtcooper/raspotify/issues/457#issuecomment-974781127) if you are using the latest version.

JasonLG1979, 2021-11-22:
> Maybe try what I did [here](https://github.com/dtcooper/raspotify/issues/457#issuecomment-974781127) if you are using the latest version.

That's a possibility but I think it's more about adding that path to `ReadWritePaths`.

I'm installing snapcast right now. My plan if I'm right and if I can is to introduce a new `FIFO_DIR` `Environment`.  `ReadWritePaths` punches a hole though the sandbox so ideally I can make it default to nothing by making the default value gibberish, and in the above case the user would add `FIFO_DIR="/tmp/snapcast"` to open up that path.

bozakov, 2021-11-22:
> Maybe try what I did [here](https://github.com/dtcooper/raspotify/issues/457#issuecomment-974781127) if you are using the latest version.

I already tried adding that, but it didn't make any difference unfortunately.

> That's a possibility but I think it's more about adding that path to `ReadWritePaths`.
>
> I'm installing snapcast right now. My plan if I'm right and if I can is to introduce a new `FIFO_DIR` [...]

That sounds like a nice approach, thank for looking into it!

JasonLG1979, 2021-11-22:
@bozakov Try this:
 
 Edit the Raspotify systemd service unit file with `sudo systemctl edit raspotify --full` and replace it's content with:
 
 ```
 ProtectKernelTunables=true
ProtectProc=invisible
ProtectHome=true
ProtectSystem=strict
ProtectClock=yes

DevicePolicy=strict

DeviceAllow=char-alsa rw
DeviceAllow=/dev/null r
DeviceAllow=/dev/random r
DeviceAllow=/dev/urandom r

UMask=077
CacheDirectory=raspotify

PrivateTmp=true
PrivateUsers=true

ProcSubset=pid
NoNewPrivileges=true
PermissionsStartOnly=true
LockPersonality=true
MemoryDenyWriteExecute=true
RemoveIPC=true

CapabilityBoundingSet=

#SystemCallArchitectures=native
SystemCallFilter=@system-service
SystemCallFilter=~@privileged @resources
SystemCallErrorNumber=EPERM

Environment="DEVICE_NAME=Raspotify (%H)"
Environment="BITRATE=160"
Environment="CACHE_ARGS=--disable-audio-cache"
Environment="VOLUME_ARGS=--enable-volume-normalisation --volume-ctrl linear --initial-volume 100"
Environment="BACKEND_ARGS=--backend alsa"
Environment="DEVICE_TYPE=speaker"
Environment="FIFO_DIR=~/"

ConfigurationDirectory=default
EnvironmentFile=-/etc/default/raspotify

ReadWritePaths=-/var/cache/raspotify ${FIFO_DIR}

ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --device-type ${DEVICE_TYPE} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS

[Install]
WantedBy=multi-user.target
```

Then edit `/etc/default/raspotify` and add `FIFO_DIR="/tmp/snapcast"` to the end of the file and then restart everything and see if that helps.

If that works then I'll make those changes and they will be included in the next release.

JasonLG1979, 2021-11-22:
Sorry about the hassle. But such is the nature of security hardening. You lock it down as tight as you can and only open up what's absolutely necessary. I kinda figured there would be some bug reports because I wasn't sure how exactly people use Raspotify (beyond the obvious use cases) so I basically had to scream test it. Basically release it and wait for people to tell me what I broke so I know what to open up.

My plan is to do a 0.31.2 after https://github.com/librespot-org/librespot/pull/883 is merged and if I don't get any more bug reports for other stuff I broke,lol!!!

JasonLG1979, 2021-11-22:
`Just saw your edit - service file loads now. However I'm still getting the same error.`

`However that just seems to be a warning because the default is ~/.`

Maybe I can just make it empty since it just add to the existing list?

Try this:

`Environment="FIFO_DIR="`

`ReadWritePaths=-/var/cache/raspotify $FIFO_DIR`

JasonLG1979, 2021-11-22:
> FWIW my FIFO file is in a subdiretory /tmp/snapcast/raspotify_fifo.

The path has to be a directory. So if the file's path is `/foo/bar/baz` then path we want is `/foo/bar`.

JasonLG1979, 2021-11-22:
The `-` the `=-` in ReadWritePaths=-/var/cache/raspotify $FIFO_DIR is also very important. the `-` means ignore it.

bozakov, 2021-11-22:
hmm - no joy yet. 

strangely I still get the error `Nov 22 19:54:36 pi3 systemd[1]: /etc/systemd/system/raspotify.service:57: ReadWritePaths= path is not absolute, ignoring: $FIFO_DIR` when re-enabling the service. This is the current config snippet:

```
#...
Environment="FIFO_DIR=/tmp"

ConfigurationDirectory=default
EnvironmentFile=-/etc/default/raspotify

ReadWritePaths=-/var/cache/raspotify $FIFO_DIR
```

and `/etc/default/raspotify`

```
BACKEND_ARGS="--backend pipe --device /tmp/snapcast/raspotify_fifo"
FIFO_DIR="/tmp/snapcast"
```

JasonLG1979, 2021-11-22:
Nov 22 19:58:39 pi3 systemd[1]: /etc/systemd/system/raspotify.service:17: Unknown lvalue 'RestrictSUIDSGID' in section 'Service', ignoring                   
```
Nov 22 19:58:39 pi3 systemd[1]: /etc/systemd/system/raspotify.service:20: Unknown lvalue 'ProtectHostname' in section 'Service', ignoring                    
Nov 22 19:58:39 pi3 systemd[1]: /etc/systemd/system/raspotify.service:22: Unknown lvalue 'ProtectKernelLogs' in section 'Service', ignoring                  
Nov 22 19:58:39 pi3 systemd[1]: /etc/systemd/system/raspotify.service:25: Unknown lvalue 'ProtectProc' in section 'Service', ignoring                        
Nov 22 19:58:39 pi3 systemd[1]: /etc/systemd/system/raspotify.service:28: Unknown lvalue 'ProtectClock' in section 'Service', ignoring                       
Nov 22 19:58:39 pi3 systemd[1]: /etc/systemd/system/raspotify.service:38: Unknown lvalue 'ProcSubset' in section 'Service', ignoring
```
These are because you're still on Buster.

```
Nov 22 19:58:39 pi3 systemd[1]: /etc/systemd/system/raspotify.service:57: ReadWritePaths= path is not absolute, ignoring: $FIFO_DIR 
```
This may also be because you're on Buster? I don't get any of those warnings, I'm running Bullseye.

JasonLG1979, 2021-11-22:
Comment out all the Unknown lvalue's and see if that works? In the long run you should prob upgrade though,lol!!!

bozakov, 2021-11-22:
Yeah makes sense that it's due to the systemd version. Commenting out doesn't make any difference (the values are ignored anyways).
Reverting to the .service file from the previous release seems to work fine (albeit without your hardening fixes).

I guess I'll stick with that until I have time to upgrade to Bullseye. Thank you for the fixes so far!

JasonLG1979, 2021-11-22:
> Yeah makes sense that it's due to the systemd version. Commenting out doesn't make any difference (the values are ignored anyways).
Reverting to the .service file from the previous release seems to work fine (albeit without your hardening fixes).

> I guess I'll stick with that until I have time to upgrade to Bullseye. Thank you for the fixes so far!

I'm still trying to figure out how to dynamically open up the path. No matter what I do I can't get it to work correctly. The way I'd like. Another option if all else fails is to just have snapcast listen to alsa.

bozakov, 2021-11-22:
You mean without having a default "allowed" path? Would specifying a non-existing path and prefixing it with `-` not work?

> Another option if all else fails is to just have snapcast listen to alsa. 

I have some other (containerized) music services running, so I would like to avoid using alsa if possible.

JasonLG1979, 2021-11-22:
> You mean without having a default "allowed" path? Would specifying a non-existing path and prefixing it with - not work?

The way it currently is basically the whole file-system is invisible and completely inaccessible except for `/var/cache/raspotify` which is `rw` and `/etc/default` which is `r`. The only reason those can even be seen is because of `CacheDirectory=` and `ConfigurationDirectory=`.

JasonLG1979, 2021-11-22:
Oh and devices of any kind except for sound devices (though `char-alsa rw`) and the pseudo devs `/dev/nul r`, `/dev/null r` and `/dev/urandom r` are the same, totally invisible and completely inaccessible.

Raspotify can't so much as sneeze and fart at the same time.

JasonLG1979, 2021-11-22:
If you use Raspotify in a container you don't really need any of the hardening. You get it already because it's running in a container. I could have 2 service files one current one (and default) and a 2nd less contained for use cases like yours? I can use `Conflicts=` so that enabling one will automatically disable the other.

bozakov, 2021-11-22:
It's good to lock everything down as much as possible
I'm upgrading to bullseye right now, I can also look into the .service file in a day or two.

JasonLG1979, 2021-11-22:
> It's good to lock everything down as much as possible

I think the problem also might be in how `ReadWritePaths=` is evaluated. For the life of me I couldn't get it to register the change in  `/etc/default/raspotify` it may need to be a static directory. Maybe `/tmp/fifo` I don't know what the convention or spec is as far as where you'd typically find a fifio.

JasonLG1979, 2021-11-22:
I'm all about security but as always it's a balancing act between security and usability. I knew I'd have to dial it back a bit to find the sweet spot. Thanks again for helping me out with debugging and testing.

mike-vivas, 2021-11-23:
Hi everyone,
having the same problem.

I got it running after removing "PrivateTmp=true" from the service file. 

According to https://www.freedesktop.org/software/systemd/man/systemd.exec.html this creates a private /tmp for the unit using this option - it does not use the standard /tmp folder. Disabling it makes the FIFO streaming to snapcast working again.
Probably the other changes with FIFO_DIR are not needed anymore - will try this as well and report.

Question is: how bad is it for the security to set PrivateTmp to false? To my eyes this should not be a big security concert but I might be wrong.

mike-vivas, 2021-11-23:
Tested: with "PrivateTmp=false" the changes related to "FIFO_DIR" would not be needed anymore as /tmp is writable for every user/process.
(I removed FIFO_DIR from the service file and from the environment file in /etc/default/raspotify)

JasonLG1979, 2021-11-23:
> Question is: how bad is it for the security to set PrivateTmp to false? To my eyes this should not be a big security concern but I might be wrong.

No one thing really is a deal breaker for me. Like I said usability is obviously important. Although `librespot` doesn't otherwise use `/tmp` at all. I do wish there was a way to open up just a single dir in '/tmp` like `/tmp/fifo` for example. `PrivateTmp=false` seems like a big hammer approach?

> Tested: with "PrivateTmp=false" the changes related to "FIFO_DIR" would not be needed anymore as /tmp is writable for every user/process.
(I removed FIFO_DIR from the service file and from the environment file in /etc/default/raspotify)

`PrivateTmp=false` is the default (unless `PrivateTmp=false` is implied by some other setting) you might just be able to comment out `PrivateTmp=true`

JasonLG1979, 2021-11-23:
In the sort run I'd say we should document this in the wiki and show how to do a drop in override:

`sudo systemctl edit raspotify`

And then:

```
[Service]
PrivateTmp=
PrivateTmp=false
```

In the override. (@bozakov can you verify that that works?)

In the midterm we should look for a better solution that doesn't just open up all of `/tmp`. And in the long run if nothing else can be found I will make `PrivateTmp=false` the default.

mike-vivas, 2021-11-23:
sorry, I have to correct my comment - onyl "PrivateTmp=false" is not enough.
I also have to add /tmp to ReadWritePaths, otherwise no stream.
Will check if it is possible to add it via systemctl override.

I do agree that the PrivateTmp=false might be a hammer approach - will keep checking if I find a better solution.

mike-vivas, 2021-11-23:
"ProtectSystem=strict" makes /tmp read-only (from what I can get from systemd docs).

If I put "ProtectSystem=full" I can do without /tmp in ReadWritePaths and the FIFO-stream works.

["PrivateTmp" always set to false as mentioned above]

mike-vivas, 2021-11-24:
Hey Jason,

Depending on what you need I can offer my help here.
Just some thoughts from my side:

- if you want to leave PrivateTmp to true, you will no be able to put /tmp in ReadWritePaths. PrivateTmp creates a private /tmp for this systemd-unit that you cannot open up to others as far as I understood. There is an option to share private tmp with other units but would require to know their names.
For the FIFO story this means that users will need to use any other folder apart from /tmp and give it free via ReadWritePaths - also ok but it should be specified in the docs that /tmp cannot be used.
Leaving PrivateTmp to false as per default allows users to use /tmp for their FIFO files

- ProtectSystem: either 'full' or 'strict'.
With 'full' you would not need the changes to release FIFO_DIR with ReadWritePaths - but of course more folders are open (less sandboxing / security)
With 'strict' you would need the FIFO_DIR changes and security would be higher.

In the end my preferred solution would be:
- PrivateTmp=false (leave /tmp as possible folder for the FIFO queue)
- ProtectSystem=strict but including the changes for FIFO_DIR

Based on your feedback I can make tests and give you feedback.

JasonLG1979, 2021-11-24:
@mike-vivas of your presented solutions I'm in favor of documenting that `/tmp` can't be used and that they should use some other location (preferably a custom location that only contains the fifo for there own security) and that we make opening that up possible with `ReadWritePaths`.

My reasoning is:

1. `/tmp` is basically a sewer.
2. Using `/tmp` for IPC (that's really what a fifo is, a IPC hack) is an abuse of `/tmp` IMHO.

JasonLG1979, 2021-11-24:
We also might make the var something more generic like READ_WRITE_PATHS and we might also include `ExecPaths=` as a EXEC_PATHS var to allow for `--onevent` scripts with the same caveat that they can't be in `/tmp` (although, I'm  not sure why they would be?) and that preferably they be in a safe-ish location.

JasonLG1979, 2021-11-24:
And ofc anything called from `ExecPaths` by the service will run under the `raspotify` user which has basically no privileges. Certainly not `sudo` so scripts with `sudo` in them will obviously fail.

JasonLG1979, 2021-11-25:
@mike-vivas can you test your changes against https://github.com/dtcooper/raspotify/releases/tag/0.31.2-beta-1

I've redone how the package works a bit there is now no actual `raspotify` user, just a dynamic user named `raspotify`. I also made `apt purge` purge the unit file. Before it left it.

JasonLG1979, 2021-11-25:
[Relevant reading on Dynamic Users](http://0pointer.net/blog/dynamic-users-with-systemd.html)

JasonLG1979, 2021-11-25:
Well scratch `ExecPaths` I get:

`/etc/systemd/system/raspotify.service:73: Unknown key name 'ExecPaths' in section 'Service', ignoring.`

I'm guessing it's not supported in the version of systemd that's in Bullseye.

And no matter what I do I can't set `ReadWritePaths` with a env var. I get:

`/etc/systemd/system/raspotify.service:74: ReadWritePaths= path is not absolute, ignoring: $READ_WRITE_PATHS`

That tells me env vars don't work there it's trying to literally find `$READ_WRITE_PATHS`.

A drop in override might be the only option with the above caveat that it can't be in `/tmp`.

So the override would look like this:
```
[Service]
ReadWritePaths=/path/to/the/fifo
```

bozakov, 2021-11-25:
> @mike-vivas can you test your changes against https://github.com/dtcooper/raspotify/releases/tag/0.31.2-beta-1
> 
> I've redone how the package works a bit there is now no actual `raspotify` user, just a dynamic user named `raspotify`. I also made `apt purge` purge the unit file. Before it left it.

After a quick test this seems to work fine, after moving the fifo file to `/var/run`. No other changes / settings were needed (besides setting the backend in `/etc/default/raspotify`)

JasonLG1979, 2021-11-25:
Ok. I'm going to make a beta-2 for everyone to test and then if all goes we'll be good to go. Thanks everyone!!!

JasonLG1979, 2021-11-25:
0.31.2-beta-2 is up. If all goes well I will release 0.31.2 in a couple days.

https://github.com/dtcooper/raspotify/releases/tag/0.31.2-beta-2

JasonLG1979, 2021-11-25:
> After a quick test this seems to work fine, after moving the fifo file to `/var/run`. No other changes / settings were needed (besides setting the backend in `/etc/default/raspotify`)

Good deal. That should be mentioned in the [Troubleshooting](https://github.com/dtcooper/raspotify/wiki/Troubleshooting) section of the wiki if you would please and we can close this as solved and I can remove the issue tags.

mike-vivas, 2021-11-25:
tested 0.31.2-beta-2 and can confirm that it works.
Indeed no extra ReadWritePaths are needed when moving the FIFO-queue to /var/...
thank you both

bozakov, 2021-11-26:
Added pointer in https://github.com/dtcooper/raspotify/wiki/Troubleshooting.
Thanks for figuring this out!

closed 2021-11-26 by bozakov

JasonLG1979, 2021-11-26:
Awesome thanks!!!

JasonLG1979, 2021-11-26:
Thanks everyone for all your help. I've released [0.31.3](https://github.com/dtcooper/raspotify/releases/tag/0.31.3) It will be pushed to the repo shortly.