dtcooper/raspotify #459: bluealsa PCM errors with 0.31.1

fragande, 2021-11-23:
Sorry in advance for my lack of knowledge, but I'm running into issues with a fresh install of Raspotify on my RPi Zero W running Raspberry Pi OS Lite "Bullseye". I borked my old working Buster setup by trying to build bluealsa from source with AAC support. My playback device is a BT speaker.

The errors I'm getting are:

```
Nov 23 17:15:29 raspberrypi librespot[443]: ALSA lib bluealsa-pcm.c:763:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
Nov 23 17:15:29 raspberrypi librespot[443]: [2021-11-23T17:15:29Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'ENODEV: No such device'

```

Running `librespot` manually does work fine however, so I'm not sure what is causing this. It seems it may be related to other recent issues? Got it working by downgrading to 0.16.0.

JasonLG1979, 2021-11-23:
This may be similar to https://github.com/dtcooper/raspotify/issues/457

If you could edit the systemd unit file with:

`sudo systemctl edit --full raspotify`

Find the line that starts with `RestrictAddressFamilies=` and make it look like this:

`RestrictAddressFamilies=AF_UNIX AF_INET AF_INET6 AF_NETLINK`

And see if that fixes your issue?

If so great. That change has already been commited to master I just haven't done a new release yet.

If not we'll do some more digging.

fragande, 2021-11-23:
> 

Forgot to mention I already tried that by editing /usr/lib/systemd/system/raspotify.service directly, but maybe there's a difference? Anyway, the method above didn't make a difference either. Still the same error after editing, saving and rebooting.

JasonLG1979, 2021-11-23:
No it shouldn't matter I don't think how you edit it?

After you edit it you have to call `sudo systemctl daemon-reload` and `sudo systemctl restart raspotify` for it to take effect.

JasonLG1979, 2021-11-23:
So if I understand correctly you're transmitting over bluetooth from alsa correct?

JasonLG1979, 2021-11-23:
It looks like you may not have the bluealsa alsa sink configure correctly?

JasonLG1979, 2021-11-23:
You might look here:

https://github.com/dtcooper/raspotify/wiki/Play-via-Bluetooth-Speaker-on-Raspberry-Pi-Desktop#play-via-bluetooth-speaker-without-raspberry-pi-desktop

I can't vouch for the howto as I've never done it as I have no need. If I wanted to stream Spotify to a bluetooth speaker I'd just stream it from the controlling device and skip using a Raspberry Pi as a middle man.

JasonLG1979, 2021-11-23:
@bablokb might be able to help better than I?

fragande, 2021-11-23:
Sorry I'm a knucklehead when it comes to stuff like this, but yes, I'm using bluealsa to stream over BT which I believe creates an ALSA PCM device? I'm not sure how it works or what the correct terminology is here ...

I believe it's configured correctly using `asound.conf` as it works when calling `librespot` manually. Also works fine when using 0.16.0 as mentioned before. It's only when using 0.31.0 I get these errors, with everything else being identical.

I use several devices to control Spotify streaming to this BT Speaker (phone, PC and tablet) so that's why I'm using Raspotify fo this. I realize a WiFi speaker is miles better, but this solution has worked OK so far.

JasonLG1979, 2021-11-23:
> Sorry, I'm a knucklehead when it comes to stuff like this,

It's all good. I'm probably the one who broke it,lol!!! 

Try adding:
`AF_BLUETOOTH`

To the `RestrictAddressFamilies=` so it looks like:

`RestrictAddressFamilies=AF_UNIX AF_INET AF_INET6 AF_NETLINK AF_BLUETOOTH`

fragande, 2021-11-23:
That seems to have done the trick! 0.31.0 up and running without any errors in systemd logs. Thank you very much.

JasonLG1979, 2021-11-23:
Good deal. I'll add that to master.

JasonLG1979, 2021-11-23:
We'll keep this open until the release that includes it is released.

JasonLG1979, 2021-11-23:
763a7c821daabe99f01386bd7fb69196dc13f6f8

bablokb, 2021-11-24:
> @bablokb might be able to help better than I?

pi-btaudio is currently broken (at least for Buster and later). I am working on a newer version that uses bluez-alsa instead of bluesalsa. There is still some testing ahead, the dbus-code doing the auto-connect also needs updates.

Using a Raspberry Pi as a middle man is indeed nothing you would normally want to do, if your controlling device can also stream bluetooth.

fragande, 2021-11-24:
> pi-btaudio is currently broken (at least for Buster and later). I am working on a newer version that uses bluez-alsa instead of bluesalsa. There is still some testing ahead, the dbus-code doing the [...]

Good to hear. `pi-btaudio` is very convenient but found out it uses packages that are obsolete in Bullseye yes. It did seem to work in Buster, at least it installed and worked well enough for my use case so thank you for that.

I've been very happy with this solution as it turned my old wall mounted IKEA Eneby BT speaker into a WiFi speaker (Well ... kind of, it's been _always on_ modded so it's pretty close) that I can control with all my devices running Spotify. Much better than manually connecting with BT every time. I realize this may be a very unusual use case though.

JasonLG1979, 2021-11-25:
Can everyone test against https://github.com/dtcooper/raspotify/releases/tag/0.31.2-beta-1

andrewn, 2021-11-25:
FWIW, I'm on Buster and was having this issue using `pi-btaudio` streaming from a _Raspberry Pi 3 Modal A+_ to a connected Bluetooth speaker.

It's fixed for me using the `AF_UNIX` change in this issue.

```
pi@raspberrypi:~ $ lsb_release -a
Description:	Raspbian GNU/Linux 10 (buster)
```

Using `pi-btaudio`, connecting to the device from Spotify would cause `librespot` to crash:

```
Nov 25 14:43:32 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Nov 25 14:43:32 raspberrypi librespot[610]: [2021-11-25T14:43:32Z INFO  librespot] librespot 0.3.1 85d6049 (Built on 2021-11-18, Build ID: YsLswf3d, Profile: release)
Nov 25 14:43:43 raspberrypi librespot[610]: [2021-11-25T14:43:43Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-a-r9m3.ap.spotify.com:4070"
Nov 25 14:43:44 raspberrypi librespot[610]: [2021-11-25T14:43:44Z INFO  librespot_core::session] Authenticated as "..my username.." !
Nov 25 14:43:44 raspberrypi librespot[610]: [2021-11-25T14:43:44Z INFO  librespot_playback::mixer::softmixer] Mixing with softvol and volume control: Linear
Nov 25 14:43:44 raspberrypi librespot[610]: [2021-11-25T14:43:44Z INFO  librespot_playback::convert] Converting with ditherer: tpdf
Nov 25 14:43:44 raspberrypi librespot[610]: [2021-11-25T14:43:44Z INFO  librespot_playback::audio_backend::alsa] Using AlsaSink with format: S16
Nov 25 14:43:44 raspberrypi librespot[610]: [2021-11-25T14:43:44Z INFO  librespot_core::session] Country: "GB"
Nov 25 14:43:44 raspberrypi librespot[610]: [2021-11-25T14:43:44Z INFO  librespot_playback::player] Loading <Bizness> with Spotify URI <spotify:track:08oSPuXfichvi5VRAaewbV>
Nov 25 14:43:44 raspberrypi librespot[610]: [2021-11-25T14:43:44Z INFO  librespot_playback::player] <Bizness> (263733 ms) loaded
Nov 25 14:43:44 raspberrypi librespot[610]: ALSA lib bluealsa-pcm.c:755:(_snd_pcm_bluealsa_open) Couldn't initialize D-Bus context: Failed to open socket: Address family not supported by protocol
Nov 25 14:43:44 raspberrypi librespot[610]: [2021-11-25T14:43:44Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'ENOMEM: Out of memory'
Nov 25 14:43:44 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Nov 25 14:43:44 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
```

Adding `AF_UNIX` to `RestrictAddressFamilies` in the systemd unit file, fixed it. No `AF_BLUETOOTH` was needed. 🎉 

```
Nov 25 14:45:37 raspberrypi librespot[643]: [2021-11-25T14:45:37Z INFO  librespot_core::session] Connecting to AP "gew1-accesspoint-a-k28s.ap.spotify.com:4070"
Nov 25 14:45:37 raspberrypi librespot[643]: [2021-11-25T14:45:37Z INFO  librespot_core::session] Authenticated as "..my username.." !
Nov 25 14:45:37 raspberrypi librespot[643]: [2021-11-25T14:45:37Z INFO  librespot_playback::mixer::softmixer] Mixing with softvol and volume control: Linear
Nov 25 14:45:37 raspberrypi librespot[643]: [2021-11-25T14:45:37Z INFO  librespot_playback::convert] Converting with ditherer: tpdf
Nov 25 14:45:37 raspberrypi librespot[643]: [2021-11-25T14:45:37Z INFO  librespot_playback::audio_backend::alsa] Using AlsaSink with format: S16
Nov 25 14:45:37 raspberrypi librespot[643]: [2021-11-25T14:45:37Z INFO  librespot_core::session] Country: "GB"
Nov 25 14:45:38 raspberrypi librespot[643]: [2021-11-25T14:45:38Z INFO  librespot_playback::player] Loading <Bizness> with Spotify URI <spotify:track:08oSPuXfichvi5VRAaewbV>
Nov 25 14:45:39 raspberrypi librespot[643]: [2021-11-25T14:45:39Z INFO  librespot_playback::player] <Bizness> (263733 ms) loaded
```

JasonLG1979, 2021-11-25:
> It's fixed for me using the `AF_UNIX` change in this issue.

Good deal!!!
0.31.2-beta-2 is up. If all goes well I will release 0.31.2 in a couple days.

https://github.com/dtcooper/raspotify/releases/tag/0.31.2-beta-2

JasonLG1979, 2021-11-25:
@fragande can you double check that `AF_BLUETOOTH` is actually needed? @andrewn only needed `AF_UNIX`.

`sudo systemctl edit --full raspotify`

Remove `AF_BLUETOOTH`.

`sudo systemctl daemon-reload`

`sudo systemctl restart raspotify`

And see if it still works. If it still works then `AF_BLUETOOTH` is not required and will be removed since I really only want to open up what I absolutely have to.

MelGrubb, 2021-11-26:
Does this mean that bluetooth speakers will be officially supported in the next version? I spent a couple nights trying to get this working and after reading this I'm not surprised I couldn't get it to work. Bluetooth support would mean that I no longer have to add some external audio HAT to support audio from the Pi Zero. That's huge. I have a few Zero Ws laying around with nothing to do, and I'd love to turn them into Spotify Connect devices.

JasonLG1979, 2021-11-26:
@MelGrubb not supported but I don't want to make it more difficult for no good reason. Unsupported doesn't mean impossible it just means that you're on your own to figure it out.

JasonLG1979, 2021-11-26:
You also don't really need a DAC HAT. A $5 USB audio dongle DAC is going to sound much, much better than Bluetooth. If the downstream audio device has an analog input that would be my choice vs configuring a Zero as a Spotify Connect Bluetooth bridge.

MelGrubb, 2021-11-26:
As a project though, needing nothing more than a Zero W and power supply is very tempting. You don't even have to dedicate a separate Pi for the job if you have one that isn't terribly busy or has sufficient headroom. I can think of three Pis that aren't slammed and are within BT range of the speaker I have in mind. Any one of them could become the Raspotify "node" for the basement without having to be physically next to the speaker.

JasonLG1979, 2021-11-26:
> As a project though, needing nothing more than a Zero W and power supply is very tempting. You don't even have to dedicate a separate Pi for the job if you have one that isn't terribly busy or has [...]

You could collaborate with @bablokb at https://github.com/bablokb/pi-btaudio as far as pi-bt-audio is concerned Raspotify is just another thing making noise. I imagine you could route Raspotify and all manner of other services over Bluetooth. Bluetooth support is out of scope of Raspotify.

JasonLG1979, 2021-11-26:
I can fully appreciate wanting to make the most of your Pi though. My Pi 4 8GB (DAC HAT attached) runs Raspotify (connected to my AVR), a Plex server (God send if you have kids and you don't want them to scratch up your movies), NFS shares for automatic backups and whatever random other things I'm playing with at the moment. And it's still mostly idle if you look at it's sue over the long run.

andrewn, 2021-11-26:
@MelGrubb, for what it's worth, [here are the steps I followed](https://andrewnicolaou.co.uk/posts/2021/spotify-pi-bluetooth).

I have a waterproof bluetooth speaker in the shower and want to use it with Spotify. Works well so far :-D

fragande, 2021-11-26:
> @fragande can you double check that `AF_BLUETOOTH` is actually needed? @andrewn only needed `AF_UNIX`.
>
> `sudo systemctl edit --full raspotify`
>
> Remove `AF_BLUETOOTH`.
>
> `sudo systemctl [...]

Sorry for the late reply. It seems that `AF_UNIX` is indeed enough. Seems I was editing `/usr/lib/systemd/system/raspotify.service` rather than `/etc/systemd/system/raspotify.service`. My understanding of the file structure is very limited. Somewhere between adding AF_UNIX and AF_BLUETOOTH to the former I probably added AF_UNIX to the latter using `sudo systemctl edit --full raspotify`, This made me believe AF_BLUETOOTH was the solution.

Sorry for the confusion.

JasonLG1979, 2021-11-26:
> Sorry for the confusion.

It's all good. I will remove `AF_BLUETOOTH` and do a release shortly.

JasonLG1979, 2021-11-26:
Thanks everyone for all your help. I've released [0.31.2](https://github.com/dtcooper/raspotify/releases/tag/0.31.2) It will be pushed to the repo shortly.

closed 2021-11-26 by JasonLG1979