dtcooper/raspotify #461: Error writing from AlsaSink buffer to PCM

fernfrost, 2021-11-26:
Hi

Get this error when connecting:

> Nov 26 22:03:39 DietPi librespot[1197]: [2021-11-26T21:03:39Z WARN  librespot_playback::audio_backend::alsa] Error writing from AlsaSink buffer to PCM, trying to recover, ALSA function 'snd_pcm_writei' failed with error 'EIO: I/O error'
>
> Nov 26 22:03:39 DietPi librespot[1197]: [2021-11-26T21:03:39Z ERROR librespot_playback::player] Audio Sink Error On Write: <AlsaSink> ALSA function 'snd_pcm_recover' failed with error 'EIO: I/O error'

aplay -l shows the following:

> dietpi@DietPi:/$ sudo aplay -l
> **** List of PLAYBACK Hardware Devices ****
> card 0: sndtpahermesrpi [snd_tpa_hermes_rpi], device 0: TPA-Hermes-RPi HiFi tpa-hermes-rpi-codec-0 [TPA-Hermes-RPi HiFi tpa
> hermes-rpi-codec-0]
>   Subdevices: 1/1
>   Subdevice #0: subdevice #0

I have have tried setting `--device hw:0,0`according to #235 but no luck

JasonLG1979, 2021-11-26:
What is the output of `aplay -Dhw:0,0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav`

fernfrost, 2021-11-26:
> ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
> aplay: main:830: audio open error: No such file or directory

fernfrost, 2021-11-26:
Output of aplay -L:

>dietpi@DietPi:/$ sudo aplay -L
>null
>    Discard all samples (playback) or generate zero samples (capture)
>hw:CARD=sndtpahermesrpi,DEV=0
>    snd_tpa_hermes_rpi, TPA-Hermes-RPi HiFi tpa-hermes-rpi-codec-0
>    Direct hardware device without any conversions
>plughw:CARD=sndtpahermesrpi,DEV=0
>    snd_tpa_hermes_rpi, TPA-Hermes-RPi HiFi tpa-hermes-rpi-codec-0
>    Hardware device with all software conversions
>sysdefault:CARD=sndtpahermesrpi
>    snd_tpa_hermes_rpi, TPA-Hermes-RPi HiFi tpa-hermes-rpi-codec-0
>    Default Audio Device
>dmix:CARD=sndtpahermesrpi,DEV=0
>    snd_tpa_hermes_rpi, TPA-Hermes-RPi HiFi tpa-hermes-rpi-codec-0
>    Direct sample mixing device

JasonLG1979, 2021-11-26:
That's a driver issue. I've never heard of `sndtpahermesrpi` is it `https://github.com/miero/tpa-hermes-rpi-dev` it looks like you updated it for 5.10. Your driver is obviously broken.

fernfrost, 2021-11-26:
Correct. I updated it to be able to install but havnt gotten around to testing it until now.
Suspected it was a driver issue. Wil dig into that.

JasonLG1979, 2021-11-26:
> Correct. I updated it to be able to install but havnt gotten around to testing it until now.
> Suspected it was a driver issue. Wil dig into that.

Ok then why did you file a bug here if you can easily tell it's not a Raspotify issue?

closed 2021-11-26 by JasonLG1979

fernfrost, 2021-11-26:
Sorry, forgot to run with sudo.

Output of `sudo aplay -Dhw:0,0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav`

    dietpi@DietPi:~$ sudo aplay -Dhw:0,0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav
    Playing WAVE '/usr/share/sounds/alsa/Front_Right.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
    HW Params of device "hw:0,0":
    --------------------
    ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
    FORMAT:  S16_LE S24_LE
    SUBFORMAT:  STD
    SAMPLE_BITS: [16 32]
    FRAME_BITS: [32 64]
    CHANNELS: 2
    RATE: [44100 384000]
    PERIOD_TIME: (83 1486078)
    PERIOD_SIZE: [32 65536]
    PERIOD_BYTES: [256 524288]
    PERIODS: [2 4096]
    BUFFER_TIME: (166 2972155)
    BUFFER_SIZE: [64 131072]
    BUFFER_BYTES: [256 524288]
    TICK_TIME: ALL
    --------------------
    aplay: set_params:1349: Channels count non available

JasonLG1979, 2021-11-26:
You shouldn't need to run that with sudo. I'm locking this issue. This is not a Raspotify issue.