dtcooper/raspotify #464: command line options/args to send 'next track'' command possible?

jakka351, 2021-11-27:
Hi I have got raspotify on a PI in my car, I want to hook up to my canbus steering wheel controls - I have the button presses available as key presses or a command or anything really. Any options here?

JasonLG1979, 2021-11-27:
No. You would need to roll your own control interface by connecting to Spotify's web api.

closed 2021-11-27 by JasonLG1979

jakka351, 2021-11-28:
For anyone that comes looking with similar question - I have hacked together this monster https://github.com/jakka351/Rpotify and have canbus steering wheel controls controlling raspotify on a RPI4 playing wirelessly from my phone.

JasonLG1979, 2021-11-28:
You could add a page to the wiki?

jakka351, 2021-11-30:
I will write something up

JasonLG1979, 2021-11-30:
Maybe page in the wiki with a list of links to projects that use Raspotify along with a very brief description of the project?

JasonLG1979, 2021-11-30:
@jakka351 Done:

https://github.com/dtcooper/raspotify/wiki/Projects-related-to-Raspotify

jakka351, 2021-12-07:
Cool.