dtcooper/raspotify #466: librespot crashes with ALSA error EINVAL: Invalid argument on playback

oebeledrijfhout, 2021-11-28:
I'm running the latest raspotify on a Vero4K and I'm seeing this error on playback and I can't play anything:

```
Nov 28 10:41:15 osmc0 systemd[1]: Started Raspotify (Spotify Connect Client).
Nov 28 10:41:15 osmc0 librespot[23855]: [2021-11-28T09:41:15Z INFO  librespot] librespot 0.3.1 bbd575e (Built on 2021-11-26, Build ID: a6e0Ery3, Profile: release)
<...>
Nov 28 10:41:41 osmc0 librespot[23855]: [2021-11-28T09:41:41Z INFO  librespot_playback::player] Loading <Nomad> with Spotify URI <spotify:track:75zwzv0Q6ZJdkBrJUYvtAC>
Nov 28 10:41:41 osmc0 librespot[23855]: [2021-11-28T09:41:41Z DEBUG librespot_audio::fetch] Downloading file 4fcb711a5f054ffa9fe1c2b651638d79718c9278
Nov 28 10:41:41 osmc0 librespot[23855]: [2021-11-28T09:41:41Z DEBUG librespot_core::channel] new ChannelManager
Nov 28 10:41:41 osmc0 librespot[23855]: [2021-11-28T09:41:41Z DEBUG librespot_core::audio_key] new AudioKeyManager
Nov 28 10:41:41 osmc0 librespot[23855]: [2021-11-28T09:41:41Z INFO  librespot_playback::player] <Nomad> (456200 ms) loaded
Nov 28 10:41:41 osmc0 librespot[23855]: [2021-11-28T09:41:41Z DEBUG librespot_playback::player] Normalisation Data: NormalisationData { track_gain_db: -6.5130596, track_peak: 0.9969521, album_gain_db: -7.3830442, album_peak: 1.0025475 }
Nov 28 10:41:41 osmc0 librespot[23855]: [2021-11-28T09:41:41Z DEBUG librespot_playback::player] Calculated Normalisation Factor for Track: 47.24%
Nov 28 10:41:41 osmc0 librespot[23855]: [2021-11-28T09:41:41Z TRACE librespot_playback::player] == Starting sink ==
Nov 28 10:41:41 osmc0 librespot[23855]: [2021-11-28T09:41:41Z ERROR librespot_playback::player] Audio Sink Error Invalid Parameters: <AlsaSink> Hardware, ALSA function 'snd_pcm_hw_params_set_buffer_time_near' failed with error 'EINVAL: Invalid argument'
Nov 28 10:41:42 osmc0 systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Nov 28 10:41:42 osmc0 systemd[1]: raspotify.service: Failed with result 'exit-code'.
Nov 28 10:41:52 osmc0 systemd[1]: raspotify.service: Service RestartSec=10s expired, scheduling restart.
Nov 28 10:41:52 osmc0 systemd[1]: raspotify.service: Scheduled restart job, restart counter is at 1.
Nov 28 10:41:52 osmc0 systemd[1]: Stopped Raspotify (Spotify Connect Client).
Nov 28 10:41:52 osmc0 systemd[1]: Started Raspotify (Spotify Connect Client).
```

Is this a bug in raspotify or should I file this @ librespot?

JasonLG1979, 2021-11-28:
What audio device are you running though?

oebeledrijfhout, 2021-11-28:
This is the librespot process:

`/usr/bin/librespot --name Raspotify (osmc0) --device-type speaker --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --volume-ctrl linear --initial-volume 100 --verbose`

The audio receiver is connected through the S/PDIF (optical) port and I just tried changing the audio output in OSMC to HDMI only (was the default HDMI, S/PDIF, analogue) and rebooted, but that made no difference.

JasonLG1979, 2021-11-28:
What are the outputs of `aplay -l` and `aplay -L`?

oebeledrijfhout, 2021-11-28:
```
osmc@osmc0:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: AMLMESONAUDIO [AML-MESONAUDIO], device 0: I2S T9015-audio-hifi-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: AMLMESONAUDIO [AML-MESONAUDIO], device 1: SPDIF dit-hifi-1 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: AMLMESONAUDIO [AML-MESONAUDIO], device 2: PCM pcm2bt-pcm-2 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
and
```
osmc@osmc0:~$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
btaudio
    Bluetooth Audio
default:CARD=AMLMESONAUDIO
    AML-MESONAUDIO,
    Default Audio Device
sysdefault:CARD=AMLMESONAUDIO
    AML-MESONAUDIO,
    Default Audio Device
hdmi:CARD=AMLMESONAUDIO,DEV=0
    AML-MESONAUDIO,
    HDMI Audio Output
dmix:CARD=AMLMESONAUDIO,DEV=0
    AML-MESONAUDIO,
    Direct sample mixing device
dmix:CARD=AMLMESONAUDIO,DEV=1
    AML-MESONAUDIO,
    Direct sample mixing device
dmix:CARD=AMLMESONAUDIO,DEV=2
    AML-MESONAUDIO,
    Direct sample mixing device
dsnoop:CARD=AMLMESONAUDIO,DEV=0
    AML-MESONAUDIO,
    Direct sample snooping device
dsnoop:CARD=AMLMESONAUDIO,DEV=1
    AML-MESONAUDIO,
    Direct sample snooping device
dsnoop:CARD=AMLMESONAUDIO,DEV=2
    AML-MESONAUDIO,
    Direct sample snooping device
hw:CARD=AMLMESONAUDIO,DEV=0
    AML-MESONAUDIO,
    Direct hardware device without any conversions
hw:CARD=AMLMESONAUDIO,DEV=1
    AML-MESONAUDIO,
    Direct hardware device without any conversions
hw:CARD=AMLMESONAUDIO,DEV=2
    AML-MESONAUDIO,
    Direct hardware device without any conversions
plughw:CARD=AMLMESONAUDIO,DEV=0
    AML-MESONAUDIO,
    Hardware device with all software conversions
plughw:CARD=AMLMESONAUDIO,DEV=1
    AML-MESONAUDIO,
    Hardware device with all software conversions
plughw:CARD=AMLMESONAUDIO,DEV=2
    AML-MESONAUDIO,
    Hardware device with all software conversions
```

JasonLG1979, 2021-11-28:
What is are the outputs of:

The "default device" full software mixing and conversion:

`aplay --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav`

I2S T9015-audio-hifi-0, SPDIF dit-hifi-1, and PCM pcm2bt-pcm-2 actual hardware no software mixing or conversion.

`aplay -Dhw:CARD=AMLMESONAUDIO,DEV=0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav`

`aplay -Dhw:CARD=AMLMESONAUDIO,DEV=1 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav`

`aplay -Dhw:CARD=AMLMESONAUDIO,DEV=2 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav`

I2S T9015-audio-hifi-0, SPDIF dit-hifi-1, and PCM pcm2bt-pcm-2 no software mixing or but with software conversion.

`aplay -Dplughw:CARD=AMLMESONAUDIO,DEV=0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav`

`aplay -Dplughw:CARD=AMLMESONAUDIO,DEV=1 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav`

`aplay -Dplughw:CARD=AMLMESONAUDIO,DEV=2 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav`

oebeledrijfhout, 2021-11-28:
first command, sound plays normally:
```
osmc@osmc0:~$ aplay --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Right.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
HW Params of device "default":
--------------------
ACCESS:  MMAP_INTERLEAVED MMAP_NONINTERLEAVED RW_INTERLEAVED RW_NONINTERLEAVED
FORMAT:  S16_LE S16_BE S24_LE S32_LE S32_BE S24_3LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [16 320000]
CHANNELS: [1 10000]
RATE: [4000 4294967295)
PERIOD_TIME: (10 512000]
PERIOD_SIZE: (0 2147483648)
PERIOD_BYTES: (0 4294967295)
PERIODS: (0 4294967295]
BUFFER_TIME: [1 4294967295]
BUFFER_SIZE: [1 2147483647]
BUFFER_BYTES: [2 4294967295]
TICK_TIME: ALL
--------------------
```
second command, nothing plays:
```
osmc@osmc0:~$ aplay -Dhw:CARD=AMLMESONAUDIO,DEV=0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Right.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
HW Params of device "hw:CARD=AMLMESONAUDIO,DEV=0":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 256]
CHANNELS: [2 8]
RATE: [32000 192000]
PERIOD_TIME: (10 512000]
PERIOD_SIZE: [2 16384]
PERIOD_BYTES: [64 65536]
PERIODS: [2 1024]
BUFFER_TIME: (20 4096000]
BUFFER_SIZE: [4 131072]
BUFFER_BYTES: [64 524288]
TICK_TIME: ALL
--------------------
aplay: set_params:1345: Channels count non available
```

third, sound plays distorted:

```
osmc@osmc0:~$ aplay -Dhw:CARD=AMLMESONAUDIO,DEV=1 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Right.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
HW Params of device "hw:CARD=AMLMESONAUDIO,DEV=1":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [16 256]
CHANNELS: [1 8]
RATE: [32000 192000]
PERIOD_TIME: (10 1024000]
PERIOD_SIZE: [2 32768]
PERIOD_BYTES: [64 65536]
PERIODS: [2 1024]
BUFFER_TIME: (20 8192000]
BUFFER_SIZE: [4 262144]
BUFFER_BYTES: [64 524288]
TICK_TIME: ALL
--------------------
```

fourth, nothing plays:
```
osmc@osmc0:~$ aplay -Dhw:CARD=AMLMESONAUDIO,DEV=2 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Right.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
HW Params of device "hw:CARD=AMLMESONAUDIO,DEV=2":
--------------------
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [16 512]
CHANNELS: [1 16]
RATE: [8000 192000]
PERIOD_TIME: (5 4096000]
PERIOD_SIZE: [1 32768]
PERIOD_BYTES: [64 65536]
PERIODS: [2 1024]
BUFFER_TIME: (10 32768000]
BUFFER_SIZE: [2 262144]
BUFFER_BYTES: [32 524288]
TICK_TIME: ALL
--------------------
```
fifth, sound plays normally:
```
osmc@osmc0:~$ aplay -Dplughw:CARD=AMLMESONAUDIO,DEV=0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Right.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
HW Params of device "plughw:CARD=AMLMESONAUDIO,DEV=0":
--------------------
ACCESS:  MMAP_INTERLEAVED MMAP_NONINTERLEAVED MMAP_COMPLEX RW_INTERLEAVED RW_NONINTERLEAVED
FORMAT:  S8 U8 S16_LE S16_BE U16_LE U16_BE S24_LE S24_BE U24_LE U24_BE S32_LE S32_BE U32_LE U32_BE FLOAT_LE FLOAT_BE FLOAT64_LE FLOAT64_BE MU_LAW A_LAW IMA_ADPCM S20_LE S20_BE U20_LE U20_BE S24_3LE S24_3BE U24_3LE U24_3BE S20_3LE S20_3BE U20_3LE U20_3BE S18_3LE S18_3BE U18_3LE U18_3BE
SUBFORMAT:  STD
SAMPLE_BITS: [4 64]
FRAME_BITS: [4 640000]
CHANNELS: [1 10000]
RATE: [4000 4294967295)
PERIOD_TIME: (10 512000]
PERIOD_SIZE: (0 2199023256)
PERIOD_BYTES: (0 4294967295)
PERIODS: (0 4294967295]
BUFFER_TIME: [1 4294967295]
BUFFER_SIZE: [1 4294967294]
BUFFER_BYTES: [1 4294967295]
TICK_TIME: ALL
--------------------
```

sixth, sound plays distorted (in the same way as the third):
```
osmc@osmc0:~$ aplay -Dplughw:CARD=AMLMESONAUDIO,DEV=1 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Right.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
HW Params of device "plughw:CARD=AMLMESONAUDIO,DEV=1":
--------------------
ACCESS:  MMAP_INTERLEAVED MMAP_NONINTERLEAVED MMAP_COMPLEX RW_INTERLEAVED RW_NONINTERLEAVED
FORMAT:  S8 U8 S16_LE S16_BE U16_LE U16_BE S24_LE S24_BE U24_LE U24_BE S32_LE S32_BE U32_LE U32_BE FLOAT_LE FLOAT_BE FLOAT64_LE FLOAT64_BE MU_LAW A_LAW IMA_ADPCM S20_LE S20_BE U20_LE U20_BE S24_3LE S24_3BE U24_3LE U24_3BE S20_3LE S20_3BE U20_3LE U20_3BE S18_3LE S18_3BE U18_3LE U18_3BE
SUBFORMAT:  STD
SAMPLE_BITS: [4 64]
FRAME_BITS: [4 640000]
CHANNELS: [1 10000]
RATE: [4000 4294967295)
PERIOD_TIME: (10 1024000]
PERIOD_SIZE: (0 4294967295)
PERIOD_BYTES: (0 4294967295)
PERIODS: (0 4294967295]
BUFFER_TIME: [1 4294967295]
BUFFER_SIZE: [1 4294967294]
BUFFER_BYTES: [1 4294967295]
TICK_TIME: ALL
--------------------
```
seventh, nothing plays:
```
osmc@osmc0:~$ aplay -Dplughw:CARD=AMLMESONAUDIO,DEV=2 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Right.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
HW Params of device "plughw:CARD=AMLMESONAUDIO,DEV=2":
--------------------
ACCESS:  MMAP_INTERLEAVED MMAP_NONINTERLEAVED MMAP_COMPLEX RW_INTERLEAVED RW_NONINTERLEAVED
FORMAT:  S8 U8 S16_LE S16_BE U16_LE U16_BE S24_LE S24_BE U24_LE U24_BE S32_LE S32_BE U32_LE U32_BE FLOAT_LE FLOAT_BE FLOAT64_LE FLOAT64_BE MU_LAW A_LAW IMA_ADPCM S20_LE S20_BE U20_LE U20_BE S24_3LE S24_3BE U24_3LE U24_3BE S20_3LE S20_3BE U20_3LE U20_3BE S18_3LE S18_3BE U18_3LE U18_3BE
SUBFORMAT:  STD
SAMPLE_BITS: [4 64]
FRAME_BITS: [4 640000]
CHANNELS: [1 10000]
RATE: [4000 4294967295)
PERIOD_TIME: (5 4096000]
PERIOD_SIZE: (0 4294967295)
PERIOD_BYTES: (0 4294967295)
PERIODS: (0 4294967295]
BUFFER_TIME: [1 4294967295]
BUFFER_SIZE: [1 4294967294]
BUFFER_BYTES: [1 4294967295]
TICK_TIME: ALL
--------------------
```

JasonLG1979, 2021-11-28:
If they play or not isn't important. All I care about is the HW Params of each thing. `apaly` is weird in that you can't get the HW Params unless you try to play something. It's just a quirk of the program.

JasonLG1979, 2021-11-28:
Now what's the output of:

`librespot --device ?`

oebeledrijfhout, 2021-11-28:
```
osmc@osmc0:~$ librespot --device ?
[2021-11-28T20:13:40Z INFO  librespot] librespot 0.3.1 bbd575e (Built on 2021-11-26, Build ID: a6e0Ery3, Profile: release)

        Compatible alsa device(s):

        ------------------------------------------------------

osmc@osmc0:~$
```

that doesn't look good

JasonLG1979, 2021-11-28:
Make sure nothing else at all is using your audio devices.

oebeledrijfhout, 2021-11-28:
I stopped the mediacenter service to ensure Kodi isn't running but I still get the same result from the librespot command. I'll post on the OSMC forum, there has been an update there as well at the same time raspotify was updated.

JasonLG1979, 2021-11-28:
You could try adding this to `/etc/asound.conf` (`/etc/asound.conf does not exist by default generally):
```
 # /etc/asound.conf

pcm.!default {
    type plug
    slave.pcm {
        type dmix
        ipc_key {
            @func refer
            name defaults.pcm.ipc_key
        }
        ipc_gid {
            @func refer
            name defaults.pcm.ipc_gid
        }
        ipc_perm {
            @func refer
            name defaults.pcm.ipc_perm
        }
        slave {
            pcm {
                type hw
                card AMLMESONAUDIO
                device 0
            }
            channels 2
            periods 0
            period_size 0
            buffer_size 0
            buffer_time 500000
            period_time 100000
            rate 44100
            format S32_LE
        }
        bindings {
            0 0
            1 1
        }
    }
}

ctl.!default {
    type hw
    card AMLMESONAUDIO
}
```

And then you'd pass `--format S32` to librespot.

oebeledrijfhout, 2021-11-28:
now I get a different error from librespot:

```
ALSA lib pcm_direct.c:1216:(snd1_pcm_direct_initialize_slave) unable to set buffer time
ALSA lib pcm_dmix.c:1120:(snd_pcm_dmix_open) unable to initialize slave
[2021-11-28T21:09:51Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'EINVAL: Invalid argument'
```

JasonLG1979, 2021-11-28:
OK I've got a [`librespot` branch](https://github.com/JasonLG1979/librespot/tree/dynamic-alsa-buffer) that may work. Give me a bit and I'll build you a Raspotify .deb based on it for you to test.

JasonLG1979, 2021-11-28:
Revert `/etc/asound.conf` to whatever state it was in.

And try this out:

https://drive.google.com/file/d/1lyj-JDCW6B__iqaQk2ZYEZrtP4is-QlM/view?usp=sharing

I came up with that a couple weeks ago while trying to figure out someone else's issue but they never got back to me to tell me if it worked or not.

JasonLG1979, 2021-11-28:
As the name suggests it tries to dynamically sets the buffer sizes based what the device reports as it's min and max values.

oebeledrijfhout, 2021-11-28:
I deleted `/etc/asound.conf` as it wasn't there before and installed the deb. I get this error now:

```
[2021-11-28T21:40:31Z ERROR librespot_playback::player] Audio Sink Error Invalid Parameters: <AlsaSink> Hardware, ALSA function 'snd_pcm_hw_params_set_buffer_size_near' failed with error 'EINVAL: Invalid argument'
```

JasonLG1979, 2021-11-28:
That's weird because basically what that branch does is ask the hardware what it's min and max buffer sizes are and it never asks for anything out of that range. In theory logically it should never be an invalid parameter if it's in between it's own reported min and max.  I'm starting to think that this is a driver/hardware issue like there are holes in that min max range that aren't supported or something. Or something else is failing silently before that?

JasonLG1979, 2021-11-28:
Let's kick this up to `librespot` and we'll figure it out there since this is obviously an upstream issue (either `librespot`, Vero4K or osmc). It's nothing to do with your configuration as far as I can tell and technically the Vero4K isn't a supported device and osmc isn't a supported OS as far as Raspotify is concerned.

We can leave this open and you can reference it in your `librespot` bug report.

oebeledrijfhout, 2021-11-28:
thank you for your effort! I'll open a bug report at `librespot`.

JasonLG1979, 2021-11-28:
> thank you for your effort! I'll open a bug report at `librespot`.

I wrote the code in `librespot` I'll see you there,lol!!!

JasonLG1979, 2021-12-02:
@oebeledrijfhout I'm going to close this as I feel like we've made good progress upstream and are close to a fix there. After we get it squared away I will release a new version that includes the fix. Thanks again for all your help. You have no idea how helpful it is to have someone willing and able to test things out especially with ALSA stuff.

closed 2021-12-02 by JasonLG1979

oebeledrijfhout, 2021-12-02:
Glad to help, feel free to ping me if you want me to test something.