dtcooper/raspotify #467: raspotify doesnt start after update on Retropie - Jessie

pelztier3011, 2021-11-28:
hey,
after update raspotify it doesnt start anymore.
here a the output of the status:
`sudo systemctl status raspotify.service
● raspotify.service - Raspotify (Spotify Connect Client)
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2021-11-28 22:25:51 CET; 3s ago
     Docs: https://github.com/dtcooper/raspotify
           https://github.com/librespot-org/librespot
           https://github.com/dtcooper/raspotify/wiki
           https://github.com/librespot-org/librespot/wiki/Options
  Process: 1093 ExecStart=/usr/bin/librespot --name ${DEVICE_NAME} --device-type ${DEVICE_TYPE} $BACKEND_ARGS --bitrate ${BITRATE} $CACHE_ARGS $VOLUME_ARGS $OPTIONS (code=exited, status=127)
 Main PID: 1093 (code=exited, status=127)

Nov 28 22:25:51 retropie systemd[1]: raspotify.service: Failed with result 'exit-code'.`

closed 2021-11-28 by github-actions[bot]

pelztier3011, 2021-11-28:
sorry, it is stretch.
my bad.

JasonLG1979, 2021-11-28:
Still not supported. Bullseye or bust my friend.