dtcooper/raspotify #468: `.service` file update broke pipe backend

jgabriel98, 2021-11-30:
after commit a844a33 the backend `pipe` doesn't work anymore.

## How to reproduce:

1. add the following lines to `/etc/default/raspotify`:

    ```bash
    OPTIONS="--device /tmp/fifo_raspotify"
    BACKEND_ARGS="--backend pipe"
    ```
2. create a pipe with read and write permissions:    `mkfifo /tmp/fifo_raspotify --mode=0666`
    result:
    ```
    $ ls -l /tmp/fifo_raspotify
    prw-rw-rw- 1 pi pi 0 nov 30 00:04 /tmp/fifo_raspotify
    ```
 
3. restart `raspotify.service`
4. connect to device and try to start a song, it will break
 
#### note:
When reverting commit a844a33 and reproducing the steps above (also running `systemctl daemon-reload` before step 3), everything works fine.
  
 --------------------------------
  
 ## Early investigations:
 - The the `raspotify.service` logs (`journalctl -u raspotify.service -f`) indicates problems with the pipe file visibility:
<pre><code>
nov 30 00:25:57 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
nov 30 00:25:57 raspberrypi librespot[17310]: [2021-11-30T03:25:57Z INFO  librespot] librespot 0.3.1 bbd575e (Built on 2021-11-26, Build ID: a6e0Ery3, Profile: release)
nov 30 00:26:07 raspberrypi librespot[17310]: [2021-11-30T03:26:07Z INFO  librespot_core::session] Connecting to AP "guc3-accesspoint-a-n88s.ap.spotify.com:4070"
nov 30 00:26:07 raspberrypi librespot[17310]: [2021-11-30T03:26:07Z INFO  librespot_core::session] Authenticated as "jgabriel98" !
nov 30 00:26:07 raspberrypi librespot[17310]: [2021-11-30T03:26:07Z INFO  librespot_playback::mixer::softmixer] Mixing with softvol and volume control: Linear
nov 30 00:26:07 raspberrypi librespot[17310]: [2021-11-30T03:26:07Z INFO  librespot_playback::convert] Converting with ditherer: tpdf
nov 30 00:26:07 raspberrypi librespot[17310]: [2021-11-30T03:26:07Z INFO  librespot_playback::audio_backend::pipe] Using pipe sink with format: S16
nov 30 00:26:07 raspberrypi librespot[17310]: [2021-11-30T03:26:07Z INFO  librespot_core::session] Country: "BR"
nov 30 00:26:08 raspberrypi librespot[17310]: [2021-11-30T03:26:08Z INFO  librespot_playback::player] Loading <falkor> with Spotify URI <spotify:track:3nGyTdjyMC0xT2TA7Q4Ru4>
nov 30 00:26:11 raspberrypi librespot[17310]: [2021-11-30T03:26:11Z INFO  librespot_playback::player] <falkor> (457200 ms) loaded
<b>nov 30 00:26:15 raspberrypi librespot[17310]: [2021-11-30T03:26:15Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: No such file or directory (os error 2)
nov 30 00:26:15 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
nov 30 00:26:15 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
</b></pre></code>

 - Also, in the sames logs, i've found out that some lines in the `.service` file are not reconized by the system (running latest raspbian):
<pre><code>
pi@raspberrypi:~/spotifypiHome $ journalctl -u raspotify.service -f
-- Logs begin at Sun 2021-11-28 20:31:25 -03. --
nov 29 21:39:47 raspberrypi librespot[7953]: [2021-11-30T00:39:47Z WARN  libmdns::fsm] couldn't parse packet from 8.218.88.19:8080: packet has non-zero reserved bits
nov 29 23:56:12 raspberrypi librespot[7953]: [2021-11-30T02:56:12Z INFO  librespot_playback::player] Loading <falkor> with Spotify URI <spotify:track:3nGyTdjyMC0xT2TA7Q4Ru4>
nov 29 23:56:15 raspberrypi librespot[7953]: [2021-11-30T02:56:15Z WARN  librespot_playback::player] Player::play called from invalid state
nov 29 23:56:15 raspberrypi librespot[7953]: [2021-11-30T02:56:15Z INFO  librespot_playback::player] <falkor> (457200 ms) loaded
<b>nov 30 00:12:40 raspberrypi systemd[1]: /lib/systemd/system/raspotify.service:15: Unknown lvalue 'RestrictSUIDSGID' in section 'Service', ignoring
nov 30 00:12:40 raspberrypi systemd[1]: /lib/systemd/system/raspotify.service:18: Unknown lvalue 'ProtectHostname' in section 'Service', ignoring
nov 30 00:12:40 raspberrypi systemd[1]: /lib/systemd/system/raspotify.service:20: Unknown lvalue 'ProtectKernelLogs' in section 'Service', ignoring
nov 30 00:12:40 raspberrypi systemd[1]: /lib/systemd/system/raspotify.service:23: Unknown lvalue 'ProtectProc' in section 'Service', ignoring
nov 30 00:12:40 raspberrypi systemd[1]: /lib/systemd/system/raspotify.service:26: Unknown lvalue 'ProtectClock' in section 'Service', ignoring
nov 30 00:12:40 raspberrypi systemd[1]: /lib/systemd/system/raspotify.service:37: Unknown lvalue 'ProcSubset' in section 'Service', ignoring</b>
</pre></code>
it apear only when the service is loaded, so running `systemctl daemon-reload` should make it appear

## system info:
```
pi@raspberrypi:~/spotifypiHome $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

closed 2021-11-30 by github-actions[bot]

JasonLG1979, 2021-11-30:
`/tmp` is a sewer. You should not use the common system `/tmp` for IPC. Many systems will not even allow services access to `/tmp`.

Raspotify has no access to `/tmp` by design.

You should put `fifo`'s under `/var/run` as per the [wiki](https://github.com/dtcooper/raspotify/wiki/Troubleshooting)

See a [relevant issue in the snapcast repo](https://github.com/badaix/snapcast/issues/737).  

> Also, in the sames logs, i've found out that some lines in the .service file are not reconized by the system (running latest raspbian):

Buster is not the latest version.

jgabriel98, 2021-11-30:
Oh, didn't know that the wiki was updated with this info about fifo under `/var/run` (it's been a long time that i've read through this repo docs).

But thank you so much, will test this as soon as possible.