dtcooper/raspotify #472: there is no /etc/raspotify/conf

pjgoodall, 2021-12-03:
The configuration file is missing from the documented location.
```
root@spotconnect:/# ls /etc/raspotify
ls: cannot access '/etc/raspotify': No such file or directory
```

there is`/etc/default/raspotify`

```
$ ls -l  /etc/default/raspotify
-rw------- 1 root root 2024 Dec  3 15:37 /etc/default/raspotify
```

pjgoodall, 2021-12-03:
fumbled an apt update it seems...

closed 2021-12-03 by pjgoodall

JasonLG1979, 2021-12-03:
That would be because 0.31.4 has not been released yet.

pjgoodall, 2021-12-03:
Yes I’d did eventually work that out. My years old install failed on a
recent upgrade. I saw your not and I latched into it :-)
 But I believe it was als changed. Fixed the config there.
Cheers…

JasonLG1979, 2021-12-03:
> Yes I’d did eventually work that out. My years old install failed on a recent upgrade. I saw your not and I latched into it :-) But I believe it was als changed. Fixed the config there. Cheers…
> […](#)

It's all good now you know. You're a step ahead of the game for when it is released.