dtcooper/raspotify #474: No Sound After Update

ljacobo, 2021-12-06:
Hi,

Having some issues after I update the raspberry pi 4.

raspotify.service - Raspotify (Spotify Connect Client)
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-12-06 14:53:21 AST; 17s ago
     Docs: https://github.com/dtcooper/raspotify
           https://github.com/librespot-org/librespot
           https://github.com/dtcooper/raspotify/wiki
           https://github.com/librespot-org/librespot/wiki/Options
 Main PID: 24413 (librespot)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/raspotify.service
           └─24413 /usr/bin/librespot --name xDv2 --device-type speaker --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --volume-ctrl linear --initial-volume 100 --username luisj01

Dec 06 14:53:22 raspberrypi librespot[24413]: [2021-12-06T18:53:22Z INFO  librespot_core::session] Connecting to AP "guc3-accesspoint-a-fwl0.ap.spotify.com:4070"
Dec 06 14:53:22 raspberrypi librespot[24413]: [2021-12-06T18:53:22Z INFO  librespot_core::session] Authenticated as "<user cleared>" !
Dec 06 14:53:22 raspberrypi librespot[24413]: [2021-12-06T18:53:22Z INFO  librespot_playback::mixer::softmixer] Mixing with softvol and volume control: Linear
Dec 06 14:53:22 raspberrypi librespot[24413]: [2021-12-06T18:53:22Z INFO  librespot_playback::convert] Converting with ditherer: tpdf
Dec 06 14:53:22 raspberrypi librespot[24413]: [2021-12-06T18:53:22Z INFO  librespot_playback::audio_backend::alsa] Using AlsaSink with format: S16
Dec 06 14:53:22 raspberrypi librespot[24413]: [2021-12-06T18:53:22Z INFO  librespot_core::session] Country: "DO"
Dec 06 14:53:34 raspberrypi librespot[24413]: [2021-12-06T18:53:34Z INFO  librespot_playback::player] Loading <Sunset Lover> with Spotify URI <spotify:track:0hNduWmlWmEmuwEFcYvRu1>
Dec 06 14:53:36 raspberrypi librespot[24413]: [2021-12-06T18:53:36Z INFO  librespot_playback::player] <Sunset Lover> (237792 ms) loaded
Dec 06 14:53:36 raspberrypi librespot[24413]: Failed to get home directory.
Dec 06 14:53:36 raspberrypi librespot[24413]: E: [pulseaudio] caps.c: Assertion 'setresuid(uid, uid, uid) >= 0' failed at daemon/caps.c:57, function pa_drop_root(). Aborting.

JasonLG1979, 2021-12-06:
Updated what? Just Raspotify? Your OS?

It looks you're using PulseAudio which isn't supported and I'm not really sure where `Failed to get home directory.` is coming from since it doesn't appear to coming from `librespot` since it doesn't have the `[2021-12-06T18:53:36Z INFO ` in front of it. It looks like you're getting error leaking in from PulseAudio though the alsa bindings.

JasonLG1979, 2021-12-06:
And please wrap terminal output in back-ticks. Otherwise github can end up doing strange things like referencing issues if you don't wrap `#6` for example.

 https://docs.github.com/en/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax

ljacobo, 2021-12-06:
> Updated what? Just Raspotify? Your OS?
> 
> It looks you're using PulseAudio which isn't supported and I'm not really sure where `Failed to get home directory.` is coming from since it doesn't appear to coming from `librespot` since it doesn't have the `[2021-12-06T18:53:36Z INFO ` in front of it. It looks like you're getting errors leaking in from PulseAudio though the alsa bindings.

Sorry, after update raspberry OS.

Then already tried purging and re-installing Raspotify with no luck.

Should I remove PulseAudio, because everything was working perfectly until the update.

Thanks.

JasonLG1979, 2021-12-06:
I would start by backing up your config and then doing `sudo apt purge raspotify` and then `sudo apt install raspotify`. At that point Raspotify should be like a fresh install. Then you can reconfigure it with your backed up config.

That's still not going to help you run Raspotify with Pulseaudio though. You'll have to figure that part out on your own. Others have done it. You may ask @kirenida how they have Raspotify and PulseAudio configured to play well together, and if you manage to get it working by all means please update the wiki but it's not something that officially supported.

closed 2021-12-06 by JasonLG1979

JasonLG1979, 2021-12-06:
> Should I remove PulseAudio, because everything was working perfectly until the update.

As I said @kirenida verified that it is possible for Raspotify and PulseAudio to play well together you'll have to ask them.

As far as if you should remove PulseAudio that really depends on what else you have installed that depends on PulseAudio. If you're running the desktop version of Raspoberry Pi OS then uninstalling PulseAudio will probably break a lot of things. On the other hand if you're using your Pi headlessly and nothing else needs PulseAudio there's no reason to have in installed.

Generally if you NEED PulseAudio you'd be better off with Spotifyd if you're a novice.

kirenida, 2021-12-07:
@ljacobo this looks like a system configuration issue, I don't think I've ever encountered something like this. Did you ceck if anybody has a similar issue on the RaspberryOS forums?

Have you tried checking if the pulse and pi users are in the audio and / or pulse-access groups?

@JasonLG1979 I'll try to get the pulseaudio + raspotify "guide" up on the wiki soon, life got in my way.

JasonLG1979, 2021-12-07:
> @JasonLG1979 I'll try to get the pulseaudio + raspotify "guide" up on the wiki soon, life got in my way.

Awesome thanks!!! Although it's ofc not supported out of the box it will be nice to have a documented way to do it as it seems to be a really popular question.