dtcooper/raspotify #475: Raspotify crashes with Alsa equalizer

marcelmu, 2021-12-13:
I have setup alsa with an equalizer according to this comment:
https://github.com/dtcooper/raspotify/issues/23#issuecomment-476951906

This has been working for quite some time. 
The other day I upgraded to 0.31.3 and after that audio playback crashes with the following error:

```
Dec 13 08:50:10 raspberrypi librespot[12520]: [2021-12-13T08:50:10Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device equal May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'EPERM: Operation not permitted'
Dec 13 08:50:10 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Dec 13 08:50:10 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
```

However running the librespot command standalone does not produce the same error. Playback works in that case.

`/usr/bin/librespot --name raspotify --device-type speaker --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --volume-ctrl log --initial-volume=30 --device equal`

Even without the "--device equal" setting, the crash occurs.

jongdesteven, 2021-12-13:
I have the same issue, manually running librespot does work too.

JasonLG1979, 2021-12-13:
What Raspberry Pi OS version?

marcelmu, 2021-12-13:
I have 5.10.63-v7 installed

JasonLG1979, 2021-12-13:
I'm not asking the kernel version.

marcelmu, 2021-12-13:
Sorry.

```
~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
```

You meant this?

JasonLG1979, 2021-12-14:
After looking in the pi user's home it seems that the plugin probably wants to put a `.alsaequal.bin` in the home of the user that's using the plugin. The docs say as much. That ofc isn't going to work with a user without a home.

I can get it to work by putting `.alsaequal.bin` in a place that is visible to the raspotify service and telling the plugin where it's at in my `/etc/asound.conf`. Something like this works with the latest git version of raspotify:

```
pcm.!default {
    type plug
    slave.pcm plugequal
}
ctl.!default {
    type hw card 0
}
ctl.equal {
    type equal
    controls /var/lib/raspotify/.alsaequal.bin
}
pcm.plugequal {
    type equal
    controls /var/lib/raspotify/.alsaequal.bin
    slave.pcm "plughw:0,0"
}
pcm.equal {
    type plug
    slave.pcm plugequal
}
```

The problem ofc with this is that you have to user sudo to change the eq values and nothing else can read `/var/lib/raspotify/.alsaequal.bin`. The previous way that the plugin worked by using `$HOME` may have worked when the plugin was 1st written (and may continue to work for real human users) but it will break for more and more system services as they adopt systemd features like dynamic users.

JasonLG1979, 2021-12-14:
Looking at the [deb package source repo](https://salsa.debian.org/multimedia-team/alsaequal) the plugin itself hasn't been updated in 11 years. I would not even know who upstream of that actually is?

Ignatz95, 2021-12-31:
> I have setup alsa with an equalizer according to this comment: [#23 (comment)](https://github.com/dtcooper/raspotify/issues/23#issuecomment-476951906)
> 
> This has been working for quite some time. The other day I upgraded to 0.31.3 and after that audio playback crashes with the following error:
> 
> ```
> Dec 13 08:50:10 raspberrypi librespot[12520]: [2021-12-13T08:50:10Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device equal May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'EPERM: Operation not permitted'
> Dec 13 08:50:10 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
> Dec 13 08:50:10 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
> ```
> 
> However running the librespot command standalone does not produce the same error. Playback works in that case.
> 
> `/usr/bin/librespot --name raspotify --device-type speaker --backend alsa --bitrate 320 --disable-audio-cache --enable-volume-normalisation --volume-ctrl log --initial-volume=30 --device equal`
> 
> Even without the "--device equal" setting, the crash occurs.

I have the exact same problem with my installation of version 0.31.4 on my RaspberryPi 3B+ running OS Raspbian GNU/Linux 10 (buster). I'm also using a hifiberry sound card on top of the Pi.

I found another workaround for the problem, just in case somebody still has this problem:

If you enter ```aplay -L``` into the terminal, then you'll get a list of every available audio sink device. Here my output:
```
pi@spotipi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
default
    Playback/recording through the PulseAudio sound server
plugequal
equal
sysdefault:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dacplus, HiFiBerry DAC+ HiFi pcm512x-hifi-0
    Default Audio Device
dmix:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus, HiFiBerry DAC+ HiFi pcm512x-hifi-0
    Direct sample mixing device
dsnoop:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus, HiFiBerry DAC+ HiFi pcm512x-hifi-0
    Direct sample snooping device
hw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus, HiFiBerry DAC+ HiFi pcm512x-hifi-0
    Direct hardware device without any conversions
plughw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dacplus, HiFiBerry DAC+ HiFi pcm512x-hifi-0
    Hardware device with all software conversions
usbstream:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dacplus
    USB Stream Output
```

Now, what I did was the following:
I copied the expression for my hifiberry sound card ```sysdefault:CARD=sndrpihifiberry``` and entered it into the ```/etc/raspotify/conf``` file:
```
# Audio device to use, use `librespot --device ?` to list options.
# Defaults to the system's default.
LIBRESPOT_DEVICE="sysdefault:CARD=sndrpihifiberry"
```
Then I restarted the raspotify systemd service and everything was working fine again.

Maybe this workaround could also work if you are just using the standard audio jack as audio output to the speakers. In this case edit the raspotify configuration file with ```LIBRESPOT_DEVICE="jack"```. Maybe this helps.

JasonLG1979, 2021-12-31:
> Maybe this workaround could also work if you are just using the standard audio jack as audio output to the speakers. In this case edit the raspotify configuration file with `LIBRESPOT_DEVICE="jack"`. Maybe this helps.

I don't know what you did but `"jack"` doesn't mean the analog audio output jack it means [JACK Audio Connection Kit](https://jackaudio.org/)

michaelfeil, 2022-08-10:
Jason I can confirm, I had a similar problem with Alsa and Hifiberry DAC+ on a propably related issue with 
Raspbian 11 Bullseye. 

```
Aug 10 22:35:51 raspberrypi librespot[2429]: [2022-08-10T20:35:51Z ERROR librespot_playback::player] Audio Sink Error Invalid Parameters: <AlsaSink> Device default Unsupported Format S16LE (S16), ALSA function 'snd_pcm_hw_params_set_format' failed with error 'EINVAL: Invalid argument'
```
librespot worked out of the box. 
I replaced it with the name as listed under `librespot --device ?`, where also the following worked:
```
# Audio device to use, use `librespot --device ?` to list options.
# Defaults to the system's default.
LIBRESPOT_DEVICE="hw:CARD=sndrpihifiberry"
```
Maybe it helps getting redirected to this error message.

JasonLG1979, 2022-08-10:
No. 

1. This not the only reason that error can occur.
2. That error originates in librespot not in Raspotify.
3. The offending eq plugin is unmaintained so this bug is invalid.