dtcooper/raspotify #476: Trigger script when not playing after x seconds

microfx, 2021-12-14:
Is it possible? Is there a function? Using shairport-sync (Airplay implementation) which can do it so well - only when using Raspotify my speakers stay on as long as I don't disconnect manually from Spotify Connect. A timeout that triggered a script (to turn off my wallplug where the speakers are connected to) would help me reduce my power consumption – getting expensive nowadays!

Cheers

JasonLG1979, 2021-12-14:
#455

closed 2021-12-14 by github-actions[bot]

microfx, 2021-12-14:
It's too complicated ... open up! 😅

microfx, 2021-12-14:
> #455

I really don't understand why I need a degree in computer science ... where as in sharepoint it's only adding a script (without if / else what not conditions) to the config ... the issue is about exactly this: Ease of use.

JasonLG1979, 2021-12-14:
> It's too complicated ... open up! 😅

There are literally code examples in #455. Figure it out.

microfx, 2021-12-14:
I did! It doesn't work. I mean ... as said in the 1st posting ... my speakers turn off. But not after a timeout. It's too complicated for the rest of us. Please take that into account.

JasonLG1979, 2021-12-14:
> > #455
> 
> I really don't understand why I need a degree in computer science ... where as in sharepoint it's only adding a script (without if / else what not conditions) to the config ... the issue is about exactly this: Ease of use.

Raspotify is nothing more than a packaged version of librespot which itself is nothing but a collection of libs designed to be used to create applications that happen to function as a "batteries not included" Spotify Connect client, all of which are given out for free. So beggars can't be choosers if you can't be bothered to roll up your sleeves and put in a little bit of effort.

JasonLG1979, 2021-12-14:
> I did! It doesn't work. I mean ... as said in the 1st posting ... my speakers turn off. But not after a timeout. It's too complicated for the rest of us. Please take that into account.
> 
> Those code examples read like they're describing the matrix. I don't get any of that.

The code examples are primary school level Python. My 9yr old could have wrote those?

JasonLG1979, 2021-12-14:
You're acting as though you are owed this functionality?

That's not how Open Source software works. If the functionality you desire is missing it is up to you to contribute to see that the functionality is added. Either by adding the functionality yourself or convincing someone else to. Complaining that a free thing doesn't do what you want is not contributing.

microfx, 2021-12-14:
> The code examples are primary school level Python. My 9yr old could have wrote those?

So... what are you trying to say? I'm dumber than your 9yr old? Is this what it's about? Aight. Thanks for the comment then! Very productive telling this other people having problem using a tool with whom they are having trouble with.

JasonLG1979, 2021-12-14:
> So... what are you trying to say? I'm dumber than your 9yr old? Is this what it's about? Aight. Thanks for the comment then! Very productive telling this other people having problem using a tool [...]

No what I'm saying is that it's your job to figure it out and not my job to explain it to you. You have been given a very good starting point. It is now up to you to decide if it is worth your effort to continue.

microfx, 2021-12-14:
And why is it not designed to be user friendly? To tell people they are expected to have more knowledge?

JasonLG1979, 2021-12-14:
I already explained that. Raspotify is just a package. (a compiled binary and systemd service file) the binary is librespot. Librespot is itself a collection of libraries for programmers to use create apps that interact with Spotify's API. (Think programmer legos) the binary you get if you compile them all together is a minimal example program. Librespot is very much roll your own, batteries not included. It's aimed at developers because it's a collection of development libraries.

JasonLG1979, 2021-12-14:
```python3
import os

if os.getenv('PLAYER_EVENT') in ('playing', 'preloading', 'started'):
    # librespot sent a playing, preloading or started event.
    # Do whatever to turn your speakers on...

elif os.getenv('PLAYER_EVENT') in ('stopped', 'paused'):
    # librespot sent a stopped or pause event.
    # Do whatever to turn your speakers off...
```

JasonLG1979, 2021-12-14:
@microfx if you have actual legitimate questions beyond "Why doesn't this free thing do exactly what I want?" and "Why do you expect me to learn things?" Feel free to join the discussion in #455. But be warned that if you continue to spam this repo with complaints you will be blocked.