dtcooper/raspotify #478: 'ENODEV: No such device' error

PaoloRondot, 2021-12-18:
Hello,
I already tried to find answers on the other issues, especially the one starting with "I'm very new to this", but I'm still struggling. :-)

Here's my situation. I need my raspberry to be able to playback music coming from four different sources: Jack input, Bluetooth, Spotify and PI's memory, and without a PI's desktop setup, so only through SSH and command line.
I've already setup the bluetooth and the Jack audio IN, using an Hifiberry DAC+ADC Pro shield and PulseAudio. Now I am trying to setup Spotify part. I installed raspotify with the latest version.
I got -without really understanding- that PulseAudio is not really suited for Raspotify. Yet it is really handy for me, as I have to deal with multiple sources and mix them.

And here's my issue. I can find the raspberry on spotify connect, on my phone but I can't connect to it, and when I look at `/var/log/daemon.log`, I see the following error : 
`Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'ENODEV: No such device'`

Do you have any idea on how I can fix this ?
Also, do you think that I should take my problem differently or use other tools ?

Any help will be very appreciated, I'm a newbie very motivated to learn from community ! Thanks !👍

JasonLG1979, 2021-12-18:
The wiki has a user contributed page on Pulseaudio that might be of some help.

kirenida, 2021-12-19:
@PaoloRondot are you running PulseAudio as a system service? Then try what I wrote on the wiki and let me know if it works.
If you are not running it as a system service, maybe try adding the `auth-cookie-enabled=0 auth-anonymous=1` option to  `load-module module-native-protocol-unix` in **/etc/pulse/default.pa**?

I don't know how DAC+DSP shows up in ALSA, but have a look at the output of `aplay -l` and if multiple cards show up, try setting up Raspotify to use the correct one.

Whatever you do, let me know if you get it working so I can update the PulseAudio wiki page here.

PaoloRondot, 2021-12-19:
@kirenida I am not running PA in system mode, but I'm going to give it a try, following the wiki page. I keep you updated ! 
Tough, I can tell you what `aplay -l` gives me : 
```
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sndrpihifiberry [snd_rpi_hifiberry_dacplusadcpro], device 0: HiFiBerry DAC+ADC Pro HiFi multicodec-0 [HiFiBerry DAC+ADC Pro HiFi multicodec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
And I've added `dtoverlay=hifiberry-dacplusadcpro` and `force_eeprom_read=0` to `/boot/config.txt`, as told on Hifiberry guide. 
I also added the following to `/etc/asound.rc`:
```
pcm.!default {
type pulse
}

ctl.!default {
type pulse
}
```
I get the audio out of the shield properly with Bluetooth and Jack in. It is just raspotify that I can't get to work.

Thanks a lot for your answer , I keep you updated.

PaoloRondot, 2021-12-19:
I have the feeling that it is because of a conflict with bluetooth. If I turn off bluetooth on the raspberry, then spotify connect works,  but once I turn it on again, if I connect to raspberry's spotify connect, as soon as I try to play a music, I get disconnected from spotify connect. Does this info help ?

kirenida, 2021-12-20:
@PaoloRondot have you tried passing the correct ALSA device to librespot through the Raspotify config file? I think it should be hw:2,0 or something like that, from your `aplay -l`

I think you should also be able to see available devices with `librespot device ?`

I don't think I can help you more that this, as I haven't used PulseAudio in that kind of configuration. Maybe you need to find an option to tell Pulse to "free up" a sink or a card when it's not in use?

JasonLG1979, 2021-12-20:
@PaoloRondot a link to the Hifiberry guide would be very helpful as I suspect that it's probably either outdated or just plain wrong as the system wide alsa config file should be `/etc/asound.conf` not `/etc/asound.rc` for starters.

PaoloRondot, 2021-12-22:
Sorry for taking time to answer..!
@JasonLG1979, the guide I followed was indeed indicating that the alsa config file was `/etc/asound.conf`, but I found on other forums to configure `asound.rc`, to get Raspotify to work with PA. I probably mixed things.. Should I then configure both, or just keep the .conf one ?
@kirenida, How should I proceed to pass the correct ALSA device to librespot via Raspotify config file ?
I also ran the `librespot device ?` command but I don't really know what to do with the answer:
```
[2021-12-22T10:07:45Z INFO  librespot] librespot 0.3.1 bbd575e (Built on 2021-11-26, Build ID: a6e0Ery3, Profile: release)
[2021-12-22T10:07:45Z WARN  libmdns::fsm] dropping truncated packet from 192.168.1.1:5353
[2021-12-22T10:07:45Z WARN  libmdns::fsm] dropping truncated packet from 192.168.1.1:5353
[2021-12-22T10:07:45Z WARN  libmdns::fsm] dropping truncated packet from 192.168.1.1:5353
[2021-12-22T10:07:45Z WARN  libmdns::fsm] dropping truncated packet from 192.168.1.1:5353
```
I'm gonna keep investigating, and keep you updated. Thanks a lot for your suggestions :-)

PaoloRondot, 2021-12-22:
@kirenida, I've set in raspotify config file this:
`OPTIONS="--device hw:2,0"`
I can now get spotify to work, but as soon as I connect the bluetooth, I get this in the daemon log:
`Error Connection Refused: <AlsaSink> Device hw:2,0 May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'`
And when I disconnect bluetooth, it starts working again.
Do you know if in this `OPTION` parameter of the config file, I can add something to tell Raspotify to use PulseAudio as a mixer or something ?

PaoloRondot, 2021-12-22:
I just tried setting `OPTIONS="--device pulse"` but I get:
```
Dec 22 10:38:52 raspberrypi librespot[1302]: ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused
Dec 22 10:38:52 raspberrypi librespot[1302]: [2021-12-22T10:38:52Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device pulse May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'ECONNREFUSED: Connection refused'
```
Maybe if I can correct this `'ECONNREFUSED: Connection refused'` error, I could solve the problem ?

JasonLG1979, 2021-12-22:
You're all over the place here. The guide, where's that link I ask for. We can't possibly know what's going on until we can decipher what you've done.

PaoloRondot, 2021-12-22:
Alright ! Just a reminder of my goal: being able to get at the same time, sound coming from Bluetooth, Jack in and spotify. Now here are the guides I followed:
**The Hifiberry guide**: https://www.hifiberry.com/docs/software/configuring-linux-3-18-x/
**The guide I've followed for Bluetooth**: https://forums.raspberrypi.com/viewtopic.php?t=235519
**The guide I've followed for Raspotify**: https://mathieu-requillart.medium.com/my-ultimate-guide-to-the-raspberry-pi-audio-server-i-wanted-spotify-ce549656af06

These three guides are the bases that gave me the following result:
- Playing at the same time sounds coming from Bluetooth and Jack input, and both using PulseAudio. 👍 
- Have an error when I try to play music from spotify, that I described on my first message of this issue: `Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'ENODEV: No such device'` 👎 

I tried to add `auth-anonymous=1` option to `load-module module-native-protocol-unix` in /etc/pulse/default.pa, following @kirenida's first suggestion. This gave the same result as before.
I also tried playing with the `asound.rc` and `asound.conf` files. In the Hifiberry guide, it's written to configure `asound.conf`, and in the Raspotify guide, the `asound.rc`. As a result, I now have these two files, with `asound.conf` like:
```
pcm.!default {
type hw card 2
}
ctl.!default {
type hw card 2
}
```
and `asound.rc` like:
```
pcm.!default {
type pulse
}
ctl.!default {
type pulse
}
```
These two files didn't help me correct my raspotify error. I admit that I don't really know how they work and how they are used by the system.

Then Following @kirenida' second advice, I tried to pass the ALSA device to librespot via Raspotify config file by adding the following in this file: `OPTIONS="--device hw:2,0"`. This new configuration allowed me to have another result:
When I only have spotify playing on the PI, it plays like a charm: the first error I opened this issue for, is solved 👍. But when I try to play simultaneoulsy a sound from another source, raspotify stops with this new error: 
`Dec 22 16:00:12 raspberrypi librespot[757]: [2021-12-22T16:00:12Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device hw:2,0 May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'` 👎 
Moreover, when only spotify is playing, I don't see raspotify in PulseAudio's sink-inputs list, showing that it bypasses PulseAudio which, I believe, then creates a conflict between Alsa and PA and give the error above.

Therefore, I thought that I could maybe set in the raspotify config file `OPTIONS="--device pulse"` instead of `OPTIONS="--device hw:2,0"`, to force it to use PA. But it didn't work, giving the error `'ECONNREFUSED: Connection refused'` that I describe in the last comment.

Voila. That's my "thread of thoughts". I believe that this problem must be very simple, I hope that this long comment will help clarify everything. Thanks again for your time!

JasonLG1979, 2021-12-22:
The HifiBerry guide is OK but the other 2 guides are not compatible. The bluetooth guide runs everything as the pi user which isn't compatible with a system level Raspotify service. The Raspotify guide is basically garbage, that's the nicest thing I can say about it. Neither of those guides together is ever going to get you what you want.

Walking you though the right way to do it is way outside the scope of Raspotify.

Your best bet is to checkout something like [rpi-audio-receiver](https://github.com/nicokaiser/rpi-audio-receiver)

closed 2021-12-22 by JasonLG1979

JasonLG1979, 2021-12-22:
I put in a [PR](https://github.com/nicokaiser/rpi-audio-receiver/pull/115) to make sure Raspotify works with [rpi-audio-receiver](https://github.com/nicokaiser/rpi-audio-receiver) assuming everything else works also I have not checked. That's neither here nor there for me. I just fixed the Raspotify bit. Now it's between you and them.

PaoloRondot, 2022-01-04:
Hello there,
Just wanted to say that I got my things to work with spotifyd.
Thanks again for your help and happy new year.