dtcooper/raspotify #479: Unable to get raspotify running after last update..

Anka001, 2021-12-18:
I am unable to get raspotify up and running after updating to 0.31.4. 

I uninstalled and reinstalled ('sudo apt purge raspotify' and 'sudo apt install raspotify'). I moved configuration to /etc/raspotify/conf, and tried to edit according to the manuals.
Whatever I try, I always end up with 'systemd[1]: raspotify.service: Failed with result 'exit-code'.'

Is there a default '/etc/raspotify/conf', that works out of the box, as a base for further experiments?
Can somebody please point me in the right direction?

Andreas

JasonLG1979, 2021-12-18:
> I moved configuration to /etc/raspotify/conf, and tried to edit according to the manuals.

The [migration page](https://github.com/dtcooper/raspotify/wiki/Migration-from-older-versions-to-0.31.4-and-beyond) does not say to move the old config to `/etc/raspotify/conf`. That clearly will not work.

> Whatever I try, I always end up with 'systemd[1]: raspotify.service: Failed with result 'exit-code'.'

What is the error?

> Is there a default '/etc/raspotify/conf', that works out of the box, as a base for further experiments?

Yes, [here](https://github.com/dtcooper/raspotify/blob/master/raspotify/etc/raspotify/conf) and [here](https://github.com/dtcooper/raspotify/wiki/Configuration) although I'm not sure what you mean by "experiments"?

JasonLG1979, 2021-12-18:
As explained in the [migration page](https://github.com/dtcooper/raspotify/wiki/Migration-from-older-versions-to-0.31.4-and-beyond) Each line in `/etc/raspotify/conf` corresponds to an environment variable which in turn corresponds to a [librespot option/flag](https://github.com/librespot-org/librespot/wiki/Options)

> /etc/raspotify/conf will contain a non-exhaustive list of librespot options and flags. If you don't see the option or flag that you normally use you can add it and so long as it's valid it will work. To avoid name collisions environment variables must be prepended with LIBRESPOT_, so option/flag foo-bar becomes LIBRESPOT_FOO_BAR. If you look at the rest of /etc/raspotify/conf it should not be difficult to understand.

kriation, 2021-12-20:
@Anka001, can you run the following and share the result?
```/usr/bin/librespot -V```

@JasonLG1979, I suspect @Anka001 is still running 0.3.1 as the gh-pages branch is not updated with the 0.31.4 package (as of this morning). When you have a chance, can you re-build the apt repository with the latest?

williamthrilliam, 2021-12-20:
I did a apt upgrade to the latest buster version and I too do not have a working spotify connect service now. Purging and installing I get the following output: 
`pi@raspberrypi:~ $ sudo apt install raspotify
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  raspotify
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/2,656 kB of archives.
After this operation, 0 B of additional disk space will be used.
Selecting previously unselected package raspotify.
(Reading database ... 40360 files and directories currently installed.)
Preparing to unpack .../raspotify_0.31.3~librespot.v0.3.1-19-gbbd575e_armhf.deb ...
Unpacking raspotify (0.31.3~librespot.v0.3.1-19-gbbd575e) ...
Setting up raspotify (0.31.3~librespot.v0.3.1-19-gbbd575e) ...
Created symlink /etc/systemd/system/multi-user.target.wants/raspotify.service → /lib/systemd/system/raspotify.service.`

Looks like I'm still on 0.3.1. Does that mean buster isn't supported?

kriation, 2021-12-20:
@williamthrilliam, the reason we're all still seeing 0.31.3 when pulling from the repository is because 0.31.4 isn't there yet.

JasonLG1979, 2021-12-20:
> @Anka001, can you run the following and share the result? `/usr/bin/librespot -V`
> 
> @JasonLG1979, I suspect @Anka001 is still running 0.3.1 as the gh-pages branch is not updated with the 0.31.4 package (as of this morning). When you have a chance, can you re-build the apt repository with the latest?

Very possibly. With the holiday and all @dtcooper has understandably had more important things to do than update the .deb repo right away. 

> I did a apt upgrade to the latest buster version and I too do not have a working spotify connect service now. Purging and installing I get the following output: `pi@raspberrypi:~ $ sudo apt install [...]

See the above and also see the read me. Buster is no longer supported. The repo itself isn't versioned at all so no matter what OS version you're running you'll get the latest version that's in the repo, so if you're running a unsupported OS/OS version the package could very well break. If you plan on running an old OS version it's best to not enable the repo but instead just install a working .deb.

So for Buster the last version that works is:

https://github.com/dtcooper/raspotify/releases/tag/0.16.0

dtcooper, 2021-12-20:
Actually working this as we speak. [Reprepro](https://wiki.debian.org/DebianRepository/SetupWithReprepro) appears to be broken, so once I have a fix for that it'll be deployed.

dtcooper, 2021-12-20:
Deployed!

JasonLG1979, 2021-12-20:
Thanks @dtcooper!!!

closed 2021-12-20 by JasonLG1979

Anka001, 2021-12-21:
> librespot 0.3.1 e5fd7d6 (Built on 2021-12-16, Build ID: 9fJyj1nn, Profile:
> release)
>
>
>
> How can I make sure I am running the correct build of raspotify I am
> running?
>
>
>
> Andreas
>
>
>
> update.. (Issue #479)

Original issue has been closed, I opened #481:

Issue still not resolved.

<https://github.com/dtcooper/raspotify/issues/479#issuecomment-997983957> ,
457XANCNFSM5KLCDS4A> .
t=8&pt=524675>  or Android
tm_campaign%3Dnotification-email%26utm_medium%3Demail%26utm_source%3Dgithub>