dtcooper/raspotify #480: Can't find config file

Faewks, 2021-12-19:
After flashing a new raspberrypi os lite and installing raspotify the "easy way" on a raspberrypi zero W I can't seem to find the config file. 
[image]

Following the wiki the directory "raspotify" should be under `/etc/` and the config should be under `/etc/raspotify/conf` .
Do I have to create the directory and file?
I'm a little bit puzzled right now.

Hope you can help me.

Regards,
Faewks.

JasonLG1979, 2021-12-19:
That is because @dtcooper has not pushed the update to the .deb repo yet. Being as though it is the holiday season there might be a small but understandable delay. You can install the latest version ahead of it's push to the repo in the meantime.

Download the 0.31.4 .deb:
```
curl -O https://github.com/dtcooper/raspotify/releases/download/0.31.4/raspotify_0.31.4.librespot.v0.3.1-34-ge5fd7d6_armhf.deb
```

And then install it:
```
sudo apt install -y ./raspotify_0.31.4.librespot.v0.3.1-34-ge5fd7d6_armhf.deb
```

closed 2021-12-19 by JasonLG1979

dtcooper, 2021-12-19:
I'll be released the latest version soon. Thanks for your patience all.

JasonLG1979, 2021-12-19:
It's all good. Enjoy the holidays.

Faewks, 2021-12-19:
Thank you for your answers and fast informations! 
I don't mind waiting! I was just confused that after following everything from the installation advice I could not find the config.

Have a nice holliday!
Regards,
Faewks.