dtcooper/raspotify #481: Raspotify not running after last update

Anka001, 2021-12-21:
Sorry for being annoying, but you closed the issue, and I am still lost. Sorry for not being a linux nerd, but a guy who just wants to run raspotify. I was able to install a year ago, and everything worked fine until the last update.

Installed the update that was deployed on monday. No change.

Should raspotify start with the provided /etc/raspotify/conf? I just edited the credentials username/password, and left everything else unchanged. 
Result is:

pi@Timor:/etc/raspotify $ sudo systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2021-12-21 20:01:33 CET; 3s ago
     Docs: https://github.com/dtcooper/raspotify
           https://github.com/librespot-org/librespot
           https://github.com/dtcooper/raspotify/wiki
           https://github.com/librespot-org/librespot/wiki/Options
  Process: 1086 ExecStart=/usr/bin/librespot (code=exited, status=127)
 Main PID: 1086 (code=exited, status=127)

Dec 21 20:01:33 Timor systemd[1]: raspotify.service: Failed with result 'exit-code'.

Any help appreciated
Andreas

JasonLG1979, 2021-12-21:
> Sorry for being annoying, but you closed the issue, and I am still lost. Sorry for not being a linux nerd, but a guy who just wants to run raspotify. I was able to install a year ago, and everything worked fine until the last update.

> Should raspotify start with the provided /etc/raspotify/conf?

You should follow the migration guide.

> Should raspotify start with the provided /etc/raspotify/conf? I just edited the credentials username/password, and left everything else unchanged

You don't need to enter your credentials unless you have also disabled discovery.

> Result is:

```
pi@Timor:/etc/raspotify $ sudo systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
Active: activating (auto-restart) (Result: exit-code) since Tue 2021-12-21 20:01:33 CET; 3s ago
Docs: https://github.com/dtcooper/raspotify
https://github.com/librespot-org/librespot
https://github.com/dtcooper/raspotify/wiki
https://github.com/librespot-org/librespot/wiki/Options
Process: 1086 ExecStart=/usr/bin/librespot (code=exited, status=127)
Main PID: 1086 (code=exited, status=127)

Dec 21 20:01:33 Timor systemd[1]: raspotify.service: Failed with result 'exit-code'.
```

What OS and OS version are you using? Raspotify is only guaranteed to be compatible with Raspberry Pi OS Bullseye.

And please surround terminal output with code quotes (triple backticks).

https://docs.github.com/en/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax

Anka001, 2021-12-21:
> What OS and OS version are you using?

pi@Timor:/etc/raspotify $ lsb_release -d && uname -r
Description:    Raspbian GNU/Linux 9.13 (stretch)
4.19.66-v7+

closed 2021-12-21 by github-actions[bot]