dtcooper/raspotify #482: Raspotify service exits with code 127

NolanGarrett, 2021-12-27:
### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

After a conversion to the latest install package, I can not start raspotify as a service. As a binary, it appears to work fine.

I followed the instructions here: https://github.com/dtcooper/raspotify/wiki/Migration-from-older-versions-to-0.31.4-and-beyond

I have also attempted a manual apt purge raspotify and reinstall using the shell script, without luck. I continue to get a 127 exit code, but cannot identify where the issue lies.

### Relevant log output

```shell
Dec 27 18:17:46 raspotify systemd[1]: Started Raspotify (Spotify Connect Client).
Dec 27 18:17:46 raspotify systemd[1]: raspotify.service: Main process exited, code=exited, status=127/n/a
Dec 27 18:17:46 raspotify systemd[1]: raspotify.service: Unit entered failed state.
Dec 27 18:17:46 raspotify systemd[1]: raspotify.service: Failed with result 'exit-code'.
Dec 27 18:17:56 raspotify systemd[1]: raspotify.service: Service hold-off time over, scheduling restart.
Dec 27 18:17:56 raspotify systemd[1]: Stopped Raspotify (Spotify Connect Client).
Dec 27 18:17:56 raspotify systemd[1]: Started Raspotify (Spotify Connect Client).
Dec 27 18:17:56 raspotify systemd[1]: raspotify.service: Main process exited, code=exited, status=127/n/a
Dec 27 18:17:56 raspotify systemd[1]: raspotify.service: Unit entered failed state.
Dec 27 18:17:56 raspotify systemd[1]: raspotify.service: Failed with result 'exit-code'.
Dec 27 18:18:06 raspotify systemd[1]: raspotify.service: Service hold-off time over, scheduling restart.
Dec 27 18:18:06 raspotify systemd[1]: Stopped Raspotify (Spotify Connect Client).
Dec 27 18:18:06 raspotify systemd[1]: Started Raspotify (Spotify Connect Client).
Dec 27 18:18:06 raspotify systemd[1]: raspotify.service: Main process exited, code=exited, status=127/n/a
Dec 27 18:18:06 raspotify systemd[1]: raspotify.service: Unit entered failed state.
Dec 27 18:18:06 raspotify systemd[1]: raspotify.service: Failed with result 'exit-code'.
Dec 27 18:18:16 raspotify systemd[1]: raspotify.service: Service hold-off time over, scheduling restart.
Dec 27 18:18:16 raspotify systemd[1]: Stopped Raspotify (Spotify Connect Client).
Dec 27 18:18:16 raspotify systemd[1]: Started Raspotify (Spotify Connect Client).
Dec 27 18:18:16 raspotify systemd[1]: raspotify.service: Main process exited, code=exited, status=127/n/a
Dec 27 18:18:17 raspotify systemd[1]: raspotify.service: Unit entered failed state.
Dec 27 18:18:17 raspotify systemd[1]: raspotify.service: Failed with result 'exit-code'.
```

JasonLG1979, 2021-12-27:
My guess would be that you are not in fact running a compatible OS.

JasonLG1979, 2021-12-27:
If you search "status=127" (Due Diligence) in the issues of this repo you will find that `status=127` is the exit code that you would get if you're running on a system with an older incompatible systemd version.

JasonLG1979, 2021-12-27:
Duplicate of #456, #467 and #481.

closed 2021-12-27 by github-actions[bot]