dtcooper/raspotify #485: Raspotify exited status=127

TimHausmann, 2022-01-01:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hi, I am sorry to bother but I can't get raspotify to work. After Upgrading my Raspberry Pi A to the latest FW, raspotify stopped working. I updated to the latest raspotify version via apt and still the service crashes. I am using a Sound Blaster Play! USB Soundcard (hw:1,0). Do I need to set this as audio device in the config?

Thank you very much in advance and a happy new year to everyone.

```
pi@mypi:~ $ systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
   Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2022-01-01 11:20:10 CET; 8s ago
     Docs: https://github.com/dtcooper/raspotify
           https://github.com/librespot-org/librespot
           https://github.com/dtcooper/raspotify/wiki
           https://github.com/librespot-org/librespot/wiki/Options
  Process: 32297 ExecStart=/usr/bin/librespot (code=exited, status=127)
 Main PID: 32297 (code=exited, status=127)

Jan 01 11:20:10 odin2 systemd[1]: raspotify.service: Failed with result 'exit-code'.
```
```
pi@mypi:~ $ sudo tail  /var/log/syslog
Jan  1 11:26:17 odin2 systemd[1]: Started Raspotify (Spotify Connect Client).
Jan  1 11:26:18 odin2 systemd[1]: raspotify.service: Main process exited, code=exited, status=127/n/a
Jan  1 11:26:18 odin2 systemd[1]: raspotify.service: Unit entered failed state.
Jan  1 11:26:18 odin2 systemd[1]: raspotify.service: Failed with result 'exit-code'.
Jan  1 11:26:28 odin2 systemd[1]: raspotify.service: Service hold-off time over, scheduling restart.
Jan  1 11:26:28 odin2 systemd[1]: Stopped Raspotify (Spotify Connect Client).
Jan  1 11:26:28 odin2 systemd[1]: Started Raspotify (Spotify Connect Client).
Jan  1 11:26:28 odin2 systemd[1]: raspotify.service: Main process exited, code=exited, status=127/n/a
Jan  1 11:26:28 odin2 systemd[1]: raspotify.service: Unit entered failed state.
Jan  1 11:26:28 odin2 systemd[1]: raspotify.service: Failed with result 'exit-code'.
```
```
# /etc/raspotify/conf -- Arguments/configuration for librespot

# A non-exhaustive list of librespot options and flags.

# Please see https://github.com/dtcooper/raspotify/wiki &
# https://github.com/librespot-org/librespot/wiki/Options
# for configuration details and a full list of options and flags.

# You can also find a full list with `librespot -h`.

# To avoid name collisions environment variables must be prepended with
# `LIBRESPOT_`, so option/flag `foo-bar` becomes `LIBRESPOT_FOO_BAR`.

# Invalid environment variables will be ignored.

# Raspotify defaults may vary from librespot defaults.
# Commenting out the environment variable will fallback to librespot's default
# unless otherwise noted.

# Flags are either on (uncommented) or off (commented),
# their values are otherwise not evaluated (but the "=" is still needed).

# Only log warning and error messages.
LIBRESPOT_QUIET=

# Automatically play similar songs when your music ends.
LIBRESPOT_AUTOPLAY=

# Disable caching of the audio data.
# Enabling audio data caching can take up a lot of space
# if you don't limit the cache size with LIBRESPOT_CACHE_SIZE_LIMIT.
# It can also wear out your Micro SD card. You have been warned.
LIBRESPOT_DISABLE_AUDIO_CACHE=

# Disable caching of credentials.
# Caching of credentials is not necessary so long as
# LIBRESPOT_DISABLE_DISCOVERY is not set.
LIBRESPOT_DISABLE_CREDENTIAL_CACHE=

# Play all tracks at approximately the same apparent volume.
LIBRESPOT_ENABLE_VOLUME_NORMALISATION=

# Enable verbose log output.
#LIBRESPOT_VERBOSE=

# Disable zeroconf discovery mode.
#LIBRESPOT_DISABLE_DISCOVERY=

# Options will fallback to their defaults if commented out,
# otherwise they must have a valid value.

# Device name.
# Raspotify defaults to "raspotify (*hostname)".
# Librespot defaults to "Librespot".
LIBRESPOT_NAME="odin2"

# Bitrate (kbps) {96|160|320}. Defaults to 160.
LIBRESPOT_BITRATE="320"

# Output format {F64|F32|S32|S24|S24_3|S16}. Defaults to S16.
#LIBRESPOT_FORMAT="S16"

# Displayed device type. Defaults to speaker.
#LIBRESPOT_DEVICE_TYPE="speaker"

# Limits the size of the cache for audio files.
# It's possible to use suffixes like K, M or G, e.g. 16G for example.
# Highly advised if audio caching isn't disabled. Otherwise the cache
# size is only limited by disk space.
#LIBRESPOT_CACHE_SIZE_LIMIT=""

# Audio backend to use, alsa or pipe. Defaults to alsa.
#LIBRESPOT_BACKEND="alsa"

# Username used to sign in with.
# Credentials are not required if LIBRESPOT_DISABLE_DISCOVERY is not set.
#LIBRESPOT_USERNAME=""

# Password used to sign in with.
#LIBRESPOT_PASSWORD=""

# Enable verbose log output.
#LIBRESPOT_VERBOSE=

# Disable zeroconf discovery mode.
#LIBRESPOT_DISABLE_DISCOVERY=

# Options will fallback to their defaults if commented out,
# otherwise they must have a valid value.

# Device name.
# Raspotify defaults to "raspotify (*hostname)".
# Librespot defaults to "Librespot".
LIBRESPOT_NAME="odin2"

# Bitrate (kbps) {96|160|320}. Defaults to 160.
LIBRESPOT_BITRATE="320"

# Output format {F64|F32|S32|S24|S24_3|S16}. Defaults to S16.
#LIBRESPOT_FORMAT="S16"

# Displayed device type. Defaults to speaker.
#LIBRESPOT_DEVICE_TYPE="speaker"

# Limits the size of the cache for audio files.
# It's possible to use suffixes like K, M or G, e.g. 16G for example.
# Highly advised if audio caching isn't disabled. Otherwise the cache
# size is only limited by disk space.
#LIBRESPOT_CACHE_SIZE_LIMIT=""

# Audio backend to use, alsa or pipe. Defaults to alsa.
#LIBRESPOT_BACKEND="alsa"

# Username used to sign in with.
# Credentials are not required if LIBRESPOT_DISABLE_DISCOVERY is not set.
#LIBRESPOT_USERNAME=""

# Password used to sign in with.
#LIBRESPOT_PASSWORD=""

# Audio device to use, use `librespot --device ?` to list options.
# Defaults to the system's default.
#LIBRESPOT_DEVICE="default"

# Initial volume in % from 0 - 100.
# Defaults to 50 For the alsa mixer: the current volume.
#LIBRESPOT_INITIAL_VOLUME="50"

# Volume control scale type {cubic|fixed|linear|log}.
# Defaults to log.
#LIBRESPOT_VOLUME_CTRL="log"

# Range of the volume control (dB) from 0.0 to 100.0.
# Default for softvol: 60.0.
# For the alsa mixer: what the control supports.
#LIBRESPOT_VOLUME_RANGE="60.0"

# Pregain (dB) applied by volume normalisation from -10.0 to 10.0.
# Defaults to 0.0.
#LIBRESPOT_NORMALISATION_PREGAIN="0.0"

# Threshold (dBFS) at which point the dynamic limiter engages
# to prevent clipping from 0.0 to -10.0.
# Defaults to -2.0.
#LIBRESPOT_NORMALISATION_THRESHOLD="-2.0"

# The port the internal server advertises over zeroconf 1 - 65535.
# Ports <= 1024 may require root privileges.
#LIBRESPOT_ZEROCONF_PORT=""

# HTTP proxy to use when connecting.
#LIBRESPOT_PROXY=""
```

JasonLG1979, 2022-01-01:
If I were to guess you're not running Raspotify on Debian 11 or something with at least as new a version of systemd as Debian 11. Debian 10 (Buster/old stable) is not supported.

TimHausmann, 2022-01-01:
`pi@mypi:~ $ cat /etc/*-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
`
Well, I will need to upgrade this one…
Thank you Sir. I will veryfy and give feedback.

closed 2022-01-01 by github-actions[bot]

TimHausmann, 2022-01-02:
As suggested, an update to a recent version of Debian fixed the issue.