dtcooper/raspotify #491: Audio Sink Error Connection Refused

spippoli, 2022-01-06:
### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

I'm using raspotify on a rpi-zero W using bluealsa to send the audio to a bluetooth speaker. It worked fine on buster but I just upgraded OS to bullseye and raspotify to v0.31.4~librespot.v0.3.1-34 and it stopped working with the following error:
```
Jan  6 18:10:54 spodisplay librespot[11577]: ALSA lib bluealsa-pcm.c:755:(_snd_pcm_bluealsa_open) Couldn't initialize D-Bus context: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Jan  6 18:10:54 spodisplay librespot[11577]: [2022-01-06T18:10:54Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device bluealsa May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'ENOMEM: Out of memory'
```
my raspotify configuration file looks like this (omitting the commented lines): 
```
LIBRESPOT_QUIET=
LIBRESPOT_AUTOPLAY=
LIBRESPOT_DISABLE_AUDIO_CACHE=
LIBRESPOT_DISABLE_CREDENTIAL_CACHE=
LIBRESPOT_ENABLE_VOLUME_NORMALISATION=
LIBRESPOT_NAME="Spodisplay"
LIBRESPOT_BITRATE="96"
LIBRESPOT_DEVICE="bluealsa"
```

- I verified that `/etc/asound.conf` is correct since if i launch `librespot -d bluealsa` it works fine and I get sound through the bt speaker
- I found issue #459 and verified that the service file contains `RestrictAddressFamilies=AF_UNIX AF_INET AF_INET6 AF_NETLINK`
- The `'ENOMEM: Out of memory'` hint suggests a problem with memory but raspotify is the only process running on my otherwise clean bullseye lite install and currently only ~45MB of RAM are used. 

Should I check anything else? Did I miss some required configuration parameter?

JasonLG1979, 2022-01-06:
How have you installed `bluealsa`? Because `bluealsa` is not in the Buster repo? I don't use `bluealsa` so I can't really offer any help.

spippoli, 2022-01-07:
I have both `bullseye` and `buster` repos in `sources.list`.
As for the error, as far as I understand setting `LIBRESPOT_DEVICE="bluealsa"` is the same as launching `librespot -d bluealsa` so if launching `librespot` directly works my problem could probably be due to some other `raspotify` service configuration parameter limiting resources or permissions. I'll try playing with those to see if it takes me somewhere. 
Any suggestion on where I should focus is welcome!

JasonLG1979, 2022-01-16:
Closing this because it's not really a Raspotify issue.

closed 2022-01-16 by JasonLG1979

AndreaGiardini, 2022-10-30:
Systemd does not play well with dbus when dynamicusers are active https://github.com/systemd/systemd/issues/9503

Removing DynamicUsers and putting back a static user makes it work for me. With dynamicusers librespot doesn't manage to get a dbus connection to the bluealsa device (or at least that's my supposition)