dtcooper/raspotify #494: Can see but will not connect

Bazmundi, 2022-01-10:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

I have installed raspotify onto RPi 3B+ running linux 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l

Spotify on my Windows 10 laptop "sees" the raspotify on the network, but fails to connect when selected.  

I am not using proxies etc.

I have a premium Spotify account etc.  I have setup my spotify username and password in /etc/raspotify/conf.

I have used defaults everywhere else in /etc/raspotify/conf.

I have tried restart, stop/start and reboot etc.

I am using the RCA audio out (not USB and not HDMI).  I have used raspi-config to set audio to the audio jack.

aplay and librespot both report that there are no ALSA compatible ALSA devices (at all).  There did not seem to be a requirement for a sound card?  

So, it seems there may need be something else in the conf file I need setup?

JasonLG1979, 2022-01-11:
The wiki has a [Troubleshooting page](https://github.com/dtcooper/raspotify/wiki/Troubleshooting) that explains how to set the default audio device.

closed 2022-01-11 by github-actions[bot]

Bazmundi, 2022-01-11:
> This question or issue can probably be easily answered or resolved by reading the docs/wiki.

If you pay attention to the information I supplied the evidence is I did use the info at the troubleshooting page.   It directs one to use alsamixer to increase volume, which I indicated I did.  It suggests a means of checking audio is working over the RCA, which I indicated I did.  I indicated that I was not using USB sound so none of the USB sound options appeared relevant.  I am not using pulse audio nor bluetooth.

So, there is actually nothing in the wiki or troubleshooting that does help.

JasonLG1979, 2022-01-11:
> If you pay attention to the information I supplied the evidence is I did use the info at the troubleshooting page.

That kind of attitude will get you nowhere. I paid plenty of attention and nothing you said was really that relevant other than `hw:0` is HDMI and `hw:1` is analog out. I will say it again. The wiki has the info you need to set the analog out as your default.

JasonLG1979, 2022-01-11:
I bet if you followed the wiki, edited `/etc/asound.conf` and added:

```
defaults.ctl.card 1
defaults.pcm.card 1
```

Things would work as you'd expect. But I don't know I'm not really paying attention...

Bazmundi, 2022-01-11:
> I bet if you followed the wiki, edited `/etc/asound.conf` and added:
> 
> ```
> defaults.ctl.card 1
> defaults.pcm.card 1
> ```
> 
> Things would work as you'd expect. But I don't know I'm not really paying attention...

Ah, so ignore the fact that that guidance was directed at people who where trying to setup the USB sound.  Gotcha.

JasonLG1979, 2022-01-12:
> Ah, so ignore the fact that that guidance was directed at people who where trying to setup the USB sound. Gotcha.

The wiki presents some common questions and answers. I can only present the information to you. I can not understand it for you. Maybe more extrapolation and less attitude.