dtcooper/raspotify #495: Options format

erdnuesse, 2022-01-11:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hi, the default files don't specify the format of some vars.
ie. enable Autoplay=

it doesn't say if it's yes\no, true\false, enabled\disabled. 

would be cool if that was fixed. 
as of now, I skip using them personally. but I'm probably not the only one. 

regards, K

JasonLG1979, 2022-01-12:
> Hi, the default files don't specify the format of some vars.
ie. enable Autoplay=

> it doesn't say if it's yes\no, true\false, enabled\disabled.

There are 2 types of "Options" options and flags. Options take a value, flags do not.

The config is an environment file. The options and flags are environment variables. Environment variables are `KEY=VALUE` pairs. The value of a flag is not evaluated the presence or lack of presence of the flag is all that matters.

> okay, I'm stupid. it's not in each line anymore like the other options, it's above, well, it's inconsistent.

> I thought when all options are explained inline, I didn't expect this to be in the text above.

A comment above a config value is a pretty common convention?

For example...

```
# This is a comment explaining option foo. "bar" is the default value.
# It is commented out so it won't do anything until you uncomment it
# by deleting the # in front of it.
# LIBRESPOT_FOO="bar"
```

I shouldn't have to say this but the above is not a real option...

It's all explained in the config file?

closed 2022-01-12 by JasonLG1979

erdnuesse, 2022-01-12:
What I was trying to say was: 
First, I was stupid for mentioning, that there was something missing. 

Secondly, for enable Autoplay, the default \ expected values (on\off) are not in the line above it, or what it defaults to. 
In this segment, on or off is only mentioned once in a segment above. 

As a result, there are two conventions in the file. On some options, there are defaults described, including allowed options, and in the other segment, the on\off is mentioned only once, atop of multiple options. 

The lack of consistency was the thing that threw me off. 

My suggestion is to describe each item right on top of it, to keep it consistent.

JasonLG1979, 2022-01-12:
Flags do not take values. This functions exactly like the command line options and flags except that they are environment variables. When translating command line flags to environment variables you could set the value to anything you want since flag values are not evaluated. That is how it is done, that is the convention. The default state is for the flag to not be present.

JasonLG1979, 2022-01-12:
FOO="true"
FOO="bar"
FOO="false"
FOO="absolutely anything"
FOO=

All mean the same thing if FOO is a flag.

JasonLG1979, 2022-01-12:
I would suggest reading up on the concepts of environment variables and command line options and flags.