dtcooper/raspotify #496: Update wiki article for Bullseye packet issues

BlackYps, 2022-01-12:
### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

bluealsa is not available in the repositories of Bullseye. At least in the Light version that I use. This should be mentioned in this wiki article in the [headless section](https://github.com/dtcooper/raspotify/wiki/Play-via-Bluetooth-Speaker-on-Raspberry-Pi-Desktop#play-via-bluetooth-speaker-without-raspberry-pi-desktop). For the same reason pi-btaudio is unusable at the moment because its install fails. This should also be mentioned. See the issue here: https://github.com/bablokb/pi-btaudio/issues/8

JasonLG1979, 2022-01-12:
The wiki is editable. You don't need to file a bug report to update it.

closed 2022-01-12 by JasonLG1979

BlackYps, 2022-01-12:
Oh, I didn't even realize