dtcooper/raspotify #498: Pipe backend not working

timglong, 2022-01-16:
### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

I'm not having any luck using the pipe backend. I found and read #458 and #468, read the wiki entry and changed `LIBRESPOT_DEVICE` to `/var/run/snapfifo`, but it seems Raspotify can't write to the locations specified. Using `LIBRESPOT_DEVICE="/var/run/snapfifo"` gives me an `os error 30` and I either receive an `os error 2` or an `os error 30` if I point at/create a new folder inside `/var/run/`.

`Jan 16 15:45:02 PiOffice librespot[796]: [2022-01-16T21:45:02Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: Read-only file system (os error 30)`

`Jan 16 16:08:38 PiOffice librespot[858]: [2022-01-16T22:08:38Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: No such file or directory (os error 2)`

Previously I was using `BACKEND_ARGS="--backend pipe --device /tmp/snapfifo"` and this worked very nicely.

After the change I'm using `LIBRESPOT_BACKEND="pipe"` and `LIBRESPOT_DEVICE="/var/run/snapfifo"` as described above.

This is a brand new Bullseye install and a brand new Raspotify install on a Pi4. I'm hoping I'm just missing something simple as I was looking forward to playing around with some of the new options.

JasonLG1979, 2022-01-16:
Ok, stupid question:

Have you configured snapcast to put it's fifo in `/var/run/`?

timglong, 2022-01-16:
As far as I'm aware, yes. I'm not super familiar with the workings of pipe/fifo, but it sounds like the creation of the connection is NOT being handled by Raspotify, so maybe I'm directing my question to the wrong place.

That seems like a very reasonable question!

JasonLG1979, 2022-01-17:
> As far as I'm aware, yes. I'm not super familiar with the workings of pipe/fifo, but it sounds like the creation of the connection is NOT being handled by Raspotify, so maybe I'm directing my question to the wrong place.
> 
> That seems like a very reasonable question!

If you don't know for sure if you've configured snapcast to create the fifo in `/var/run` you probably haven't. That's not it's default location. You can look at the config file changes for clues as to what you should change on the snapcast side:

https://github.com/jgabriel98/spotifypiHome/pull/7/files

timglong, 2022-01-17:
I shouldn't have qualified the statement--yes, Snapcast and Raspotify are pointing at the same fifo. I've tried a number of different locations for the fifo; in all but the original `/tmp/' location, Snapcast fails to create the fifo. Raspotify will not read the fifo in the `/tmp/` location.

The code linked above appears to be using earlier versions of both Raspotify and Snapcast, and the conf modifications don't work, at least in my situation.

From https://github.com/badaix/snapcast/issues/737, running `sudo sysctl -n fs.protected_fifos` returns 1 on the Pi that was just updated to Bullseye and 0 on a client running Buster. I haven't tried the workaround listed there yet.

I'm not sure how to proceed.

timglong, 2022-01-17:
Also, while attempting to get something besides white noise/garbled sound, I noticed that the wiki references modifying `/etc/asound.conf` vs. `/usr/share/alsa/alsa.conf` when using a USB DAC. On both of my systems (Bullseye and Buster) the former file (referenced in the wiki) is empty, while the latter file looks like what I was expecting. Not sure if this should be updated.

closed 2022-01-17 by JasonLG1979

footy42, 2022-02-01:
> Ok, stupid question:
> 
> Have you configured snapcast to put it's fifo in `/var/run`?

why is /var/run recommended?  it gets cleared on reboot.

JasonLG1979, 2022-02-01:
> why is /var/run recommended? it gets cleared on reboot.

`/tmp` get's cleared pretty often too automatically.

http://www.h-online.com/open/news/item/Linux-distributions-to-include-run-directory-1219006.html

JasonLG1979, 2022-02-01:
My understanding is that `/run` (which is what `/var/run` links to) links to is supposed to be a tmpfs for apps to store things they need to run. (hence the name the name I guess?) in the case of snapcast it means that they would need to create the fifo when the service 1st started at least once after each boot.

I don't make the rules. I'm just following best and current practices.

footy42, 2022-02-01:
Makes sense. Thanks :+1:

JasonLG1979, 2022-02-19:
> If anyone figures this out, please let me know. The snapcast folks don't seem to care, so for the time being I'm going to revert to Raspotify 0.16 or I guess figure out how to build/configure librespot on my own. It was a good year of multi-room audio, RIP.

You don't have to revert. In a pinch if all else fails (with a little effort and some trial and error) you can disable the Raspotify service and write your own systemd service file and use the librespot binary packaged with Raspotify. Remember Raspotify is just a compiled binary, a systemd service file and a config. There's nothing stopping you from using the binary however you like.

timglong, 2022-04-24:
@JasonLG1979 I think the best way to handle this moving forward is to just not use pipe--I was able to get my system back up and running by using librespot directly, just like you suggested. Maybe we could add this to the wiki?

Here's the snapserver conf line for my setup, including all the parameters I usually set with Raspotify:

`source = spotify:///usr/bin/librespot?name=Spotify&username=***&password=***&devicename=***&bitrate=320&wd_timeout=7800&volume=100&normalize=false&autoplay=false&disable_audio_cache=true&params=--device-type%20avr&params=--volume-ctrl%20fixed&params=--format%20S32&params=--disable-discovery%20true`

The nice thing about this is you can use the `params` argument to add any options from librespot very easily.

JasonLG1979, 2022-04-24:
Don't forget to disable the raspotify service unless ofc you want 2 instances of librespot running?

JasonLG1979, 2022-04-24:
I'd love to see this in the wiki. Feel free to add a page about snapcast integration.

timglong, 2022-04-25:
To disable it's just `sudo systemctl disable raspotify`, right?

I want to do a bit more testing but once that's done I will make a wiki post.

JasonLG1979, 2022-04-25:
Yep.

JasonLG1979, 2022-04-25:
Sounds good.