dtcooper/raspotify #499: Error after update: couldn't parse packet from 192.168.0.174:5353: type 47 is invalid

wieluk, 2022-01-17:
### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

I updated to the newest version and now I get this error message:

`Jan 17 17:41:40 VpnPi librespot[18501]: [2022-01-17T16:41:40Z WARN  libmdns::fsm] couldn't parse packet from 192.168.0.174:5353: type 47 is invalid`

Now raspotify doesn't show up on my phone or computer anymore.

Does anyone know how to fix this?

JasonLG1979, 2022-01-17:
What is `192.168.0.174`?

JasonLG1979, 2022-01-20:
Not much in `libmdns` (the zeroconf implementation that `librespot` uses) has changed in a while. I'm not a mdns expert but my gut tells me that it's not a `librespot` issue since your phone for some reason seems to be sending invalid packets, or at least they're invalid by the time `librespot` gets them?

Stupid question: Have you tried resetting everything?

wieluk, 2022-01-21:
yea it works now but only when I add my spotify account

JasonLG1979, 2022-01-22:
> yea it works now but only when I add my spotify account

Then something on your network is mangling mdns.

JasonLG1979, 2022-02-08:
Ok I'm going assume from the lack of response that either you've solved the problem, it's solved itself or you've given up.

No matter the case I'm going to close this since it doesn't appear to be a `librespot` and certainly isn't a Raspotify issue.

closed 2022-02-08 by JasonLG1979

benklop, 2022-04-22:
i have this issue as well. I just installed and haven't done any configuration yet.

benklop, 2022-04-22:
https://github.com/librespot-org/libmdns/issues/19 appears relevant. looks like switching to avahi fixes it

JasonLG1979, 2022-04-22:
> [librespot-org/libmdns#19](https://github.com/librespot-org/libmdns/issues/19) appears relevant. looks like switching to avahi fixes it

It may be. It is possible to build with the `with-dns-sd` feature but that uses the way way deprecated Apple Bonjour compatibility layer of Avahi so that's a no-go. What really needs to happen is for someone to rewrite the mDNS bits to use the proper Avahi API.

benklop, 2022-04-29:
it does in fact fix it for me.