dtcooper/raspotify #500: Onevent doesn't work anymore after update

vasilisvg, 2022-01-18:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

In the previous version of Raspotify I was able to execute a script that needed sudo permissions with the --onevent argument. The trick that worked for me was giving the raspotify user these sudo permissions. 
In the new version there doesn't seem to be such a user. Which results in all kinds of permission errors that I haven't been able to fix yet.
I was hoping that there was a simple, similar fix, but I can’t really find an answer. Not in the documentation and not in the (closed) issues. Is there a new user that I can give more rights? Is there any other way to give more permissions when the script is executed? Or should I better ask this at Librespot?

JasonLG1979, 2022-01-18:
Giving a system service user sudo privileges is not really a thing you do. Does the script ***really*** need root privileges or is just easier to use the big root hammer?

What does the script do?

`librespot` in Raspotify operates as a dynamic systemd user now. It purposely operates under the least possible privileges. There's no way that it will execute a script as root.

JasonLG1979, 2022-01-18:
You would basically need to completely rewrite the service unit file.

vasilisvg, 2022-01-19:
> What does the script do?

The script shows some metadata about the sons that’s playing on an e-paper display. It needs extra privileges to terminate another script that might be working with the display at the same time. 
I guess I'll have to rewrite the script, or find another way to make it work. Some workaround must be possible. Thank you for your quick answer!

JasonLG1979, 2022-02-08:
@vasilisvg If you've come up with a solution then please do share the script and the solution so that it might be other who might have similar issues. I'd be personally interested because I'd bet an e-ink display would be great for Spotify track info since it doesn't change very often and e-ink displays being low power.

vasilisvg, 2022-02-08:
Unfortunately I haven't been able to find a workaround for the new Raspotify.

JasonLG1979, 2022-02-08:
You could always just use a drop in service override?

When I get home from work today I'll see what I can come up with.

vasilisvg, 2022-02-14:
Did you by any chance manage to work out this _drop in service override?_ I have no idea how this works, tbh…

JasonLG1979, 2022-02-14:
Sorry, I got sidetracked. Give me a few mins.

JasonLG1979, 2022-02-14:
Well, we may have to come up with another way. Turns out overrides don't function as expected on because of a systemd bug.

JasonLG1979, 2022-02-14:
An alternative is to have a script just echo/print to the systemd log and then tail the log and do whatever you want with that info. That way whatever is tailing the log can run as root and Raspotify does not have to.

```python3
#!/usr/bin/python3
import os
import json

# PLAYER_EVENT:
# changed, started, stopped,
# playing, paused, preloading, volume_set
print(
    json.dumps(
        {
            'event': os.getenv('PLAYER_EVENT') or '',
            'old_track_id': os.getenv('OLD_TRACK_ID') or '',
            'new_track_id': os.getenv('TRACK_ID') or '',
            'duration_ms': os.getenv('DURATION_MS') or '',
            'position_ms': os.getenv('POSITION_MS') or '',
            'volume': os.getenv('VOLUME') or '',
        }
    )
)
```

That should dump something like this to the logs:
```
{"event": "volume_set", "old_track_id": "", "new_track_id": "", "duration_ms": "", "position_ms": "", "volume": "58958"}
{"event": "started", "old_track_id": "", "new_track_id": "0IVAlb0cFCH1cQ974iR1TU", "duration_ms": "", "position_ms": "", "volume": ""}
{"event": "playing", "old_track_id": "", "new_track_id": "0IVAlb0cFCH1cQ974iR1TU", "duration_ms": "223546", "position_ms": "171571", "volume": ""}
{"event": "changed", "old_track_id": "0IVAlb0cFCH1cQ974iR1TU", "new_track_id": "6C3yPXYG2NJYlIYVe5J7hv", "duration_ms": "", "position_ms": "", "volume": ""}
{"event": "playing", "old_track_id": "", "new_track_id": "6C3yPXYG2NJYlIYVe5J7hv", "duration_ms": "309946", "position_ms": "0", "volume": ""}
{"event": "changed", "old_track_id": "6C3yPXYG2NJYlIYVe5J7hv", "new_track_id": "2YC6ET3q1F29B0V7UcPV70", "duration_ms": "", "position_ms": "", "volume": ""}
{"event": "playing", "old_track_id": "", "new_track_id": "2YC6ET3q1F29B0V7UcPV70", "duration_ms": "166653", "position_ms": "0", "volume": ""}
```

As long as you leave the quiet flag turned on there should not be much of anything else logged by raspotify other than that.

vasilisvg, 2022-02-15:
Thanks, I'll give it a try. I hope I'll find some time this weekend.

JasonLG1979, 2022-02-19:
> I also hit this recently. I don't necessarily need sudo but I do need to save some state to disk. Is there some file path I can write to now that raspotify is a dynamic systemd user? (The path needs to be accessible to another user)

In a pinch if all else fails (with a little effort and some trial and error) you can disable the Raspotify service and write your own systemd service file and use the librespot binary packaged with Raspotify. Remember Raspotify is just a compiled binary, a systemd service file and a config. There's nothing stopping you from using the binary however you like.

just-inlewis, 2022-02-19:
> 

You've done such a great job hardening the runtime though! I was able to accomplish my goal by adding a service override.

`[Service]
ReadWriteDirectories=+/MYPATH`

JasonLG1979, 2022-02-19:
> You've done such a great job hardening the runtime though! I was able to accomplish my goal by adding a service override.
> 
> `[Service] ReadWriteDirectories=+/MYPATH`

Nice!!!

closed 2022-05-28 by github-actions[bot]