dtcooper/raspotify #502: Problems with playing audio from USB speaker

blue1stone, 2022-01-20:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hey there!
First of all, great job with the package, it is a charm to use!

However, I am encountering a problem when I want to play the music not through the normal 3.5mm jack on my Raspberry Pi 3B+ but rather through my USB speakers.

When I try running
`librespot --username **** --password ****` it works like a charm, however if I try to run it using the raspotify service, it gives me the following error:
```
Jan 20 17:24:36 picroft librespot[2309]: E: [pulseaudio] caps.c: Assertion 'setresuid(uid, uid, uid) >= 0' failed at daemon/caps.c:57, function pa_drop_root(). Aborting.
Jan 20 17:24:36 picroft librespot[2309]: [2022-01-20T17:24:36Z ERROR librespot_playback::player] Audio Sink Error Invalid Parameters: <AlsaSink> Device default Unsupported Format S16LE (S16), ALSA function 'snd_pcm_hw_params_set_format' failed with error 'EINVAL: Invalid argument'
``` 

My `/etc/asound.conf` looks as follows:
```
pcm.!default {
 type hw
 card 1
 device 0
}

ctl.!default {
 type hw
 card 1
 device 0
}
```
My `/lib/systemd/system/raspotify.service`:
```
[Unit]
Description=Raspotify (Spotify Connect Client)
Documentation=https://github.com/dtcooper/raspotify
Documentation=https://github.com/librespot-org/librespot
Documentation=https://github.com/dtcooper/raspotify/wiki
Documentation=https://github.com/librespot-org/librespot/wiki/Options
Wants=network.target sound.target
After=network.target sound.target

[Service]
DynamicUser=yes
SupplementaryGroups=audio

Restart=always
RestartSec=10

# Some of these may be implied by DynamicUser=yes
RestrictRealtime=true
RestrictSUIDSGID=true
RestrictNamespaces=true
RestrictAddressFamilies=AF_UNIX AF_INET AF_INET6 AF_NETLINK

ProtectHostname=true
ProtectControlGroups=true
ProtectKernelLogs=true
ProtectProc=invisible
ProtectHome=true
ProtectSystem=strict
ProtectClock=yes

DevicePolicy=strict
DeviceAllow=char-alsa rw
DeviceAllow=/dev/null r
DeviceAllow=/dev/random r
DeviceAllow=/dev/urandom r

UMask=077
CacheDirectoryMode=0700
CacheDirectory=%N
StateDirectoryMode=0700
StateDirectory=%N
ConfigurationDirectoryMode=0600
ConfigurationDirectory=%N

PrivateTmp=true
PrivateUsers=true

ProcSubset=pid
NoNewPrivileges=true
PermissionsStartOnly=true
LockPersonality=true
MemoryDenyWriteExecute=true
RemoveIPC=true

CapabilityBoundingSet=

#SystemCallArchitectures=native
SystemCallFilter=@system-service
SystemCallFilter=~@privileged @resources
SystemCallErrorNumber=EPERM

Environment=LIBRESPOT_NAME="%N (%H)"
Environment=LIBRESPOT_CACHE=%C/%N
Environment=LIBRESPOT_SYSTEM_CACHE=%S/%N

EnvironmentFile=-%E/%N/conf

ExecStart=/usr/bin/librespot

[Install]
WantedBy=multi-user.target
```
And my `/etc/raspotify/conf`:
```
# /etc/raspotify/conf -- Arguments/configuration for librespot

# A non-exhaustive list of librespot options and flags.

# Please see https://github.com/dtcooper/raspotify/wiki &
# https://github.com/librespot-org/librespot/wiki/Options
# for configuration details and a full list of options and flags.

# You can also find a full list with `librespot -h`.

# To avoid name collisions environment variables must be prepended with
# `LIBRESPOT_`, so option/flag `foo-bar` becomes `LIBRESPOT_FOO_BAR`.

# Invalid environment variables will be ignored.

# Raspotify defaults may vary from librespot defaults.
# Commenting out the environment variable will fallback to librespot's default
# unless otherwise noted.

# Flags are either on (uncommented) or off (commented),
# their values are otherwise not evaluated (but the "=" is still needed).

# Only log warning and error messages.
#LIBRESPOT_QUIET=

# Automatically play similar songs when your music ends.
#LIBRESPOT_AUTOPLAY=

# Disable caching of the audio data.
# Enabling audio data caching can take up a lot of space
# if you don't limit the cache size with LIBRESPOT_CACHE_SIZE_LIMIT.
# It can also wear out your Micro SD card. You have been warned.
#LIBRESPOT_DISABLE_AUDIO_CACHE=

# Disable caching of credentials.
# Caching of credentials is not necessary so long as
# LIBRESPOT_DISABLE_DISCOVERY is not set.
#LIBRESPOT_DISABLE_CREDENTIAL_CACHE=

# Play all tracks at approximately the same apparent volume.
#LIBRESPOT_ENABLE_VOLUME_NORMALISATION=

# Enable verbose log output.
#LIBRESPOT_VERBOSE=

# Disable zeroconf discovery mode.
#LIBRESPOT_DISABLE_DISCOVERY=

# Options will fallback to their defaults if commented out,
# otherwise they must have a valid value.
# Play all tracks at approximately the same apparent volume.
#LIBRESPOT_ENABLE_VOLUME_NORMALISATION=

# Enable verbose log output.
#LIBRESPOT_VERBOSE=

# Disable zeroconf discovery mode.
#LIBRESPOT_DISABLE_DISCOVERY=

# Options will fallback to their defaults if commented out,
# otherwise they must have a valid value.

# Device name.
# Raspotify defaults to "raspotify (*hostname)".
# Librespot defaults to "Librespot".
LIBRESPOT_NAME="Smart Speaker"

# Bitrate (kbps) {96|160|320}. Defaults to 160.
#LIBRESPOT_BITRATE="160"

# Output format {F64|F32|S32|S24|S24_3|S16}. Defaults to S16.
#LIBRESPOT_FORMAT="S16"

# Displayed device type. Defaults to speaker.
#LIBRESPOT_DEVICE_TYPE="speaker"

# Limits the size of the cache for audio files.
# It's possible to use suffixes like K, M or G, e.g. 16G for example.
# Highly advised if audio caching isn't disabled. Otherwise the cache
# size is only limited by disk space.
#LIBRESPOT_CACHE_SIZE_LIMIT=""

# Audio backend to use, alsa or pipe. Defaults to alsa.
#LIBRESPOT_BACKEND="alsa"

# Username used to sign in with.
# Credentials are not required if LIBRESPOT_DISABLE_DISCOVERY is not set.
LIBRESPOT_USERNAME="****"

# Password used to sign in with.
LIBRESPOT_PASSWORD="****"

# Audio device to use, use `librespot --device ?` to list options.
# Defaults to the system's default.
#LIBRESPOT_DEVICE="default"

# Initial volume in % from 0 - 100.
# Defaults to 50 For the alsa mixer: the current volume.
#LIBRESPOT_INITIAL_VOLUME="50"

# Volume control scale type {cubic|fixed|linear|log}.
# Defaults to log.
#LIBRESPOT_VOLUME_CTRL="log"

# Range of the volume control (dB) from 0.0 to 100.0.
# Default for softvol: 60.0.
# For the alsa mixer: what the control supports.
#LIBRESPOT_VOLUME_RANGE="60.0"

# Pregain (dB) applied by volume normalisation from -10.0 to 10.0.
# Defaults to 0.0.
#LIBRESPOT_NORMALISATION_PREGAIN="0.0"

# Threshold (dBFS) at which point the dynamic limiter engages
# to prevent clipping from 0.0 to -10.0.
# Defaults to -2.0.
#LIBRESPOT_NORMALISATION_THRESHOLD="-2.0"

# The port the internal server advertises over zeroconf 1 - 65535.
# Ports <= 1024 may require root privileges.
#LIBRESPOT_ZEROCONF_PORT=""

# HTTP proxy to use when connecting.
#LIBRESPOT_PROXY=""
```

Help is greatly appreciated!
Best regards, Elias Ball.

JasonLG1979, 2022-01-20:
Judging by the error message you're running PulseAudio on your system.

The reason it works for a user with PulseAudio is because PulseAudio is also a user level service. That's why `librespot --username **** --password ****` works when you call it manually.

Raspotify runs `librespot` as a system level service so that it runs unattended at boot with no need for a user to login.

If you want to run Raspotify with PulseAudio you either need to make Pulseaudio a system level service as mentioned in the wiki or make Raspotify a user level service.

blue1stone, 2022-01-21:
Thank you for the quick reply!
I am trying to make Raspotify a user-level service, however, I can't seem to get it working with the information I found so far. Could you point me in the right direction?
Sorry for the inconvenience, I am pretty new to Linux...

JasonLG1979, 2022-01-22:
I have no experience really as far as using a Raspberry Pi as a desktop. I've only ever used used them as headless systems. It's the intention of Raspotify to run as a system level service on a headless system. It's not really designed to run on desktop system with PulseAudio. That doesn't mean it can't be done ofc. 

[`librespot` has a very simple user-level systemd service file example](https://github.com/librespot-org/librespot/blob/dev/contrib/librespot.user.service).

[The ArchWiki is also a great general resource for all things Linux](https://wiki.archlinux.org/title/systemd/User#Writing_user_units).

Although Debian is a little behind Arch so you may find something that don't transfer exactly from Arch to Debian.

blue1stone, 2022-01-22:
Thank you!
I have decided to run `librespot` through the user service script you pointed me to and it works great.
Thanks a lot for the help!

closed 2022-01-22 by blue1stone